MarlinFirmware/Marlin #8734: Using D7 as extruder fan on MKS_BASE

ismangil, 2017-12-09:
On my board MKS Gen L, which is defined as board 40 MKS_BASE, I want to use extruder fan feature.

Is this the right way, on Configuration_adv.h

```#define E0_AUTO_FAN_PIN MOSFET_D_PIN```

I am also interested in the controller fan feature, there are some extra pins on this board, what kind of pins can be used to control a controller fan feature?

Tannoo, 2017-12-10:
PWM, I believe.

bjarchi, 2017-12-10:
> PWM, I believe.

I think this is correct, although there's also a 'soft pwm' option for fans that might open up more pins.

Depending on the size and number of fan(s) you might need a MOSFET or etc. between the pin and the fan to get enough current capacity. 

On the GEN_L (which BTW recently got it's own board define - very nearly identical to BOARD_MKS_BASE - if you're on bugfix-1.1.x) D7 is already the default pin for a *part cooling* fan - it even has the MOSFET built into the board so it can switch higher current.

When you say extruder fan, do you mean a part cooling fan, or do you want your controller to switch the extruder cooling fan(s) for you? That is, the fan(s) that cool the extruder cold bar(s)/heatsink(s)?

If it's the former, that's already the expected use case and shouldn't need any changes to the configuration. Should just work IIRC. Otherwise, you probably want to select another pin unless you want to switch the extruder cooling fan **only** and not a part cooling fan.

ismangil, 2017-12-10:
@bjarchi I want to wire three fans:

- Part cooling fan, I was going to wire this into the ports marked FAN, datasheet calls this D9 which is same as RAMPS. Will be controlled by G-Code

- Extruder fan, cooling the heatsink. This fan is supposed to be on all the time, I can wire it directly to power pins on the board but I want Marlin to only turn it on at certain temperature. I see this feature in advance conf as Extruder auto fan. I was going to wire this to D7 which in Gen L is marked HE1 for the second extruder heater. 

- Controller board cooling fan feature, only turns on when steppers are on. This fan I don't know where else I can tell Marlin to point to. Looks like there's a Mosfet board addon that turns servo pins D4, D5 suitable to drive fan outputs. Is this the only way for this third fan?

ismangil, 2017-12-10:
@bjarchi OK I can see now if I configure only one hotend, automatically D7 is now FAN1_PIN in addition to usual D9  FAN_PIN

Still short of FAN2_PIN :) I'll look into extender boards.

Thanks!

bjarchi, 2017-12-10:
> @bjarchi OK I can see now if I configure only one hotend, automatically D7 is now FAN1_PIN in addition to usual D9 FAN_PIN

Ah yes, if you only have one extruder then it can use two of the FET channels for fans - D7 as well as whatever would have been the heater for E1 (D9 apparently). 

> Still short of FAN2_PIN :) I'll look into extender boards.

I think this is how you're going to have to go. The problem isn't availability of pins, it's the amount of current they can source/sink. D7 and D9 both have large-ish MOSFETS on the GEN_L that can switch a bunch of current, so they're safe to drive a fan. Other digital pins won't have this - the only other two pins with built-in FETs are E0 heater, and bed heater. So to use another pin for fan drive you will probably need an external MOSFET. 

Are you really wedded to having MARLIN control the fans for extruder cooling and/or controller cooling? I have both of those continuously on for my machine; they're not too loud. Controller especially could use a quiet and lower airflow fan if you're not pushing your steppers too hard and you have the airflow well controlled.

ismangil, 2017-12-10:
It's not about the loudness really, I am thinking this as a sleep/standby mode, and less wear on these cheapish fans. In the meanwhile, yes I'll just wire any extra fans always on, and switch off the whole printer.

closed 2017-12-10 by ismangil

zalexzperez, 2020-03-27:
Hmm, interesting. I have a 24V powered Ender 3 with an MKS Gen L board. All 12V fans except the blower part cooling fan.

I'm also going to set the D7 pin as my E0_AUTO_FAN_PIN

I'll be controlling my extruder fan along with other 2 case fans. These three 12V fans are currently powered from a Buck converter so I'll have to power this converter from the Extruder1 port (D7).

I won't be able to cool down my drivers when the hotend temperature is below 50º but I don't think that would be an issue, as I won't be moving my motors too much with a cold machine!