MarlinFirmware/Marlin #8737: problem with manual control of extruder steppers with mixing extruder

cars4fun2001, 2017-12-10:
when trying to setup mixing extruder if I set conditionals_lcd.h to
 #elif ENABLED(MIXING_EXTRUDER)        // Multi-stepper, unified E axis, one hotend
    #define HOTENDS      1
    #define E_STEPPERS   3
    #define E_MANUAL     3
    #define TOOL_E_INDEX 0
in order to have manual control of the 3 steppers on the extruder I get

Arduino: 1.8.3 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ultralcd.cpp: In function 'void manual_move_to_current(AxisEnum, int8_t)':

ultralcd.cpp:2564: error: lvalue required as left operand of assignment

           manual_move_e_index = eindex >= 0 ? eindex : active_extruder;

                               ^

exit status 1
lvalue required as left operand of assignment

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

if I set it to 

 #elif ENABLED(MIXING_EXTRUDER)        // Multi-stepper, unified E axis, one hotend
    #define HOTENDS      1
    #define E_STEPPERS   3
    #define E_MANUAL     1
    #define TOOL_E_INDEX 0

it works but I only have manual control of E0 extruder stepper and its the only one that will  turn when I try to manually advance the extruder

thinkyhead, 2017-12-16:
> in order to have manual control of the 3 steppers on the extruder I get

Hacking `Conditionals_LCD.h` is bound to fail.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek

AnHardt, 2019-07-06:
Set up the mixing relation to use only one filament. ve[0,1,0]
Commit that relation to a virtual tool.
Select that virtual tool.
Move that tool.

If you have virtual_tools > mixing_extruders, there is a predefined virtual tool using only one filament.