MarlinFirmware/Marlin #8739: Nozzle hits bed after G29 with PROBE_MANUALLY and AUTO_BED_LEVELING_BILINEAR / AUTO_BED_LEVELING_LINEAR

lumbric, 2017-12-10:
### Bug Report
- Description: Using a delta printer a G29 command moves the head below Z0 (but in center X0 Y0) with high speed when PROBE_MANUALLY and either AUTO_BED_LEVELING_BILINEAR  or AUTO_BED_LEVELING_LINEAR is enabled. Different behaviour but similar setup is described in #8732.
- Expected behaviour: Head does move to the first probe point to a certain level above Z0 to avoid crashing the bed.
- using v1.1.6
- Actual behaviour: Head crashes bed after a G29.
- Steps to reproduce:
  - Configure PROBE_MANUALLY and AUTO_BED_LEVELING_BILINEAR  or AUTO_BED_LEVELING_LINEAR
  - G28
  - G29

[Configuration.h](https://github.com/lumbric/Marlin/blob/5c94a09b63b89b530b2834ca9a1190c9d70925e4/Marlin/Configuration.h)  [Configuration_adv.h](https://github.com/lumbric/Marlin/blob/5c94a09b63b89b530b2834ca9a1190c9d70925e4/Marlin/Configuration_adv.h)

AnHardt, 2017-12-10:
`DELTA_HEIGHT` - `Z_CLEARANCE_BETWEEN_PROBES` > "real delta height"?

lumbric, 2017-12-10:
Thanks for the hint! I was searching already how the offset to the plate can be configured. I didn't know that Z_CLEARANCE_BETWEEN_PROBES is used also for PROBE_MANUALLY.
https://github.com/lumbric/Marlin/blob/5c94a09b63b89b530b2834ca9a1190c9d70925e4/Marlin/Configuration.h#L841

lumbric, 2017-12-19:
Yeah, this seems to work better with 1.1.7! I will test and then comment again.

lumbric, 2018-01-12:
Movements are still a bit surprising. Head moves up and down. But it does no hit the bed and moves correctly in XY. I was able to test with  AUTO_BED_LEVELING_BILINEAR. The map/correction looks as expected. For AUTO_BED_LEVELING_LINEAR something seems to be wrong with the radius, might be my config... But the bug here is fixed.

closed 2018-01-12 by lumbric