MarlinFirmware/Marlin #8740: [FR] TMC2130 StallGuard as Endstop for CoreXY

Freddrup, 2017-12-10:
------------------------------------------------------------
### Feature Request
Is it possible to add a Sensorless endstop option to use the TMC2130's Stallguard feature with CoreXY setups ?
After looking a bit closer at the configuration files, I haven't seen any option for that; only for standard Cartesian setups, where you just place a wire from the driver's DIAG1 pin to the X/Y endstop pin (and, I may be wrong, but I think this wouldn't work with a CoreXY setup). 

P.S.: Or maybe I haven't looked deeply enough at the config' files ^^

teemuatlut, 2017-12-10:
I don't think there's anything currently that would stop you from compiling it that way but I don't know how it would behave because movement in X or Y is always executed by both motors and the stallGuard would trigger both X and Y endstops. I've got a CoreXY frame that I'm designing so I'll look into it at some point.

Freddrup, 2017-12-10:
Compiling Marlin with CoreXY, TMC2130 and StallGuard as endstop options defined works without reporting any errors, but, as stated by teemualtut :
> I don't know how it would behave because movement in X or Y is always executed by both motors and the stallGuard would trigger both X and Y endstops. 

I plan to build a custom 3D printer, and I considered using a CoreXY setup (because of the "design simplicity"), so having it behaving correctly would be nice ^^

bjarchi, 2017-12-11:
It seems like it will probably take some code to get this working properly on core machines (assuming it doesn't now), but it should be pretty simple. FAST_HOME would have to be disabled (whatever that option is that initially homes x and y together), then I think you would just OR the stallguard signals from each motor as the 'endstop' proxy signal.

teemuatlut, 2017-12-11:
It might be as simple as adding some "debounce" in there. Or it might not even need that.
What I think would happen is when homing let's say in the X direction, the carriage would move and collide with the frame, as it's supposed to, and you'd then have pulses on both X and Y endstop. Now, Marlin doesn't actually need to care that the Y endstop was triggered. All it needs to look for is that the X endstop signal went active. Then we do the same with Y direction and not care about X endstop signal. What I don't know is that when the X endstop is triggered, how long a pause do we need so that Marlin doesn't immediately see the pulse on Y as well.

bjarchi, 2017-12-11:
Well on core, since both motors are involved in moving both axes, you don't have distinct 'x endstop' and 'y endstop' signals. This is why you have to disable fast homing - when homing either axis, you expect to see StallGuard signals from either or both motors.

Thus, I think the program logic would be something like this (obviously simplified pseudocode):
```
home_x {
  while (!stallguard_a && !stallguard_b) {
    move_x_toward_home()
  }
  save_home_position_x()
}
```
And of course similar logic for Y.

campy, 2017-12-13:
Hi, today I test tmc2130 in corexy for x and y axis works ok, you don't need change anything for home, only you need down sensibility of tmc2130 homming features to 5 or 3 value (default it's 19).
This way homming it's very soft

Freddrup, 2017-12-13:
That's an excellent thing to know !!

ismangil, 2017-12-13:
@campy where did you purchase the drivers, Waterott or elsewhere?

What is the proper configuration of the pins/soldered pads?

Thanks!

campy, 2017-12-14:
Hi, @ismangil  purchase in waterott, and follow toms3d tutorial for configure it

https://toms3d.org/2017/12/09/tmc2130-guide/

Now only i need know how to change pins for work with LCD, because AUX3 its busy with tmc2130

lonelymyp, 2017-12-14:
It is possible to connect one driver to both endstop input
In the CoreXY, it does not matter which driver will give the signal, the signal will be from both driver during parking on each axis.

larsleukert, 2017-12-15:
@campy 
You only need to change the pins for CS in you pins_RAMPS.h that are on AUX3
I used the one for E0/E1 as i wont use a TMC2130 to extrude.
#define X_CS_PIN           53 -> 42
#define Y_CS_PIN           49 -> 44
The other pins vom AUX3 (SPI) can be used together with the display. I replaced the 8pin female header on the smartdisplay with a stackable one and connected the 3 SPI pins on top

@lonelymyp 
tried that yesterday, but had no success so far. Need to get deeper into this - maybe i will retry with a blank fw config instead of my patched hypercube config

thinkyhead, 2017-12-15:
On a CoreXY I don't think there's any way for the steppers to distinguish whether it was the X axis or the Y axis that was interrupted. So when homing with CoreXY the XY endstops must be combined. Then, if only the X axis is moving when the "XY endstop" gets triggered, it can assume it's the X axis.

larsleukert, 2017-12-16:
Imho just combining won't solve it.
Currently there is only the diag1 pin of the driver for the homing axis enabled (together with switching to spreadCycle)

`#if ENABLED(SENSORLESS_HOMING)
    #if ENABLED(X_IS_TMC2130)
      if (axis == X_AXIS) tmc2130_sensorless_homing(stepperX);
    #endif
    #if ENABLED(Y_IS_TMC2130)
      if (axis == Y_AXIS) tmc2130_sensorless_homing(stepperY);
    #endif
  #endif
`

larsleukert, 2017-12-17:
So far:
Homing in STEALTHCHOP works kind of (with (X/Y)_HOMING_SENSITIVITY set to 7)
But printing in STEALTHCHOP is not an option on my hypercube: hat to reduce X/Y _CURRENT to 650, acceleration to<1000 and slow down printing not to lose steps.
So i switched to spreadCycle - and homing did not work anymore. M43 E showed that the x endstop was triggered immediately. With much higher (X/Y)_HOMING_SENSITIVITY ~25 homing starts, but x endstop is triggerd to late (y stalls before - missing combination of endstops in code). Reduced (X/Y)_HOMING_SENSITIVITY step by step to 20 -> same problem x endstop was triggered immediately .
I didn't expected that cause the TMC2130s are always switched to spreadCycle while homing even when configured to stealthChop.
Don't understand whats going on, maybe there is a difference in the initialization of the TMCs which leads to this behavior.
Any ideas?

larsleukert, 2017-12-17:
@campy Do you use stealthChop or spreadCycle?

campy, 2017-12-19:
@larsleukert  i use stealthChop and selected diferent sensitiviy for X and Y in my case i try with a Sparkcube coreXY

X_HOMING_SENSITIVITY 3
Y_HOMING_SENSITIVITY 4

Never test spreardcycle

vtuan10, 2017-12-25:
When I use G28 to home, my coreXY first homes in x-direction, but after bumping at the end of X-axis, instead of going straight to Y, the hotend goes diagonally to the middle while moving Y0. 
It seems like the X motor is stopped after homing X (if only one motor rotates in coreXY kinematics, the hotend goes diagonally).

thinkyhead, 2017-12-25:
@vtuan10 You should consider filing an Issue.

vtuan10, 2017-12-25:
I have this issue only when using tmc2130 stallguard as endstops, With normal endstop switches everything works.

KrX3D, 2018-01-12:
hello,

i also got the tmc2130 from china, they are running (only problem when the current is first set to 0 "M906 X0" than i get after moving the axis an PRINTER HALTED error, but setting it to 800 works)

my problem is that moving into the direction to 0 gives me always an "Endstop hit" even setting the sensetivity to 100 and more.

i have STEALTHCHOP enabled and at the moment im trying 

    #define X_HOMING_SENSITIVITY  500
    #define Y_HOMING_SENSITIVITY  500

tried it with values from 3 to 100 many times

i have got an hypercube.

also what i noticed is that the X axis is moving the wrong way like with my a4988 but the y axis is moving right

EDIT: what i also noticed is that my drivers reg are simetimes the same i.e. 0x80:0C:00:00 but sometimes after flashing a new moded version of marlin they show up like this

READ: Driver registers:
READ: 	X = 0x80:0C:00:00
READ: 	Y = 0xC0:0C:00:00

Seri-, 2018-01-21:
I had great success with sensor sensitivity set to 3 on x and y and a slow homing speed of "40x60" with "quick home" disabled on latest Bugfix build. 
I also connected both steppers to both endstop pins so it won't matter which one is triggered.

I have done about 50 test homings and non of them failed.

KrX3D, 2018-01-23:
hi,

where can i config the homing speed/disable quick home?

and what exactly do you mean with

> I also connected both steppers to both endstop pins so it won't matter which one is triggered.

you have connected the pin of both steppers together and connected that to both endstop pins?

alfredanil, 2018-01-24:
@Seri- 
Do you think stallguard homming could work on CoreXZ setup as well? I mean would it be possible to do sensorless bed levelling apart from homming with Core XZ?

thinkyhead, 2018-02-14:
> Do you think stallguard homming could work on CoreXZ setup as well?

If it's implemented for one core, it is implemented for all.

> sensorless bed levelling apart from homming with Core XZ?

Please test #9582 and give some feedback there.

jcphlux, 2018-03-04:
So I see this issue is still open and wanted to know if this issue is close to a resolution. I am design a new CoreXY printer and was leaning towards using tmc2130 and would love to use sensor less homing.

campy, 2018-03-05:
`So I see this issue is still open and wanted to know if this issue is close to a resolution. I am design a new CoreXY printer and was leaning towards using tmc2130 and would love to use sensor less homing.`

Yes, this problem its solved, and works perfect for corexy, only need use low sensibility like this:

X_HOMING_SENSITIVITY 3
Y_HOMING_SENSITIVITY 4

closed 2018-03-11 by thinkyhead

thinkyhead, 2018-04-26:
> I want to run in spreadcycle but then sensorless homing stops working

Maybe something @teemuatlut can answer…?

teemuatlut, 2018-04-26:
```cpp
Configuration_adv.h

   * X/Y/Z_HOMING_SENSITIVITY is used for tuning the trigger sensitivity.
   * Higher values make the system LESS sensitive.
   * Lower value make the system MORE sensitive.
```

However this isn't the first time I hear about this so I'll take a look as soon as I have the time.

dchoskingsage, 2018-04-26:
Yeah that is how i have it, what i initially thought was happening was that changing for stealthchop to spreadcycle causes the motors to stall at a different level of torque which makes sense to me that it would change the homing sensitivity, however no matter what i set it to it seems to not make a difference.
Also the other thing that i noticed that was weird (but could totally be my setup) is that the sensitivity setting for X_HOMING_SENSITIVITY seems to set the sensitivity for Y and Y_HOMING_SENSITIVITY sets it for X.

thinkyhead, 2018-04-27:
> Also the other thing that i noticed that was weird (but could totally be my setup) is that the sensitivity setting for `X_HOMING_SENSITIVITY` seems to set the sensitivity for Y and `Y_HOMING_SENSITIVITY` sets it for X.

Everything looks okay in Marlin. Check your SPI EN pin connections.

dchosking, 2018-04-27:
> Everything looks okay in Marlin. Check your SPI EN pin connections.

When you say SPI EN pins, is that the pin from the driver to the endstop or the pins that select the driver to write settings to?

thinkyhead, 2018-04-28:
> the pins that select the driver to write settings to?

Yes, this. The SPI ENable pins.