MarlinFirmware/Marlin #8741: Marlin 1.1.6 LCD Smart Controller No Menu

Solved-IT, 2017-12-10:
- Description: On my full graphics smart controller, the menu is not displayed. Whenever I turn the encoder it only adjusts the feedrate. Pressing the encoder does not change anything.
- Expected behaviour: Display the menu
- Actual behaviour: Only the feedrate is adjusted.

I've added my configuration in the ZIP file. Am I overlooking something ?

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1546071/Marlin.zip)

Tannoo, 2017-12-11:
In `Configuration.h`, comment this out: `#define DOGLCD      // Full graphics display`.
It's not needed.

Uncomment this line:
`#define PINS_DEBUGGING`

After uploading, issue M43 in a serial console.

Look for `BTN_ENC` in the M43 serial output:
```
PIN:  35   Port: C2        BTN_ENC                                Input  = 1
```

Now, while holding down the encoder button, send `M43` again. It should look like this:
```
PIN:  35   Port: C2        BTN_ENC                                Input  = 0
```

If it does not change, it's possible that your display doesn't have a click-type encoder, it's not working, or that pin on the controller board is not working.

bjarchi, 2017-12-11:
It's also possible that something is wrong with the LCD itself. I recently got one of the RRDFG clones off amazon (fastest shipping option), and it was built with both IDC cables incorrectly polarized and the encoder directions reversed. Reading through reviews of similar products on Amazon, it seems like these kinds of 'small' problems are rampant on the clone products. You might want to look for reports from other people using the same brand of LCD, to see if there are any little 'tricks' people have found to work around manufacturer errors :\

fiveangle, 2017-12-11:
> it was built with both IDC cables/connectors incorrectly polarized and with the encoder directions reversed

This I thought was common knowledge.  Regardless, it would only turn on backlight but wouldn't display feedrate if the IDCs were flipped (as they are on so many displays) so probably bad pin setup.

Solved-IT, 2017-12-11:
Thank you all for your answers. I made the changes and issued the M43 command and unfortunately the BTN_ENC doesn't change.

I already tested the functionality of the encoder using a mulitmeter, That works.

According to the schematic, EN1 pin 5 schould be connected to EXP1 pin 2 .
That didn't work...I soldered a wire from EXP1 pin 2 to EN1 pin 5 and it works as expected.

Thank you all so much for your help...I really thought I messed up something in the configuration.

closed 2017-12-11 by Solved-IT