MarlinFirmware/Marlin #8743: Unstable PID after autotune bugfix-2.0.x

makz27, 2017-12-10:
### Bug Report
- Description: Unstable PID even after multiple autopid
- Expected behaviour: Stability of the temperature
- Actual behaviour: The temperature start to oscillate before reaching temp & oscillate a lot even after multiple autotunes
- Steps to reproduce:
  - Turn ON
  - M303 E0 S200 C8
  - M301 Pxx Ixx Dxx
  - M500
  - Set temperature to 200°
  - Watch the waves

- Version : branch bugfix-2.0.x commit : 64d49a2

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1546109/Configuration.zip)  
(on this configuration i tried to set the PID_FUNCTIONAL_RANGE to 300 but same issue at the default value 10)

I switched from Smoothieware to Marlin for a MKS Sbase.

On smoothieware, the first PID was successful, a very stable temperature for days (hotend & bed).  
At first i used the data from the smoothie firmware but it was not stable so i tried multiple autotunes with/without fan ect ... everytimes i get a unstable temperature

Here the temperature oscillate between 180.5 and 184.8 for a target temperature at 180  
[image]

makz27, 2017-12-11:
[image]

FIY : Switched back to Smoothieware with original PID value to try if the hardware is good, it's perfect

bjarchi, 2017-12-11:
I've found that the PID autotune can be a little hit-or-miss - may be a limitation of 8-bit hardware which is the lineage of Marlin and still informs a lot of the code. One thing I've noticed is that it takes a while for the algo to converge, at least on my setup. Try increasing the cycle count - say 12 instead of 8 - and see if that helps at all. 

I haven't played with this yet but you might also try *reducing* PID_K1 a bit - I think that will smooth things a bit by using relatively more of the previously calculated derivative term each cycle.

makz27, 2017-12-12:
Hi,

I tried to do more cycle on the autotune, same issue.

I've set the K1 to 0.98, boom unstable pid, waves for a surfer.
For now i'm stuck with the K1 at 1 to get a stable temperature but it need 20 minutes to heat the last 2 degrees to the target temperature.

bjarchi, 2017-12-13:
> I've set the K1 to 0.98, boom unstable pid, waves for a surfer.
For now i'm stuck with the K1 at 1 to get a stable temperature but it need 20 minutes to heat the last 2 degrees to the target temperature.

I meant lower than the default value of 0.95. Again, I haven't played with this, but I suspect that going lower than default may smooth the ringing a bit. I'll make a study of it when I have a chance. 

Re: taking too long to hit the target, that sounds like your I term may be too low. The I term (integral) is the one that forces the PID algo to converge that final bit in the absence of external perturbations. You might try bumping up I a bit and see what happens.

makz27, 2017-12-13:
It's worst than before.

But my bed is just okay (with K1 back at 1) if go to my target temp as it should, just the hotend is stable (at 1 only, not stable at any value below 1) but always stay 1 or 2degree below the target and incrementing the I value make it worst.

fiveangle, 2017-12-13:
Similar issues on 1.1.x at #8081

teddybee, 2017-12-19:
I have similar issues with the 2.0.x and MKS Sbase, the hotend is warming very slow around the target temp. Also can't keep the temp during the print. Fore example on 240 temp the hotend falls to 233-235.
[image]

GarrethX, 2018-01-03:
I can confirm this problem as well at Arduino DUE... My PID settings from some older version with marlin were nearly perfect, on 2.0.x I received very unstable temperature (even had problems with heatup without serious manual PID tweaking).

makz27, 2018-01-15:
Fixed by #8996

PR #8996 (2018-01-15): [2.0.x] PID control fix and more LPC1768 ADC filtering
This PR introduces even more sample rejection and filtering for the LPC1768 ADCs (increased median filter from 3 to 23 samples) it also fixes the PID control for all non-avr platforms as the frequency was wrong in the PID_dT calculation. 

fix #8855, fix #8743
closed 2018-01-15 by makz27