MarlinFirmware/Marlin #8746: [2.0.x] Anet A8 SD still not working

Phr3d13, 2017-12-11:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

### Bug Report
- Description: SD Not Working
- Expected behaviour: SD card initialised/read and contents displayed
- Actual behaviour: Prolonged part cooling fan spin, no sd detected

- Steps to reproduce:
  - Flash 2.0.x bugfix
  - Insert SD
  - Power On

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Anet A8 Marlin 2.0 default configs.zip](https://github.com/MarlinFirmware/Marlin/files/1546461/Anet.A8.Marlin.2.0.default.configs.zip)

Phr3d13, 2017-12-27:
Yes.

Phr3d13, 2017-12-30:
Can confirm still happening with bugfix-2.0.x, and still works with bugfix-1.1.x

Phr3d13, 2018-01-06:
Anyone have any ideas as to where I should start looking to squash this bug?

thinkyhead, 2018-02-11:
A lot of changes to the 2.0.x work branch lately. Has this issue been resolved yet?

Phr3d13, 2018-02-11:
I can test after some sleep. But as of my last test, still no.

MikeLud, 2018-02-16:
Thinkhead,
 
I am using a 2.0.x that was downloaded today and the SD card still is not working

thinkyhead, 2018-02-24:
There are some clear differences in `Sd2Card.cpp` so that's where I'd expect the trouble to be. I'm not sure why it only affects Anet A8's, but by process of elimination — replacing new (2.0) bits of code with old (1.1) bits of code — it should be possible to narrow down the culprit.

thinkyhead, 2018-02-25:
See if changing the `SD_DETECT_INVERTED` setting makes any difference, just in case…

MikeLud, 2018-02-25:
Tried changing the SD_DETECT_INVERTED and the SD card still not working. 
Version 1.1.8 SD works fine.

thinkyhead, 2018-02-25:
I would test it myself, but all my SD card readers work fine with 2.0.x at this time. If you feel ambitious you can try tweaking the code in `Sd2Card.cpp` and see if anything shakes out.

thinkyhead, 2018-02-25:
Also, don't forget to play around with the various SD card-related settings like check-and-retry, etc.

benlye, 2018-02-25:
I dusted off my old Anet board to see if I could help with this problem.  I can confirm that I was able to reproduce it right away.

First stop was pins debugging, but on bugfix-1.0.x I get garbage output from M43, e.g.:
```
PIN:   4   Port: B4        TXD                  protected 
.                                                   protected 
.                          

          protected 
```

On bugfix-2.0.x it looks sane:
```
PIN:   4   Port: B4        AVR_SS_PIN                             protected
.              FAN_PIN                                protected
.              SS_PIN                                 protected
```

@Bob-the-Kuhn, any pointers?  Is this a known issue with M43 on bugfix-1.1.x?

Compiling with PlatformIO in VS Code, if that makes any difference.

benlye, 2018-02-26:
@thinkyhead pushed some commits for pin debuggung and that now 'works' (the output is no longer garbage, but the pin readings still look odd - that's a conversation for another issue).

Anyhoo, I figured out how to make the Anet SD card work.  

I have no idea what effect this has on other boards, but I suspect it may fix #9209 as well.

At line line [253 of Sd2Card.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/sd/Sd2Card.cpp#L253), insert this:
```
pinMode(chipSelectPin_, OUTPUT);
```

So that it looks like this:
```
//todo: should use chipSelectPin ?
  pinMode(chipSelectPin_, OUTPUT);
  spiBegin();

  // set SCK rate for initialization commands
  spiRate_ = SPI_SD_INIT_RATE;
  spiInit(spiRate_);
```

I suppose the `pinMode` call could go inside the `spiBegin` function, but that's beyond my pay grade.

benlye, 2018-02-26:
Just tested my change on a RAMPS board with RRDFG Smart Controller and it didn't seem to cause anything adverse - SD is detected, mounted, read as expected.

thinkyhead, 2018-02-26:
Duplicate of #8746

Let's copy your solution to that issue for the assembled gallery.

thinkyhead, 2018-02-26:
Thanks immensely @benlye — How did we miss that comment? Marlin 1.1.x clearly sets the pin to an output, and why it was dropped for 2.0.x is a mystery.

thinkyhead, 2018-02-26:
Looking at 4b16fa3272032a66200515750608f35dffbf1c09 I can see why @p3p left it out, since the SPI pins are otherwise pre-defined. On the other hand there should have been no harm in leaving the line setting the pin as an output. And now we know, it's better. Well I'm utterly relieved this is sorted out. Only a few small things remain before we can put out 1.1.9.

closed 2018-02-26 by thinkyhead

MikeLud, 2018-02-27:
Benlye & Thinkyhead,
 
Thanks for fixing the SD issue, just downloaded the current 2.0 version and the SD works.
 
Mike

RamonPanganiban, 2018-08-08:
I just upgraded to marlin 1.1.9 on anet a8 and lost my sd card. been browsing on solutions and found this thread. I looked at sd2card.cpp and can't find where spi is invoked as in " spiBegin();"
No errors on compile and upload in arduino though. I am missing out on something, so i cannot try benlye's solution above.