MarlinFirmware/Marlin #8749: static assertion failed: FRONT_PROBE_BED_POSITION can't be reached by the Z probe.

Gigahertz77, 2017-12-11:
Hello I have installed 3dtouch my parameters is

#define X_PROBE_OFFSET_FROM_EXTRUDER 14   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 53   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

// The size of the print bed
#define X_BED_SIZE 220
#define Y_BED_SIZE 220

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -27
#define Y_MIN_POS -9
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 240

 // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 190
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 190

But i can't flashing marlin because I have this error 
**static assertion failed: FRONT_PROBE_BED_POSITION can't be reached by the Z probe.**

I don't now how to calculate this value

Tannoo, 2017-12-12:
Set the `FRONT_PROBE_BED_POSITION` to what the probe can get to...not the nozzle.

Tannoo, 2017-12-12:
```
 *   In the following example the X and Y offsets are both positive:
 *   #define X_PROBE_OFFSET_FROM_EXTRUDER 10
 *   #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
 *
 *      +-- BACK ---+
 *      |           |
 *    L |    (+) P  | R <-- probe (20,20)
 *    E |           | I
 *    F | (-) N (+) | G <-- nozzle (10,10)
 *    T |           | H
 *      |    (-)    | T
 *      |           |
 *      O-- FRONT --+
```

houseofbugs, 2017-12-13:
This is not a Marlin bug, this is a user configuration error. 

Probe is at 53 and the position is 15. It can't reach that. Max is 220-53 so 167.

Gigahertz77, 2017-12-13:
Ok thank you

thinkyhead, 2017-12-14:
```cpp
#define Y_PROBE_OFFSET_FROM_EXTRUDER 53   // Y offset: -front +behind [the nozzle]
#define Y_MIN_POS -9
```
When your nozzle is at Y=-9 your probe will be at Y=44. So 44 is the smallest value you can set for `FRONT_PROBE_BED_POSITION`.

closed 2017-12-14 by thinkyhead