MarlinFirmware/Marlin #8751: MKS Sbase compile error with ENDSTOP_INTERRUPTS_FEATURE

teddybee, 2017-12-11:
When I compile with `ENDSTOP_INTERRUPTS_FEATURE` I get an error:
```
Error: Traceback (most recent call last):
  File "c:\users\akos\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main

    cli(None, None, None)
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\akos\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\akos\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 122, in cli
    raise exception.ReturnErrorCode(1)
ReturnErrorCode: 1
```

teddybee, 2017-12-11:
[http://users.ece.utexas.edu/~valvano/Volume1/E-Book/C12_Interrupts.htm](http://users.ece.utexas.edu/~valvano/Volume1/E-Book/C12_Interrupts.htm)

fiveangle, 2017-12-12:
The error is the answer:

> #error Unsupported Platform!

It means `ENDSTOPS_INTERRUPT_FEATURE `is not supported on MKS Sbase.

Bob-the-Kuhn, 2017-12-12:
The ENDSTOPS_INTERRUPT_FEATURE is not needed in Marlin 2.0.x.  Since the stepper ISR runs every millisecond then BLTouch detection is already guaranteed.

Maybe the message should be changed to something like "Warning - ENDSTOPS_INTERRUPT_FEATURE ignored (not needed)"

AnHardt, 2017-12-12:
I did not create the `ENDSTOPS_INTERRUPT_FEATURE` for Marlin to support the BL-touch.
I made it for saving the time for checking the endstops in every step during the stepper ISR.
What is now done for the BL-touch, now checking even in between of the steps, feels perverse to me.
The positive side effect of having no minimal speed (limited by timer overflow) anymore is not used up to now.

thinkyhead, 2017-12-13:
> What is now done for the BL-touch, now checking even in between of the steps, feels perverse to me.

Some felt that brute force was needed to make that fickle little hall effect probe meet the performance claimed in its promotional materials — within Marlin and not just on a test rig. I haven't checked out the additions lately. Maybe they are overkill.

thinkyhead, 2017-12-13:
Ah, I see…
```cpp
  #define ENDSTOP_NOMINAL_OCR_VAL 3000
  // Check endstops every 1.5ms to guarantee two stepper ISRs within 5ms for BLTouch
```
According to this the extra logic for the ISR to run earlier (to check endstops again) may not be needed without `BLTOUCH`. If that's the only reason the extra logic exists, and not for the benefit of endstops/probes generally, then it could be simplified for other setups.

teddybee, 2017-12-13:
Thank you for the clarification. This was enabled on my old Anet controller, I thought it would be useful if the controller doesn't waste some cpu time with the continuous loop.

closed 2017-12-17 by thinkyhead