MarlinFirmware/Marlin #8752: Stepper motor reversing

Bob-the-Kuhn, 2017-12-11:
The current algorithm appears to ignore direction changes when setting up the acceleration/deceleration profile.

In this example the jerk rate for the X axis would result in a 0.625mS spacing between steps when starting up from a standstill.  To the left of the direction change the step spacing is about 0.270mS while to the right it is 0.625mS.
[image]

My expectation in this example is:
1) The stepper motor will slow down when the first reverse direction pulse occurs.
2) The motor will come to a halt, maybe even start slowly in the reverse direction when the second  reverse direction step occurs.
3) The motor be up to the jerk speed between the third and fourth reverse direction steps.
4) After the fourth reverse direction step the stepper will be accelerating normally.

My conclusion is there'll be a loss of two steps.

----

Questions:

Is my conclusion reasonable?

Is this of enough concern that we spend CPU cycles to modify the algorithm?  This would affect primarily the planner speed.  

If we do modify the algorithm then we need to consider things like how slow do we want to decel before switching direction, do we drop all axis by the same ratio, do we treat X, Y Z & E the same, ...

Bob-the-Kuhn, 2017-12-11:
I just saw this on Wikipedia

> The pull-in curve defines an area called the start/stop region. Into this region, the motor can be started/stopped instantaneously with a load applied and without loss of synchronism.

To me that means we need to get the stepper down to the jerk step spacing before we can reliably stop the stepper without loosing steps.  

In this example it took 25 steps to accelerate to speed which means it'll take about that number to bring it to a standstill.

AnHardt, 2017-12-12:
Interesting questions!

Does a move end with a step or with a pause? I suppose pause.
Does a move begin with a step or with a pause? I suppose step.

When only one axis is involved (here X) and this changes direction and both moves should have the same max speed and neither acceleration nor jerk is changed. Then i would expect to see the pauses before and after the first step pulse (direction change) to have about the length of the junction speed (x-jerk), differing not more than one acceleration step can make it different. Maybe a bit more because using the new block needs a little time.

What i see in your picture (probably the time window is not long enough to judge it), i'd expect to see when the first move is faster than the one in reverse direction and both feed rates are below x-jerk.

victorpv, 2017-12-12:
But if you are right, shouldn't we be losing steps currently, some prints should have defects?
I have not noticed that with either an AVR running 1.1.x, or an STMF1 running 2.0.0

Bob-the-Kuhn, 2017-12-12:
See PR #8761 for some example code to address this.

Bob-the-Kuhn, 2017-12-12:
> shouldn't we be losing steps currently, so prints should have defects?

That's why I'm wondering if this is just a "nice thing to do" and could fall into the "if it works don't break it" category.

One possible explanation is that we're not running our printers close enough to their limits to notice.  As far as I know things like jerk and acceleration settings are usually determined by the "try it and see if it hurts method" or we just grab the default and stick with it unless we run into problems.

I'm hoping for some feedback from users that are pushing their printers.

victorpv, 2017-12-12:
A while back when using DRV8825 steppers with the 2.0 HAL in an STM board, before changing the min_stepper_pulse form the default 0uS, I was getting many missed steps, and the print was totally wavy in 1 axis (forgot if X or Y) while perfect in the other two. I guessed I was missing steps, was easy to figure from seeing the couple of prints I did on those conditions, but I bet I was missing loads of steps.

My guess is this issue may be causing the steppers to miss a very small number of steps, possibly only microsteps since most people use microsteps, and once enough microsteps accumulate in the opposite direction, the motor may just have a little jerk and jump back to the right position, so we dont miss a full step, but may be jerling more that we should, and missing microsteps in the changes of direction, that may recover as a single full step back.

AnHardt, 2017-12-12:
@victorpv 
A stepper can't lose microsteps. It's losing n*4 full steps. (n = ..., -2, -1, 0, 1, 2, ...)
A stepper can be displaced |+-| < one full step and will return to its original position.
Beyond of this, it will snap to the next stable position (+-4). If the load is not reduced then, it will go to the next stable position, until the load is reduced.

victorpv, 2017-12-12:
@AnHardt what I meant by missing microsteps is that it may not actually move, if the force in the opposite direction is strong enough. What I understand, and correct me if I'm wrong, is that the stepper may try to advance the motor a microstep, then another, then another, until eventually there is enough torque to move the motor.
So if the motor is rotating in one direction and we change direction as Bob was showing above, the motor may not change direction until the stepper has accumulated a few microsteps in the new direction and has enough torque, then the motor will move at once to that point.

closed 2018-05-20 by Bob-the-Kuhn