MarlinFirmware/Marlin #8754: [FR] Manual input of UBL adjustment (not a probe, actual measurement)

WheresWaldo, 2017-12-11:
### Feature Request

I searched and did not find this, nor could I find a way to do this using G-Code. I have one single point in my mesh that consistently probes too high. I want to adjust it, but do not have an LCD or encoder. Why is there no way to just use a g-code? 
Example, I know I need to move this one single point, fourth row, fourth column ( X ,Y (3, 3)) 0.04 mm down to have a nearly perfect validation grid. 

G29 P4 X3 Y3 Z-0.04

Or something like this. The problem with probing is that it probes the same every time and with no LCD/Encoder there is no way to input, edit or change this single or multiple points.

WheresWaldo, 2017-12-11:
Can this be done with M421?

WheresWaldo, 2017-12-11:
Works as expected.

closed 2017-12-11 by WheresWaldo

thinkyhead, 2017-12-16:
> Can this be done with M421?

http://marlinfw.org/docs/gcode/M421.html

```
M421 I<xindex> J<yindex> Z<value>
M421 I<xindex> J<yindex> Q<offset>
```