MarlinFirmware/Marlin #8755: [FR/RFC] Better handling of RGBW LEDs

bjarchi, 2017-12-11:
### Feature Request / Request for Comment
I installed controllable LEDs in my printer over the weekend - mainly so I had the option to turn the lights off if I ever want to print overnight - and I chose RGBW 'neopixel' strip-type LEDs since they're officially supported (there's a type defined for RGB+W neopixel/WS2812/SK6812 drivers).

Unfortunately the current state of things isn't great from the RGBW standpoint. The default define for WHITE doesn't use the W LED, even when present, so if one is using status LED functionality the 'white' while printing is just R+G+B. Enabling LED_CONTROL_MENU and LED_COLOR_PRESETS allows the user to define a default startup state that includes white, but this isn't used once status LED functionality takes over. 

I've also noticed some weirdness in the status LED functionality that seems to come from oscillations between 'heating' and 'heated' states - may need some hysteresis/debounce there. 

This is probably something I can handle in terms of writing the code and generating a PR, although participation from people who wrote or work with the existing LED code would be much appreciated.  It's probably not something I have time to work on right away, but I want to gauge interest. My main questions are:

* Would this be useful/appreciated, assuming it's done in a way that's transparent to the standard RGB case? 
* I assume that, as a new feature (ish), development would belong in bugfix-2.0.x rather than bugfix-1.1.x. Correct?
* Possibility of working in HSV for easier abstract representation of 'colors' with lower-level conversion to output LED type?

------------------------------------------------------------

Tannoo, 2017-12-20:
I see that RGBW's are fine. Neopixels needs a little work though.

thinkyhead, 2017-12-25:
Feel free to tweak the LED code as needed.

Tannoo, 2018-01-05:
>Would this be useful/appreciated, assuming it's done in a way that's transparent to the standard RGB case?

Any improvement is usually welcomed.

>I assume that, as a new feature (ish), development would belong in bugfix-2.0.x rather than bugfix-1.1.x. Correct?

Well, currently, both are needed.

>Possibility of working in HSV for easier abstract representation of 'colors' with lower-level conversion to output LED type?

My only question on that is, will it get too processor intensive for a show of lights?

Tannoo, 2018-01-05:
>Unfortunately the current state of things isn't great from the RGBW standpoint. The default define for WHITE doesn't use the W LED, even when present...

In that testing that I have done, all works just fine with RGB, RGBW, and Neopixel RGB and RGBW.