MarlinFirmware/Marlin #8756: [FR][RFC][2.0] "advanced" SPI support in the HAL

revilor, 2017-12-12:
I'm working on a hardware extension for my printer which requires SPI features beyond what is provided by the HAL at the moment. My printer is based on a MEGA 2560 board, therefore I could just use the Arduino SPI library instead of the HAL SPI functions. But considering there are devices like for example some of the Trinamic stepper drivers which can be configured via SPI, I guess it would make sense to have a more feature-rich SPI implementation in the HAL.

I wonder if there are plans for extending the HAL's SPI feature set or if maybe somebody is already working on that. I'm willing to put some effort into adding better SPI support to the HAL if that is considered a useful extension to Marlin. So please let me know what you think about it.

p3p, 2017-12-12:
An 'improved' HAL SPI interface was in the works just before my HAL work was merged and became the base of 2.0, but was no where near ready so I reverted it (https://github.com/MarlinFirmware/Marlin/commit/61c0a10efeae9400d517f376278e8f7cb4532a9b) in order to quickly fix builds, in theory all hardware features (SPI, I2C ect) need to be behind some kind of multi-channel interface and then the individual platform will either write an implementation or pass through to a library (so no core Marlin code ever accesses libraries directly) but I ran out off hobby time about 5 months ago and have yet to sit down and plan out how to do it with minimal/no impact on the AVR platform. The use of external device driver libraries also complicates matters as we have no control over how they access the hardware. It is something I planned to look into again .. eventually .. feel free to have a go if you like.

Bob-the-Kuhn, 2017-12-12:
It would be really nice if the manager had some provision for handshaking and/or locking a channel.  I expect that we're only a couple of years away from going to some type of RTOS.  

There aren't many of the current SPI users that are well enough behaved to make the transition as is.  

Some of the current LCD routines go through a init/setup phase, then a begin transmission and finally an end transmission phase.

I'm looking forward to the next generation of 32 bit boards.  The current generation is so pin limited that it's hard to do anything more than a basic system with them.  NXP has what looks like a LPX1768 compatible chip that has almost twice the I/O, has EEPROM and is priced less than the LPC1768.  DEATH TO THE RAMPS (R3) FOOTPRINT!!!

victorpv, 2017-12-12:
@Bob-the-Kuhn an API like I described could internally set a flag on entering the function, and clear the same flag on exit.
I have done it with FreeRTOS with mutexes in the STMF1 and worked well at sharing an SPI port between 2 threads, for LCD and SDcard.

thinkyhead, 2018-02-10:
Still needs work, I presume.

boelle, 2019-02-19:
@revilor still need this?

boelle, 2019-06-22:
wrong label

sjasonsmith, 2020-09-11:
@rhapsodyv this may be of interest to you.

rhapsodyv, 2020-09-11:
We already have a rich SPI implementation in STM32F1 and LPC. 

It's multi bus (one spi device can be used by different configurations) and support:

- Data Sizes: 16 and 8 bit 
- All 4 Modes
- byte order
- Speed

And we follow the Arduino SPIClass interface, so it's kind of standard too.

The client code just need to do the PIN select and call SPI.begin().

We need to port that class for other HAL, but in my view, this issue is done and could be close.

closed 2021-09-23 by thisiskeithb