MarlinFirmware/Marlin #8757: [FR] Belt Driven Extrusion Kinematics

Quintox303, 2017-12-12:
I recently had an idea of mounting the extruder motor on the x gantry of a Prusa style machine rather than on the frame (bowden) or x carriage (direct drive).  A looped belt would run across to the other side of the x gantry and go around an idler pulley, just like the x belt. This extruder belt could be meshed with a toothed pulley on the x carriage that is connected to the drive gear. So it would be a belt driven extrusion system. More information about my concept can be found here: https://mattshub.com/2017/09/30/belt-driven-extrusion/

The one different thing about this design is that the kinematics in firmware will need to change. If the extruder motor doesn’t move when the x motor moves, then this will result in unintentional extrusion as the x carriage is moving past the stationary belt and making the pulley turn. The movement commands sent to the extruder motor will actually have to be the same as the x motor, plus any extrusion distance.

So the x motor movement would remain exactly the same, but the movement of the extruder motor at any given time would be the regular extruder movement + x movement.

I've had a go at trying to make these changes in the firmware myself but it's too complicated for me. I'd be very grateful for anyone who could take the time to make this small change

bruce356, 2017-12-12:
You really do not want to know what I think

fiveangle, 2017-12-14:
I posted my response in the FB post of the same.  You also do not want to know what I think ;)

Quintox303, 2018-01-17:
Closing because I finally managed to do this myself :)

closed 2018-01-17 by Quintox303