MarlinFirmware/Marlin #8758: Marlin Firmware 1.1.5 Z-Axis will not home.

laynethompson17, 2017-12-12:
I have a homemade printer, that was actually a project for one of my engineering courses. I’m new to the 3-D printing community so please bear with me. It has Marlin Firmware and runs off of a Arduino MEGA. It has a proximity sensor that it uses when homing its Z-Axis. Every time
I home though the probe does not read the bed. It moves upward on the Z and reads out a probing failed command when I try to auto level my bed. Any help would be appreciated thanks

Tannoo, 2017-12-12:
Send `M119` to see the current state of the endstops.
```
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open
filament: TRIGGERED
```

Manually lower Z until the probe _should_ trigger and issue `M119` again.
Did `z_min` change?

Tannoo, 2017-12-12:
It's possible you may have to change `Z_MIN_ENDSTOP_INVERTING` in `Configuration.h`.

boelle, 2019-02-19:
@laynethompson17 

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle