MarlinFirmware/Marlin #8763: Problem with PROBE_BED_POSITION and Bltouch

campy, 2017-12-12:
Hi i try to configure 3DTouch sensor, but when i set offset from extruder and probe Bed position i have a error, in other printers i can configure, but always with X negative offset (probe of left side)

Wit this value the probe can go to every point, i dont know why Probe error

This is my config

Bltouch Enable
#define BLTOUCH

Offset from extruder
#define X_PROBE_OFFSET_FROM_EXTRUDER 37,8// X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -4,1  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1   // Z offset: -below +above  [the nozzle]

Bed Size

// The size of the print bed
#define X_BED_SIZE 210
#define Y_BED_SIZE 210

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

Linear autolevel enable
#define AUTO_BED_LEVELING_LINEAR

And offset probe: 
 // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 10
  #define RIGHT_PROBE_BED_POSITION 150
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 190

Error:
Arduino:1.8.2 (Windows 10), Tarjeta:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:39:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:742: error: static assertion failed: RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.

       static_assert(RIGHT_PROBE_BED_POSITION <= MAX_PROBE_X, "RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.");

       ^

SanityCheck.h:744: error: static assertion failed: BACK_PROBE_BED_POSITION can't be reached by the Z probe.

       static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION can't be reached by the Z probe.");

       ^

exit status 1
static assertion failed: RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.

bruce356, 2017-12-12:
increase the max bed size by the offset amount + a bit extra if needed so bltouch can reach areas of bed that you want.

example of mine, my bed size is actually x = 250, y = 320
#define X_MAX_POS 291  
#define Y_MAX_POS 335

AnHardt, 2017-12-12:
```
/**
 *   Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *   X and Y offsets must be integers.
 *
...
```

AnHardt, 2017-12-12:
> 
#define X_PROBE_OFFSET_FROM_EXTRUDER 37,8// X offset: -left +right [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -4,1 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1 // Z offset: -below +above [the nozzle]

Writing floats with ',' instead of '.' does not help either.

campy, 2017-12-12:
Solved with integers number @AnHardt  Thanks you!

closed 2017-12-12 by campy