MarlinFirmware/Marlin #8764: one time loop between reprap logo and info screen... Pls help

Shrikant888, 2017-12-12:
Hi sir, When I am switch ON machine. Its flow is showing REPRAP logo and then INFO SCREEN. But when click encoder its coming like its showing again REPRAP logo and then its going to PREPARE and all...

I wanted like when I click encoder in DEFAULT HOME SCREEN, its should go straightly go to PREPARE, CONTROL MENU... but now its coming different like when I click on DEFAULT HOME SCREEN , firstly its showing REPRAP logo for 1 second and then its going to PREPARE, CONTROL MENU..

What to do... pls help

Shrikant888, 2017-12-12:
The flow should be Power ON->REPRAP Logo->HOME SCREEN->Click ENCODER->INFO SCREEN (PREPARE, CONTROL, SD CARD)

but I am getting as Power ON->REPRAP Logo->HOME SCREEN->Click ENCODER->REPRAP Logo for 1 second->INFO SCREEN (PREPARE, CONTROL, SD CARD).. Pls note it sir

cjsoong, 2017-12-12:
You can try M502 & M500 instructions back to the default value

AnHardt, 2017-12-12:
If you really see a 'REPRAP Logo' and not a 'Marlin logo' you should use a newer Marlin. Yours is ancient.

Shrikant888, 2017-12-13:
Sir, REPRAP MARLIN logo is coming. its coming two times. As of now flow is like this

Power ON LCD boot
REPRAP MARLIN Logo
HOME SCREEN (temperature, Fan, X, Y, Z )
Clicking Rotary Encoder
REPRAP MARLIN Logo displays for 1 second
INFO SCREEN (Prepare, Control etc)

I wanted like this (skipping 5th step from above)

Power ON LCD boot
REPRAP MARLIN Logo
HOME SCREEN (temperature, Fan, X, Y, Z )
Clicking Rotary Encoder
INFO SCREEN (Prepare, Control etc)

Where to change in MARLIN FIRMWARE. Please tell

psavva, 2017-12-13:
Please provide your configuration files.

thinkyhead, 2017-12-16:
Are you using `bugfix-1.1.x`? We heard about this problem and fixed it a while ago.

closed 2018-03-07 by thinkyhead