MarlinFirmware/Marlin #8767: Filament Width Sensor enabled LCD incorrectly reports "V---" all the time

empakoso, 2017-12-12:
Bug Report
- Description: 
Filament Width Sensor is enabled but LCD displays "V---" throughout entire print.
This only began as of the last Filament Width Sensor update #8459 
Currently running Branch: bugfix-1.1.x downloaded 10Dec2017
Running since yesterday without any luck.

Could someone help to identify what I am doing incorrectly.  I would like to know why the volume output resulting from my sensor is not being calculated and displayed (for example "V102%").  Maybe I could call upon the experience of @Erlandsson since this issue has very recent activity.

I believe the multiplier is not being applied to the actual print since I am having another issue (blobbing output due to rogue readings).  This where the sensor sends large random reading and creates a blob on all my prints when I use it.  Knowing this I am sure volumentric multiplier is not being used by the extruder.

- Expected behaviour: ---
When the following startup gcode was used in the previous release it would show a "V" value of typically 102% when it displayed 1.71mm actually measured.

According to #8459 regarding the implementation of the Filament Width Sensor
@thinkyhead wrote: This PR fixes the LCD display code to show the simple ratio between the Nominal Filament Width (as set by M404 W[width]) and the measured filament width (as determined by a Filament Width Sensor, or as set directly by M200 D[width]). When volumetric mode is disabled it will display ---.

Previously noted in #8453 regarding the LCD "V" value displayed
@thinkyhead wrote : That makes sense, seen from that perspective. So we need the reciprocal, and the volume relationship. That is, "how far must the extruder move to push out 7.068 cu.mm when we know that 1mm will produce 6.157 cu.mm?" 1.14mm — which is, 114%.

- Actual behaviour: 
When the above startup gcode is used the LCD always displays "V---"

- What I am doing to reproduce the issue:
The following startup gcode is being used in my OctoPi setup:
M404 W1.75          ; set filament width sensor nom dia is 1.75mm
M405 D0			; filament width sensor (M405 D7.5 is ON-Delay 7.5cm) 
M407                ; displays measured diameter
M400                ; buffer wait and empty
M117 FWS enabled    ; LCD note

A more complete list of items is attached below, including the start of my gcode from Slic3r.
I have also attempted to use M405 D7.5 which is my actual distance from sensor to nozzle tip.

I have attached a copy of the config files and a quick pic of what my LCD shows all the time now.
[Filament Width Sensor Not Displaying Volume Multiplier.zip](https://github.com/MarlinFirmware/Marlin/files/1552710/Filament.Width.Sensor.Not.Displaying.Volume.Multiplier.zip)

Thank you in advance.

thinkyhead, 2017-12-13:
It currently only shows the filament width percentage when `M200 Dnnn` is used to turn on volumetric extrusion. This then requires the slicer to output G-code with E in volumetric units. A patch is pending which will enable proper E scaling in both volumetric and non-volumetric modes.

empakoso, 2017-12-14:
I should be able to test upon returning to my printer this weekend.  Thank you for the very quick response!

empakoso, 2017-12-17:
Just a quick verify of the new 1.1.7 code with just the filament width sensor enabled (no other modifications made to the download) there was this fault reported:

sketch\temperature.cpp: In static member function 'static void Temperature::manage_heater()':

temperature.cpp:802: error: 'calculate_volumetric_for_width_sensor' was not declared in this scope

       calculate_volumetric_for_width_sensor(measurement_delay[meas_shift_index])

                                                                                ^

exit status 1
'calculate_volumetric_for_width_sensor' was not declared in this scope

empakoso, 2017-12-18:
@thinkyhead Should this issue (noted above) become a separate thread?

I have now found time away work to begin a thorough test of your new upgrades to the Filament Width Sensor config option.  I will be using linear mode initially, and am looking forward to trying it out the new volumetric-mode feature!  Thanks.

empakoso, 2017-12-20:
Fixed. Works now with LCD reporting realtime updates from my filament width sensor in linear mode. 

Now testing with a known print which “blobbed” in the past. Will update shortly on the effectiveness in using our new MIN and MAX settings which should prevent blobs.

Then it’s off to check out the new feature of volumetric mode using this filament width sensor.

closed 2017-12-20 by empakoso

thinkyhead, 2017-12-20:
It took me a while to grasp what was going on with filament width and its relationship to volumetric extrusion, and that the filament width sensor code wasn't actually finished. Once that became clear, then it was just a matter of completing it!

As such, this is a relatively new feature. As it develops, I think it would be better to have the code assume that any reading more than —for example— 1mm over or under the nominal width is a bad one, rather than having `MEASURED_UPPER_LIMIT` and `MEASURED_LOWER_LIMIT` as hard settings.

With some tweaking, the filament width sensor could also double as a filament runout sensor.

empakoso, 2017-12-20:
I am all for this 1mm “blackout” feature.  I could use the run out option as well.  Keep me posted I will continue testing anything you guys can throw at this feature!

thinkyhead, 2017-12-20:
I changed the lower/upper settings with a single `FILWIDTH_ERROR_MARGIN` setting. I hope it works at least as well as the old bounds.

empakoso, 2017-12-20:
Perfect! I will give it a try today. Thanks again. This is really becoming an awesome thing to have.
Quick side note, the big Stratasys does not even have this feature.  Yeah I get it, it’s not density control by it’s the next best thing!

empakoso, 2017-12-21:
Haven’t loaded this version yet.  Will try very shortly once I get a big print off to double check issue which only displays “V100%” all the time.

empakoso, 2017-12-22:
Loaded this new method.  Really like that now we can change filament without having to reflash the fw.
Thanks.