MarlinFirmware/Marlin #8771: SPI / on board Sd Card problem

Spawn32, 2017-12-13:
@Bob-the-Kuhn 
You seem like a racer on this spi / sd card stuff, could you look at this 2 pictures and tell me whats going on ?
It starts out correct, 80 start clocks and then sends the correct command but gets a 1F insted of a 01, after that i have no ider what it's doing, it's not sending the correct commands after that, but do get respons ? :/

Tried to google the 1f answer, couldn't find anything..

This log is just from a cheap buspirate, wont get my real analyzer before a week or two..

[image]

[image]

Spawn32, 2017-12-13:
Solved:

echo:SD card ok
Begin file list
CUBE20~1.GCO 36058
End file list
ok

Note to my self, never assume...

I was amusing that "SPI.transfer" was actually using DMA, it's not, so had a extra look at @victorpv 's F1 hal i found that he was using "SPI.dmaSend"

also had to change the MCU apb divider, lowest spi clk i could get was 432khz, sd card needs 200-400khz at init, so now init is at 216khz...

Next step, crank up the "High" SPI clk to the sd cards fails 😄

closed 2017-12-13 by Spawn32

Bob-the-Kuhn, 2017-12-13:
I don't understand what is happening either.

It took a while for me to grab some SD card accesses on my mega2560 & then start to understand what is going on.

As best I can tell a valid CMD0 command is being sent but the card isn't responding with the "I'm ready" status.  The host should keep repeating the CMD0 command until the ready status is returned.

Took a while but I finally found some SD card SPI info I could understand.  
http://elm-chan.org/docs/mmc/mmc_e.html
http://nerdclub-uk.blogspot.com/2012/11/how-spi-works-with-sd-card.html

Just in case it's of use to you, I've attached an Excel spreadsheet with the SPI dumps from various cards and the logic analyzer captures that they came from.  You'll need the Saleae logic analyzer software to view the captures.

[SD_card.zip](https://github.com/MarlinFirmware/Marlin/files/1557430/SD_card.zip)

Bob-the-Kuhn, 2017-12-13:
Just saw your second post.  Glad to see you got it working.

Spawn32, 2017-12-14:
Yes it was nice to finally get it working, but i am going to study those links and files, a lot to learn about sd card communication, thanks @Bob-the-Kuhn  :)

Now i can move the eeprom emulator out of flash and to the sd card where it belongs so i dont wear out the flash...

victorpv, 2017-12-14:
@Spawn32 About the eeprom emulator to sdcard, I did something similar, not sure if I uploaded the files to my github. Basically I just implemented it in RAM, and then functions to save the whole ram block to a bin file, and to read it back. So rather than read and write individual values from the sdcard, on init it would read the whole content to ram. Most people's opinion was that having settings in an sdcard that may be removed when the board boots up was not a reliable mechanism. So in my case I just added an i2c eeprom to my board, and for other boards people can use the flash. If we added some wear leveling mechanism I don't think the flash is a bad solution.

Spawn32, 2017-12-15:
@victorpv sounds nice, will have a look and se if the copy to ram function is on your git, i am slowly working on a new revision of my pcb to fix up a few mistakes, think i will add space for a eeprom as well, nice to have just in case :) going to change the type og sd card connector i am using as well, was shocked, can get 20 pcs with a slightly different pcb layout from china for the same price digikey takes for 1, LOL