MarlinFirmware/Marlin #8774: GT2650_REV_A_PLUS - BLTouch wrong pins

OctavioMaia, 2017-12-13:
Bug Report
- Description: I currently own a GT2560 REV A+ and I couldn't get my BLTouch to work at all. It would never respond to M280 commands. The REV A+ has a proper servo pin header for the BLTouch alone, which seems to be maped to pin 32 on the config.
I managed to fix this by editing the SERVO0_PIN from 32 to 11 when the BLTouch is enabled.

- Expected behaviour: BLTouch should trigger
- Actual behaviour: It does not trigger/does not responde to M280
- Way to fix:

> #include "pins_GT2560_REV_A.h"
> 
> #undef BOARD_NAME
> #define BOARD_NAME  "GT2560 Rev.A+"
> 
> #if ENABLED(BLTOUCH)
>#define SERVO0_PIN  11
> #else
> #define SERVO0_PIN  11
> #endif

wittsan, 2017-12-14:
Hello,
I found the same problem. I'm watching a discussion on this topic https://github.com/MarlinFirmware/Marlin/issues/7981.

I allow myself to correct your code below because if the BlTouch is not activated, pin 32 must be activated to return to the normal configuration (in case you unplug the bltouch).

```
#include "pins_GT2560_REV_A.h"

#undef BOARD_NAME
#define BOARD_NAME "GT2560 Rev.A +"

#if ENABLED (BLTOUCH)
   #define SERVO0_PIN 11 // instead of 32
#else
   #define SERVO0_PIN 32 // instead of 11
#endif
```
W.

OctavioMaia, 2017-12-15:
Thanks for your update.
I'll do that on my config

wittsan, 2017-12-15:
With pleasure ;-)
It is good if we are several to confirm the error on the current version for a future patch ...
Have a good day.

thinkyhead, 2017-12-15:
It's almost as easy to make a patch as it is to make an issue, if anyone wants to try and get one in ahead of 1.1.7 release…

wittsan, 2017-12-15:
Hello Thinkyhead,
_(Sorry I'm not sure I understand the meaning)_ You say we should submit the patch directly, is that it?  how?
Thanks

landodragon141, 2017-12-16:
I'll do it later tonight, I have that same board.

thinkyhead, 2017-12-16:
> how?

That sounds like a good question for Github. https://help.github.com/articles/creating-a-pull-request/

wittsan, 2017-12-16:
I just made the submission. thx!

wittsan, 2017-12-16:
I do not know if I did well .. I have to make a "pull request" on the bugfix 1.1.x (and also bugfix 2.0.x) or is it good?
_I think it's a 1.1.7 submission I made._

boelle, 2019-02-19:
@thinkyhead we have an PR so could we not close this one?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead