MarlinFirmware/Marlin #8775: Printer keeps pausing

jrubinstein, 2017-12-13:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: printer (anet a8 with ramps running marlin 1.1.x) keeps pausing mid-print. It will pause several times during the print, keeping the bed and hotend hot. I've tried re-slicing, used two different slicers (cura and slic3r), and added a M100 command on level changes. It still pauses during the print.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

HekTron802, 2017-12-13:
Are you printing over USB or using the SD card? If you've been printing over the USB try the SD card. That will be a good way to narrow down what the possible issue is. If it works fine with the SD card (and not USB) then there is some sort of lag issue going on.

Another option is to grab the newest Bug_fix build, that might solve this. Also there are a couple of new features, such as Skew correction which I've found very useful. Linear advance which your version supports is also a feature I would highly recommend. http://marlinfw.org/docs/features/lin_advance.html

jrubinstein, 2017-12-13:
I've been using the SD card exclusively. I'll take a look at linear advance. The ANet group on facebook suggests increasing buffer. Does that sound reasonable?

HekTron802, 2017-12-13:
That does sound reasonable, the only time I've had an issue like the one you're describing is when I was playing around with the buffer size. However you currently have the default values:

```
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 16 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 16 // maximize block buffer
#endif
```

so my guess is that's not going to solve the issue, but I could be wrong and trying it won't hurt.

If it doesn't,  I would pull the newest bugfix and try that. Check out the Marlin 3d printer tool, it has a nice feature that helps out with migrating your config setup.  http://marlin3dprintertool.se/

Quick question: Have you ever been able print with this machine?

Good luck!

jrubinstein, 2017-12-13:
@HekTron802 I'll try fiddling with the buffer size.  I have been able to print with this machine without problems like this before. I've printed almost this same exact model before without problems.

thinkyhead, 2017-12-14:
>  running marlin 1.1.x

We can only fix bugs that exist in `bugfix-1.1.x` and `bugfix-2.0.x`, so please test the `bugfix-1.1.x` branch and see if the problem still exists.

jrubinstein, 2017-12-14:
I tried installing the latest bugfix, setup the new config file, but now my extruder is not working. The motor is fine, the driver seems fine, so it must be something in software. I'm tired of screwing with it, so I'm giving up for now.

HekTron802, 2017-12-14:
I’ve had that feeling many times! Sometime tomorrow let us know how the extruder isn’t working. I’m sure it’ll get sorted out.

thinkyhead, 2017-12-16:
> I'm tired of screwing with it, so I'm giving up for now.

Ok, well feel free to open a new issue if you have any other questions.

closed 2017-12-16 by thinkyhead