MarlinFirmware/Marlin #8776: Configuring 1.1.x for Ultimaker Original

CCS86, 2017-12-13:
Hi guys, sorry for misusing 'issues', but I'm not sure how else to reach you Marlin gurus.... Hopefully I can get this going and share the knowledge at the Ultimaker forums, so your work can reach more people.

I am trying to configure 1.1.6 to use on my Ultimaker Original and want to make sure I get everything set before trying to flash.

I've been going through the configuration.h (and _adv) files and comparing to my latest flashed version. Here's what I have noticed / changed so far:

**configuration.h:**

-Motherboard definition is now by name, not number): `#define MOTHERBOARD BOARD_ULTIMAKER`

-Power supply had to be turned on: `#define POWER_SUPPLY 1`

-Temp sensor needed to change for UM's TC: `#define TEMP_SENSOR_0 -1`

-Bed sensor enabled for my official heated bed: `#define TEMP_SENSOR_BED 20`

-I like to run this for bed control: 
```
#define PIDTEMPBED
#define  DEFAULT_bedKp 40    
#define  DEFAULT_bedKi .55    
#define  DEFAULT_bedKd 10
```

-Not sure here. Old version defined endstop inverting like this: 
`const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.`

-New version like this, but defaulted to false: 
`#define X_MIN_ENDSTOP_INVERTING true`

-I used to have to invert the stock values for X and Y axes (but not Z), because of my direct drives:
```
#define INVERT_X_DIR false   
#define INVERT_Y_DIR true  
```

-Now the new version has those values already, but had the wrong (false) value for Z. Changed to:
`#define INVERT_Z_DIR true`

-I enabled this:: `#define EEPROM_SETTINGS`

-I also did some modifications for my full graphic smart controller. It seems that "dogm_lcd_implementation.h" has become "ultralcd_impl_DOGM.h". Is that correct?

In my **configuration_adv**, I added these changes:

```
-#define BABYSTEPPING
-#define BABYSTEP_MULTIPLICATOR 2 (big change from the default "100")
-#define LIN_ADVANCE
-#define LIN_ADVANCE_K 0
-#define DIGIPOT_MOTOR_CURRENT { 135,135,135,135,135 } 
```

Am I missing anything? Anyone had success? 

Thanks!
Lars

kcheeeung, 2017-12-13:
What you are asking is a paradox. You have both the **machine** and **firmware**, but yet are asking us whether or not these settings will work.

Or you could just use whatever Ultimaker provides, and it's likely guaranteed to work for your Ultimaker Original machine...

CCS86, 2017-12-13:
I don't see how this is a paradox. 

I am trying to be diligent and help myself. I am going line by line to compare my existing 1.0 firmware to 1.1.6 (not all that easy, given how much restructuring has happened). I distilled my findings down to a handful of notable changes, and a few lingering questions.

I figure that there is a pretty good shot that someone on the Marlin github has some familiarity with the UMO hardware, along with familiarity of the later Marlin builds, and could help me answer these questions.

Ultimaker stopped development on this machine a longgg time ago. Why would I limit myself to firmware from that time period, just because somebody else stopped maintaining that fork of Marlin? 

I would think that the Marlin developers would like for their work to be used and appreciated by as many people as possible. That's the whole point, right? This isn't a for-profit venture.

Thank you very much for taking the time to write a dismissive and unhelpful reply. Cheers!

kcheeeung, 2017-12-13:
There is documentation for your questions: http://marlinfw.org/docs/configuration/configuration.html

The config files straight off github is for a different printer setup, and of course, Ultimaker Original has its own. So all you have to do, is **change those settings (just the true/false/#s) to be the SAME as your original config files.** It will work. Post the original config files if you don't know.

CCS86, 2017-12-13:
I understand the process of matching settings in the configuration files. I thought that would be clear from my original post. I listed the changes I made, and asked questions about items I wasn't clear on.

I even compared the 1.1.6 files to base 1.0 files from github as well, to take UM customization out of the equation. 

The link you provided doesn't answer the questions I have, but thank you.

psavva, 2017-12-14:
@CCS86, I suggest you upload the firmware and test.  If you need a rollback plan, reinstall Ultimaker's version if your not able to manage.

Overall, it looks ok, with some comments..
If you have an LCD, you probably need to enable it.
Sd card too..
You'll need to determine proper PID values using autotune and re-upload when you've got the optimal values from the autotune.

Check your printer dimensions and ensure that it's configured correctly.

Start off by checking and validating they all axes move correctly in the right direction.

Ensure that your endstops trigger correctly.

Give it a try and us know if you find to any specific problem we can help with.

CCS86, 2017-12-14:
@psavva, thanks for the reply!

I think I've got those bases covered, so I'll try to flash it. 

Any suggestions on checking the endstop triggering safely?

mroote, 2017-12-15:
@CCS86 You can use M119 to get the current status of your endstops.

```
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_max: open
Recv: z_min: open
Recv: ok
```

thinkyhead, 2017-12-15:
The default config should be pretty close to the original Ultimaker, as for a long time it was tuned specifically for the UM1.

CCS86, 2017-12-19:
It's looking like the changes I listed above did the trick. I've fired off a couple prints so far and everything is working well!

Thanks guys.

closed 2017-12-19 by CCS86

Lilithet, 2018-08-28:
You wouldn't happen to still have the config file for the setup lying around would you?