MarlinFirmware/Marlin #8777: Preheat presumably halts Marlin

skc5741, 2017-12-14:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---

I have a printer of my own design with a RUMBA control board running off of Marlin, and it's been working fine for the past month or so. For the past week or so whenever I try to set the extruder temperature it crashes, and I lose all communication. I've tried using Cura and Pronterface, whenever I try to start a print in either the extruder doesn't heat up, and the extruder temp freezes on the interface. I have to reset the printer to get the printer to jog again. When I do it in Pronterface, I also get this error:

[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1173, in startcb

I would think it would be a Pronterface error if I didn't have the same problem on Cura.

- Expected behaviour: ---

Starting the print with Cura should preheat the extruder and begin printing.
Starting the print with Pronterface should preheat the extruder and begin printing.
Preheating the extruder with Pronterface should preheat the extruder.

- Actual behaviour: ---

Starting the print with Cura freezes the extruder temp feedback and extruder doesn't heat, even after aborting print the print cannot be jogged or homed.
Starting the print with Pronterface freezes the extruder temp feedback and extruder doesn't heat, even after aborting print the print cannot be jogged or homed. Receive the error message:

[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1173, in startcb

Preheating the printer occasionally works, but the thermistor seems to be reporting a wide range of values up to 365 degrees, even when the extruder is not heated.

- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1557575/Configuration.zip)

revilor, 2017-12-14:
> Preheating the printer occasionally works, but the thermistor seems to be reporting a wide range of values up to 365 degrees, even when the extruder is not heated.

There might be the root of your problems: when Marlin does not get consistent temperature readings when heating the extruder the failsafe kicks in and blocks the printer. You should check if the thermistor has proper contact with the heater block and if the wiring is ok and properly connected to your printer board. It's also possible the thermistor is broken. If you own a multimeter you could measure the resistance of the thermistor. I suppose you use some kind of 100k NTC, so at room temperature you should measure a resistance in the region of about 98kOhm or so.

skc5741, 2017-12-16:
That sounds exactly like the problem. Thanks so much for your reply!

When I put my multimeter on the terminals of the thermistor, I get no resistance. This totally has to be the problem, but I can't put my finger on why that would be. How do thermistors go from working fine to offering no resistance at all?

I'll replace the thermistors, thanks so much for your help!

revilor, 2017-12-16:
Might be just a bad specimen. Using a thermistor which is not specified for the temperature you are printing at might be another possible reason.

closed 2017-12-17 by thinkyhead

skc5741, 2017-12-17:
I identified the problem while attempting to remove the thermistors.

As I loosened the securing screw around the thermistor wires, the auto fan abrubtly came to a stop. I realized the thermistor readings had returned to their true values. I had tightened the securing screw too much, and it was somehow interrupting the thermistor readings.

Thanks so much for your help!