MarlinFirmware/Marlin #8779: Sainsmart 2-in-1 board pulling up gen7-custom board pins definition?

shupack, 2017-12-14:
### Compile Error
When I compile with 
`
#ifndef MOTHERBOARD
  #define BOARD_SAINSMART_2IN1
#endif

` I get an error:
```
#error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."

from tab pins_GEN7_CUSTOM.h ```

If I comment out:

#if !defined(__AVR_ATmega644P__) && !defined(__AVR_ATmega644__) && !defined(__AVR_ATmega1284P__)
  #error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."
#endif

it will compile, but not sure if the pins will be set correctly.

revilor, 2017-12-14:
Just a stupid question: did you select 'Sanguino' as your board in the Arduino IDE as the error message suggests?

TGMods, 2017-12-14:
I use a Sainsmart 2-in-1 controller in my HyperCube without any problem. The error actually means that "Arduino/Genuino Mega 2560 or Mega 2560" has not been selected in the Arduino Tools --> Boards menu. It is not a Marlin issue.

shupack, 2017-12-15:
> Just a stupid question: did you select 'Sanguino' as your board in the Arduino IDE as the error message suggests?

not a stupid question, but no, I didnt' choose a Sanguino, as I don't even have it installed in my IDE.

 the 2-in-1 is a Mega 2560.

>The error actually means that "Arduino/Genuino Mega 2560 or Mega 2560" has not been selected in the Arduino Tools --> Boards menu. It is not a Marlin issue.

I do have Arduino/Genuino Mega 2560 or Mega 2560 selected, 2nd thing I checked, as I was working with an UNO previously on another project, and the error was "no pins for this board defined" (paraphrase). 

I'm currently running an older version of Marlin on my 2-in-1 with the board set as RAMPS EFB, trying to update.  

changing to: 

#ifndef MOTHERBOARD
  #define BOARD_RAMPS_13_EFB
#endif

Gives the same error:

  #error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."

Which prompted me to check in boards manager, was 1.6.19.  After updating I still get the same error.

revilor, 2017-12-15:
Which version of Marlin are you trying to build?

I tried a build of Marlin 1.1.6 using Arduino IDE 1.8.5 and Arduino IDE 1.6.9.

Using the default Configuration.h with 

#ifndef MOTHERBOARD
#define BOARD_SAINSMART_2IN1
#endif

and Mega 2560 as selected board compiles without problems with both revisions of Arduino IDE. 

Maybe reinstall Ardiuno IDE and try again?

fiveangle, 2017-12-15:
Yes, reinstall everything again and dump Arduino 1.6

However, there is a different error in 1.1.6 for your board:
```

In file included from .piolibdeps\TMC26XStepper\TMC26XStepper.cpp:33:0:
Marlin/SPI.h:33:5: error: 'FORCE_INLINE' does not name a type
FORCE_INLINE static void init() { softSPI.begin(); }
^
Marlin/SPI.h:34:5: error: 'FORCE_INLINE' does not name a type
FORCE_INLINE static void send(uint8_t data) { softSPI.send(data); }
^
Marlin/SPI.h:35:5: error: 'FORCE_INLINE' does not name a type
FORCE_INLINE static uint8_t receive() { return softSPI.receive(); }
^
Marlin/SPI.h:41:11: error: 'MISO_PIN' was not declared in this scope
class SPI<MISO_PIN, MOSI_PIN, SCK_PIN> {
^
Marlin/SPI.h:41:21: error: 'MOSI_PIN' was not declared in this scope
class SPI<MISO_PIN, MOSI_PIN, SCK_PIN> {
^
Marlin/SPI.h:41:31: error: 'SCK_PIN' was not declared in this scope
class SPI<MISO_PIN, MOSI_PIN, SCK_PIN> {
^
Marlin/SPI.h:41:38: error: template argument 1 is invalid
class SPI<MISO_PIN, MOSI_PIN, SCK_PIN> {
^
Marlin/SPI.h:41:38: error: template argument 2 is invalid
Marlin/SPI.h:41:38: error: template argument 3 is invalid
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp: In member function 'void TMC26XStepper::start()':
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:179:8: error: missing template arguments before '.' token
SPI.setBitOrder(MSBFIRST);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:180:5: error: missing template arguments before '.' token
SPI.setClockDivider(SPI_CLOCK_DIV8);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:180:22: error: 'SPI_CLOCK_DIV8' was not declared in this scope
SPI.setClockDivider(SPI_CLOCK_DIV8);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:183:5: error: missing template arguments before '.' token
SPI.begin();
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp: In member function 'void TMC26XStepper::send262(long unsigned int)':
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:956:37: error: 'SPI_MODE_MASK' was not declared in this scope
unsigned char oldMode =  SPCR & SPI_MODE_MASK;
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:959:20: error: 'SPI_MODE3' was not declared in this scope
if (oldMode != SPI_MODE3) {
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:960:12: error: missing template arguments before '.' token
SPI.setDataMode(SPI_MODE3);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:975:18: error: missing template arguments before '.' token
i_datagram = SPI.transfer((datagram >> 16) & 0xff);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:977:19: error: missing template arguments before '.' token
i_datagram |= SPI.transfer((datagram >>  8) & 0xff);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:979:19: error: missing template arguments before '.' token
i_datagram |= SPI.transfer((datagram) & 0xff);
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:992:20: error: 'SPI_MODE3' was not declared in this scope
if (oldMode != SPI_MODE3) {
^
.piolibdeps\TMC26XStepper\TMC26XStepper.cpp:993:12: error: missing template arguments before '.' token
SPI.setDataMode(oldMode);
^
*** [.pioenvs\megaatmega2560\lib\TMC26XStepper\TMC26XStepper.o] Error 1
 [ERROR] Took 101.41 seconds 
 [SUMMARY] 
Environment megaatmega2560      [ERROR]
Environment megaatmega1280      [SKIP]
Environment printrboard         [SKIP]
Environment printrboard_revf    [SKIP]
Environment brainwavepro        [SKIP]
Environment rambo               [SKIP]
Environment anet10              [SKIP]
Environment sanguino_atmega644p [SKIP]
 [ERROR] Took 101.41 seconds 
```

Which appears resolved in bugfix1.1.x, but there is yet another current regression:

```
Marlin\temperature.cpp: In static member function 'static void Temperature::updateTemperaturesFromRawValues()':
Marlin\temperature.cpp:966:29: error: 'analog2tempBed' is not a member of 'Temperature'
current_temperature_bed = Temperature::analog2tempBed(current_temperature_bed_raw);
^
*** [.pioenvs\megaatmega2560\src\temperature.o] Error 1
 [ERROR] Took 39.08 seconds 
 [SUMMARY] 
Environment megaatmega2560      [ERROR]
Environment megaatmega1280      [SKIP]
Environment printrboard         [SKIP]
Environment printrboard_revf    [SKIP]
Environment brainwavepro        [SKIP]
Environment rambo               [SKIP]
Environment anet10              [SKIP]
Environment sanguino_atmega644p [SKIP]
 [ERROR] Took 39.08 seconds 
```

Which if you have a heated bed (enable the heated bed thermistor) then you don't hit:

```
avr-objcopy -O ihex -R .eeprom .pioenvs\megaatmega2560\firmware.elf .pioenvs\megaatmega2560\firmware.hex
AVR Memory Usage
----------------
Device: atmega2560
Program:   56518 bytes (21.6% Full)
(.text + .data + .bootloader)
Data:       2667 bytes (32.6% Full)
(.data + .bss + .noinit)
 [SUCCESS] Took 116.23 seconds 
 [SUMMARY] 
Environment megaatmega2560      [SUCCESS]
Environment megaatmega1280      [SKIP]
Environment printrboard         [SKIP]
Environment printrboard_revf    [SKIP]
Environment brainwavepro        [SKIP]
Environment rambo               [SKIP]
Environment anet10              [SKIP]
Environment sanguino_atmega644p [SKIP]
 [SUCCESS] Took 116.23 seconds 
```

You should jettison your entire workchain/build environment and start from scratch with Arduino 1.9 beta, or PlatformIO and try again.

As it should build w/o a heated bed, after get your buildchain issues resolved and confirm the above info, please update the title of this issue accordingly as the `analog2tempBed ` issue still needs to be resolved.

Thanks,

-=dave

fiveangle, 2017-12-15:
Fixed at #8791

closed 2017-12-15 by thinkyhead

shupack, 2017-12-16:
upgraded to Arduino 1.9beta, compiled with no issues.  Have a print running now, will upload and test tomorrow. 

Thank you.

shupack, 2017-12-21:
done and sorted, thank you all.  I had some issues with USB drivers, had to go back to stable arduino to upload.