MarlinFirmware/Marlin #8781: [FR] TMC2130 Homing Sensorless For Z Probe

campy, 2017-12-14:
Hi, i test the new driver tmc2130 with homing sensorless and work perfect for X and Y axis, its possible implement new feature for use sensorless for Z probe?, i don´t know how about the precission of sensorless  homming repetitive values. 

But may be we can implement this features and no need put any Z probe like a BLtouch, inductive sensor etc...

For now in marlin only i can use for X and Y axis, if any user tell me what code need for this funtion i can test it

Thanks!

lonelymyp, 2017-12-14:
use any of the autocalibration sensor setup instructions. Instead of the sensor, connect the signal from the Z driver.

teemuatlut, 2017-12-14:
Sensorless homing requires a setup routine or at least be configured to work and it deliberately has not been done for the Z axis. The reason being that Z axis can have a very large torque multiplier with the screw ratios and when you ram you nozzle into the heat bed, the motors may not see much resisting force. I did not want people to start breaking their expensive E3D heatbreaks because of a feature that may not be suitable for Z axis homing.
If you really want to try it, at your own risk, you can use the TMC2130_ADV macro to configure your Z driver.

marcio-ao, 2017-12-14:
I agree with teemualut. I attempted to do sensorless homing on Z and it did not work very well because of the torque multiplier. I was able to get the motor to stall by drastically lowering the motor current, but it wasn't very precise at all, certainly not precise enough for Z probing or leveling.

lonelymyp, 2017-12-14:
Which screw do you use? 
A screw with a pitch of 8 mm does not create a large force.

megablue, 2017-12-15:
Depending on the type of Z axis, for example, doing this on belts driven Z axis,  like delta, might even cause the belt to deform by a lot before the stall detection kicks in.

garv3, 2018-01-21:
Well, for some printers sensorless homing on Z might be a nice feature. See the Anet A8 (with the correct modifications). When homing Z while X is at min, nothing will crash into the bed since the carriage is outside of the bed dimensions. By printing two identical physical tubes that you put on your Z rods you could additionally make sure, both sides are at the same height. After homing you would level using a probe anyways. I would love to see this feature in marlin.
The Prusa i3 mk3 does the calibration at the top of the Z axis. A modified Anet A8 for example could do this at the bottom while homing. This would be awesome!

teemuatlut, 2018-01-21:
You can use M915 for the Z calibration. It's supposed to work roughly the same way as in Prusa printers. But the feature is very much in beta! Use with caution!
Basically what it will do is
* Reduce current
* Move the axis to `Z_MAX_POS` + `CALIBRATION_EXTRA_HEIGHT`
* Restore normal current level
* Re-home Z axis because we intentionally lost steps when ramming the axis to its physical limit

Bazeone, 2018-01-25:
to use the M915 feature does the data pin need to be connected to the Max Z endstop?

teemuatlut, 2018-01-25:
No because the firmware will just try to push the axis higher than it physically can. But it'll first (greatly) reduce the current to prevent parts from breaking. There are no endstops or other sensors used.
I'd still recommend caution when you use it the first time. I know it worked for me and I hope it will work for others as well.

tcm0116, 2018-02-13:
>I attempted to do sensorless probing on Z and it did not work very well because of the torque multiplier. I was able to get the motor to stall by drastically lowering the motor current, but it wasn't very precise at all, certainly not precise enough for Z probing or leveling.

@marcio-ao I was having a similar issue when trying to get this to work on my delta machine. I tried to lower the current, but that didn't really work. What I learned is that the stallGuard works better when the motor is moving at higher velocities. Increasing the probing speed to 100 mm/s actually produced consistent and usable results. It seems counter-intuitive, but at the lower speeds, the steppers were skipping, especially at the perimeters. At the higher speed, it bounces the nozzle nicely off the bed without applying too much force to the bed. Seeing as my delta runs at 100 steps/mm, perhaps you could use that as a reference for tuning the homing/probing speed on your cartesian machine to see if that makes a difference.

marcio-ao, 2018-02-13:
@tcm0116 : Interesting! At the moment, we have decided to continue with Z endstops on our printers, but this is good to know!

Nenzyz, 2019-01-16:
While experimenting with Z sensorless homing on my MPCNC, noticed strange behaviour: on G28 everything works fine, but when using G29 looks like signal from driver is ignored?

tcm0116, 2019-01-16:
@Nenzyz - you'd have to enable sensorless probing to use G29.

boelle, 2019-02-19:
i might be wrong but i think i remember that someone got this to work

but i'm not sure and cant remember how

BenderskiyMA, 2019-05-20:
Hello!
I use SKR 1.3 board and TMC 2130 SPI on my delta printer.
i want to use simultaneously sensorless_homing and sensorless_leveling
how can i configure firmware and hardware for this?

boelle, 2019-06-22:
missing label

prasarnh, 2019-08-12:
i think this can be done use separate motor such as nema 8 and use through hole shaft to make independent  z probe utilized stallguard feature.

sjasonsmith, 2020-09-11:
I assume this is complete, since `SENSORLESS_PROBING` is supported (although not necessarily recommended) in our configuration files.

closed 2020-09-11 by sjasonsmith