MarlinFirmware/Marlin #8782: Bed leveling ENABLE_LEVELING_FADE_HEIGHT problem

lonelymyp, 2017-12-14:
Marlin 1.1.3

I enabled option #define AUTO_BED_LEVELING_BILINEAR
enabled option #define ENABLE_LEVELING_FADE_HEIGHT
Set the compensation height M420 Z4
got 2 problems: 
1. the height of the model is 2 mm higher than it should be
2. The first 5.5 mm models are printed with badges. the height of the layer does not correspond to extrusion.
[image]

if I turned off the option #define ENABLE_LEVELING_FADE_HEIGHT, everything is fine.

HekTron802, 2017-12-14:
Pull down the bug fix branch, that will most likely fix this issue. Marlin is currently at 1.1.6 and 1.1.7 is close to release.

lonelymyp, 2017-12-15:
I tried version 1.1.6
The problem became the opposite, is slight overextrusion.

HekTron802, 2017-12-15:
If the over extrusion is happening only on the first couple layers, try adjusting the Z endstop or the probe offset if you're using a probe. Otherwise you can adjust the steps per mm or the flowrate. 
[marlinfw.org](http://marlinfw.org/) contains allot of useful information on gcode and setting marlin up.

thinkyhead, 2017-12-15:
Please test `bugfix-1.1.x`. Version 1.1.6 is quite old now.

lonelymyp, 2017-12-16:
bugfix corrected the problem. but now another problem, before the retract the printer hangs for a second.
https://youtu.be/WiMiXhZ2tW0

lonelymyp, 2017-12-17:
checked again, turned off the calibration option, hanging before the retract is in the same place
 :(

the bug appeared only in the bugfix firmware.

lonelymyp, 2017-12-17:
I just tried 1.1.7, the problem disappeared.

closed 2017-12-17 by lonelymyp