MarlinFirmware/Marlin #8785: MIN temp err with marlin 1.1.6 when using MAX6675 with SDcard on ramps 1.4

AlphaFlight75, 2017-12-14:
------------------------------------------------------------
### Bug Report
I'm using Marlin v1.1.6 with RAMPS 1.4 with full graphic Smart Controller. I've changed from a thermistor to a thermocouple K-type using a MAX6675. I've changed the sensor type to -2 in marlin and after uploading to the arduino I get a MIN_TEMP error. I've been reading a lot of comments about this problem and have tried most of the fixes but I'm still getting the error with SD card support enabled. If I turn off the SD support in Marlin my printer works fine. I've tried using another MAX6675 board and even a MAX31855 (with -3 setting) and I still getting the same error.

 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// example_configurations/SCARA and customize for your machine.
//

// @section info

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define SHOW_BOOTSCREEN
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
#define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2

//
// *** VENDORS PLEASE READ *****************************************************
//
// Marlin now allow you to have a vendor boot image to be displayed on machine
// start. When SHOW_CUSTOM_BOOTSCREEN is defined Marlin will first show your
// custom boot image and then the default Marlin boot image is shown.
//
// We suggest for you to take advantage of this new feature and keep the Marlin
[... 1585 lines cut ...]
 *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
 *
 * Note: May require analog pins to be defined for other boards.
 */
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00   // (mm) Diameter of the filament generally used (3.0 or 1.75mm), also used in the slicer. Used to validate sensor reading.

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor (0,1,2,3)
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define MEASURED_UPPER_LIMIT         3.30 // (mm) Upper limit used to validate sensor reading
  #define MEASURED_LOWER_LIMIT         1.90 // (mm) Lower limit used to validate sensor reading
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

psavva, 2017-12-15:
May I suggest that you upload a zip file with the config.  It's really difficult to read the pasted config from the issue itself.

AlphaFlight75, 2017-12-15:
Thanks for the suggestion, here it is.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1563019/Config.zip)

thinkyhead, 2017-12-15:
Please test `bugfix-1.1.x` in case we already fixed the issue.

AlphaFlight75, 2017-12-18:
Hi, I've just checked the latest bugfix and it seems to have cured my problem. I just need to put the printer back together so I can do a test print.

Many thanks

closed 2017-12-25 by thinkyhead