MarlinFirmware/Marlin #8786: [FR] Start Print At Layer

joeedh, 2017-12-15:
It'd be nice to be able to pick a layer to start printing at when printing files off an SD card.  I'm thinking of coding this myself, so I thought I'd run it by people for comment.  Thanks.

Roxy-3D, 2017-12-15:
I think the big problem with doing this is all the non-printing stuff.   For sure, you still need to heat the nozzles and the bed.   Does the algorithm just skip (and subtract off) the first few layers?

What about bed leveling?  Does the printer still home and probe the bed?  

I think it is all the side issues that makes this difficult.  Especially if we are trying to put a generic solution into place that handles most people's needs...

For me...  being able to start a print halfway in so I can glue it to the previously failed print would have a lot of value.   So, just as a suggestion...  It would be good if the 'solution' can at least handle that one case.

psavva, 2017-12-15:
Would this not be better achieved with a gcode on your Slicer?

AnHardt, 2017-12-15:
Marlin does not know about layers.
How do you want to detect a 'layer change'?
How do you want to detect 'spiral vase' mode? What should happen then?
How do you want to distinguish from a 'change position' with lifted nozzle?

fiveangle, 2017-12-15:
> I'm thinking of coding this myself

Go for it ! 😃

thinkyhead, 2017-12-15:
I have been contemplating this feature. A new G-code could be added that sets a minimum Z height, below which all movement commands will be ignored. Up until that height it would only run non-move commands like setting the fan speed. Once it reached the "start height" it would subtract the "start height" from the current height — give or take 0.2-0.3mm for the first layer (data which you would have to supply) — as it printed the part.

This G-code could be used in conjunction with a feature that remembers the last Z height printed, and in case of a failure would allow printing the rest (or any portion) of the model starting on a clean bed.

Issues you will encounter:
- There will be no proper first layer, no brim or raft, and the initial layer may be only infill and a perimeter, not solid. So you will need to use a good adhesive to ensure good results.
- Commands before the "start height" may cause weird things to occur without having the movements included.

boelle, 2019-02-19:
i agree this could be usefull so please @joeedh submit a PR for it :-D