MarlinFirmware/Marlin #8787: [2.0.X] Graphic LCD does not wrk correctly with last bugfix (368c124)

Bergerac56, 2017-12-15:
When compiled with option REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, the LCD does not work properly. It displays partially correct informations, then becomes full of garbage and finish "white".

Marlin is still controlable through host.

The UBL problem (Z jumps in place of smooth movements) is still present.

Tannoo, 2017-12-15:
Likely a bad LCD or cables.

Bergerac56, 2017-12-15:
Hehehe, I do not think so. When I take a previous version (2 days ago) or replace the RE-ARM by an avr, there is no problem anymore...

Tannoo, 2017-12-15:
Well.. hmmm. I am running yesterday's bugfix-2.0.x with the rrfg display ans ramps. There was nothing for me to update this morning either. No issues for me.

Tannoo, 2017-12-15:
Ah... so this is a Re-ARM issue.

Bergerac56, 2017-12-15:
Yes, it could be. Perhaps something changed in the way (timing?) the LCD is addressed.

Roxy-3D, 2017-12-15:
> The UBL problem (Z jumps in place of smooth movements) is still present.

Is this Z jumping with the nozzle in the center of the bed?  Or does it happen with the nozzle 'off the mesh' ?

Bergerac56, 2017-12-15:
Hello Roxy, I just replied on the other post:  #8684

Penagwin, 2017-12-17:
I'm having this issue too, also on a re-arm board. It kinda sorta works sometimes for a few seconds then goes to a mess. The cable seems to have no effect. Nore does restarting, resetting, power cycling, etc.

Nutz95, 2017-12-17:
Hello, i can confirm this issue. I've got some garbage on the screen.
I changed my cables and so on but still with the same issue. I tried 2 other Re-arm boards but still the same.
Then i reverted to a previous Marlin version (c555a214d2cb5704bb323f99b15f0132300336b4) from December 11th, and the display is back to normal.
So somewhere along the way there was a change that broke the behavior of the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

Bob-the-Kuhn, 2017-12-17:
Short term fix is to comment out lines 172 & 174 in the file **ultralcd_impl_DOGM.h**

Someone changed the routine used for the 32 bit boards.  I need to find out why it was changed before I do a permanent fix.

Bergerac56, 2017-12-18:
@Bob-the-Kuhn Thanks Bob, the short fix works

thinkyhead, 2017-12-20:
The change comes from #8651 — merged on faith. Check with @ejtagle to find out about the rationale for the changes.

ejtagle, 2017-12-20:
@Bob-the-Kuhn : I did the changes. I have a RAMPS 1.4 modified to work with an Arduino Due. I also have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER directly connected to the RAMPS (no level conversion is neccesary, i checked that very carefully, as all control and data lines are unidirectional and signals are output from the Due side, and only inputs on the graphics LCD.
The only lines that goes backwards are the SDCARD lines, but those lines are 3.3v, so no problems at all.

Without this fix, the display shows garbage. And the problem seems to be that Software implementation of the SPI clocks data too fast. So i forward ported the AVR LCD sw spi version to the ARM platform.
After this patch was done, i also authored an improvement to the SW SPI implementation of the Due that allows to control the SPI speed (both are merged)

What i do suspect, and please correct me, is that, even if the RE-ARM board has no level translator for the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, you are using an interrmediate level translator that adds some delay, or maybe the level translator is too fast, or is too slow. (Cables should not be the problem... i am using 50cm shielded ribbon cables and no problems at all at this SPI clock speed)

What you describe is obviously caused by a timing issue. You can "fine-tune" it by adjusting ST7920_SWSPI_SND_8BIT function on the file u8g_dev_st7920_128_64_sw_spi.cpp . I have used a 1mhz SPI clock (that is the purpose of the DELAY_NS(500) ... Delay in nanoseconds , but perhaps your setup needs to run even slower... Or faster... Experiment a bit. My setup will also work if clock is slower, so maybe we can get an intermediate speed that works reliably for both of us.

AnHardt, 2017-12-20:
I really wonder why all these wannabe hardware programmers wonder about the hardware they try to program.
The [ST7920 datasheet](https://www.crystalfontz.com/controllers/Sitronix/ST7920/323/) at page 43-44 clearly states: "Timing differs with VDD!".
Maybe it makes sense to set the SPI-clock, and the pauses, dependent on the used voltage.

Bob-the-Kuhn, 2017-12-20:
I'm in the process of writing a DUE specific com driver for the ST7920.  That'll allow any DUE specific timing to be handled in DUE specific code.

ejtagle, 2017-12-20:
@AnHardt I have carefully read all the ST7920 datasheet. According to that datasheet, if using 5v as power supply to the LCD (99% of the cases), then 1Mhz SPI clock is perfectly acceptable. Even 1.25Mhz should work with no problems (800ns minimum clock period @ 4.5v Vdd, the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is always powered with 5 volts)
But in practice, it does not work reliably. Probable causes are that we are using ST7920 chinese clones, but most likely, the problem is related to the long cabling used. 
Using ribbon cables with no proper shielding, using ribbon cables with no proper impedance termination, degrades signals. It even depends on the speed of the drivers used as port drivers on the SAM3x ... or whatever microcontroller you are using. And obviously depends on the length of the cables themselves.
I take firmware development seriously. I am myself an electronic engineer and part of my professional work is dealing with this kind of problems. When hardware configurations are variable, the most general advice is to slow down communications. But here that goes against responsiveness of the movement planner, so a delicate compromise between safe operation of the LCD (communication slowdown) and speed (communication speedup) is required... Unfortunately... I think that increasing the delay to 600ns should help, but i don't have a RE-ARM board to test, that is why i asked @Bergerac56  to try different speeds and report here.
I absolutely agree that having 4!!! different implementations of the interface for the ST7290, all of them using SPI is just a horrible thing. I hate this kind of code duplication. 
And the problem and the one to blame is ug8lib, that instead of using the system spiWrite functions, uses its own... 
I have considered to donate a new implementation of a graphics library (i have written several implementations, most are faster, more flexible and take way less FLASH space), but supporting all kinds of strange configurations is always a problem...
@Bob-the-Kuhn : Feel free to use any part of the code i have already contributed...

Bergerac56, 2017-12-20:
@ejtagle - Hello. I see you are expecting something from me. As I certainly missed it, could you (re) explain me what you are expecting? So I can test it for you

ejtagle, 2017-12-20:
@Bergerac56  Just try to use the untouched ultralcd_impl_DOGM.h and modify the delay inside the function ST7920_SWSPI_SND_8BIT that is in the file u8g_dev_st7920_128_64_sw_spi.cpp
There you will find 2 calls to the function DELAY_NS(500).
This call to DELAY_NS(500) what it does is just to delay the execution 500 nanoseconds for each phase of the SPI clock that is used to transfer data to the graphic LCD.
By increasing that number, SPI clock speed will slow down. I do suspect that increasing to 600 ... maybe 700 nanoseconds will also solve your problems (and will give us an idea on how much to slow down SPI to make communications between ARM cpus and the ST7290 display controllers reliable

Bergerac56, 2017-12-20:
@ejtagle - You do suspect well ;)

500ns does not work (we know that)
580ns becomes erratic after playing a bit with the menus and the encoder
600ns is the minimum to have a stable behavior
650ns is fine 
700ns obviously also OK

ejtagle, 2017-12-20:
Well, then i will change it to 700ns. It is better to have an stable display than a fast but not working sometimes one.
I am right now testing code that reflects the SD card as a Mass storage device--- Seems to be working pretty well :) ...I'd guess this will also benefit RE-ARM boards (and also managed to improve SD card performance quite a bit!)

closed 2017-12-22 by Bergerac56