MarlinFirmware/Marlin #8789: [2.0.X] Compile Error

pbrgle1, 2017-12-15:
I'm getting the following compile error. I attached my config files below.

Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\module\temperature.cpp:35:0:

sketch\src\module\../libs/private_spi.h: In static member function 'static uint8_t SPIclass<50u, 51u, 52u>::receive()':

sketch\src\module\../libs/private_spi.h:50:7: error: 'SPDR' was not declared in this scope

       SPDR = 0;

       ^

In file included from sketch\src\module\thermistor/../../inc/MarlinConfigPre.h:27:0,

                 from sketch\src\module\thermistor/../../inc/MarlinConfig.h:26,

                 from sketch\src\module\thermistor/thermistors.h:26,

                 from sketch\src\module\temperature.h:30,

                 from sketch\src\module\temperature.cpp:27:

sketch\src\module\../libs/private_spi.h:51:19: error: 'SPSR' was not declared in this scope

       for (;!TEST(SPSR, SPIF););

                   ^

sketch\src\module\thermistor/../../inc/../core/macros.h:101:22: note: in definition of macro 'TEST'

 #define TEST(n,b) (((n)&_BV(b))!=0)

                      ^

sketch\src\module\../libs/private_spi.h:51:25: error: 'SPIF' was not declared in this scope

       for (;!TEST(SPSR, SPIF););

                         ^

sketch\src\module\thermistor/../../inc/../core/macros.h:99:24: note: in definition of macro '_BV'

   #define _BV(n)  (1<<(n))

                        ^

sketch\src\module\../libs/private_spi.h:51:14: note: in expansion of macro 'TEST'

       for (;!TEST(SPSR, SPIF););

              ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1562789/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1562790/Configuration.txt)

thinkyhead, 2017-12-15:
It appears you are using a MAX6675 Thermocouple with a DUE board. I'm not sure if that's been made to work yet.

pbrgle1, 2017-12-16:
It's actually the MAX31855.

Now i have a new issue. I changed the sensort to Semitec (sensor 5) and the second extruder reads 488. (actually 12°C). When i unplug it reads 499. I think it's just reversed because of the 12 difference. I tried swapping the first end second sensor but it's always T1 that reads 488...

thinkyhead, 2018-02-11:
Has the 32-bit brigade made any progress here?

DarioX7, 2018-08-12:
Will this problem be fixed? I have a new board for Arduino DUE with 2 pieces of MAX6675 and I have the same problem with the compilation.

xarwell, 2018-08-16:
I also have the same problem.

DvDriel, 2018-10-25:
Same here, any progress?

thinkyhead, 2018-10-26:
I picked up a clue:
"`SPDR` on the Uno is `REG_SPI0_RDR` on the Due and `SPSR` on the Uno is like `REG_SPI0_SR` on the Due."
…So we just need to experiment with those by doing this in the Due HAL:
```cpp
#define SPDR REG_SPI0_RDR
#define SPSR REG_SPI0_SR
```
…and figuring out where to connect things up.

DarioX7, 2019-02-05:
> 
> 
> I picked up a clue from http://forum.arduino.cc/index.php?topic=267937.0
> 
> > "`SPDR` on the Uno is `REG_SPI0_RDR` on the Due and `SPSR` on the Uno is like `REG_SPI0_SR` on the Due."
> 
> …So we just need to experiment with those by doing this in the Due HAL:
> 
> ```c++
> #define SPDR REG_SPI0_RDR
> #define SPSR REG_SPI0_SR
> ```
> 
> …and figuring out where to connect things up. Not sure what `SPIF` is yet.

Hey.
So, whether it is known under which register is assigned `SPIF` ?

thinkyhead, 2019-02-06:
See if the issue is resolved by the above-linked PR.

DarioX7, 2019-02-06:
> 
> 
> See if the issue is resolved by the above-linked PR.

Thank you for fast help.
I downloaded the new version of Marlin 2.0 (bugfix) with your patches where I imported to PlatformIO IDE for Atom.
In the Configuration.h file for the quick test, I set only the board definitions (based on DUE3DOM).
https://pastebin.com/TGWxF22A
And I set the MAX6675 type for hotend E0.
The compilation was successful, the firmware uploaded correctly, but unfortunately Arduino DUE all the time resetting itself over and over again. The display showed only the start logo.
https://pastebin.com/MD6r2802
I will try to somehow debug it, why it happens.

DarioX7, 2019-02-24:
> 
> 
> > See if the issue is resolved by the above-linked PR.
> 
> Thank you for fast help.
> I downloaded the new version of Marlin 2.0 (bugfix) with your patches where I imported to PlatformIO IDE for Atom.
> In the Configuration.h file for the quick test, I set only the board definitions (based on DUE3DOM).
> https://pastebin.com/TGWxF22A
> And I set the MAX6675 type for hotend E0.
> The compilation was successful, the firmware uploaded correctly, but unfortunately Arduino DUE all the time resetting itself over and over again. The display showed only the start logo.
> https://pastebin.com/MD6r2802
> I will try to somehow debug it, why it happens.
> 
> > start
> > echo: Software Reset
> > Marlin busmatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded
> > mory: 82140  PlannerBufferBytes: 1600
> > echo:EEPROM version mismatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded
> > start
> > echo:PowerUp
> > Marlin bugfix-2.0.x
> > echo: Last Updated: 2018-01-20 | Author: @brandstaetter, @grbd
> > echo:Compiled: Feb  7 2019
> > echo: Free Memory: 82140  PlannerBufferBytes: 1600
> > echo:EEPROM version mismatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded
> > start
> > echo: Watchdog Reset
> > Marlin bugfix-2.0.x
> > echo: Last Updated: 2018-01-20 | Author: @brandstaetter, @grbd
> > echo:Compiled: Feb  7 2019
> > echo: Free Memory: 82140  PlannerBufferBytes: 1600
> > echo:EEPROM version mismatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded
> > start
> > echo: Watchdog Reset
> > Marlin bugfix-2.0.x
> > echo: Last Updated: 2018-01-20 | Author: @brandstaetter, @grbd
> > echo:Compiled: Feb  7 2019
> > echo: Free Memory: 82140  PlannerBufferBytes: 1600
> > echo:EEPROM version mismatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded
> > start
> > echo: Watchdog Reset
> > Marlin bugfix-2.0.x
> > echo: Last Updated: 2018-01-20 | Author: @brandstaetter, @grbd
> > echo:Compiled: Feb  7 2019
> > echo: Free Memory: 82140  PlannerBufferBytes: 1600
> > echo:EEPROM version mismatch (EEPROM=? Marlin=V64)
> > echo:Hardcoded Default Settings Loaded

I give up.
I do not have that knowledge to find the reason for reboots.
Anyone help?

thinkyhead, 2019-03-03:
For additional help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-07-06 by shitcreek