MarlinFirmware/Marlin #8795: [2.0.x] Kossel Pro (BRAINWAIVE_PRO) example compile failure due to Z_MIN_PIN define issue

fiveangle, 2017-12-15:
Found while trying to fix Travis builds:

```
$ use_example_configs delta/kossel_pro
The command "use_example_configs delta/kossel_pro" exited with 0.
3.95s$ build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}
In file included from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_AVR/HAL_spi_AVR.cpp:40:
Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_AVR/MarlinSerial.cpp:36:
Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_AVR/persistent_store_impl.cpp:5:
Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_AVR/servo_AVR.cpp:56:
Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_AVR/watchdog_AVR.cpp:25:
Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
*** [.pioenvs/teensy20/src/src/HAL/HAL_AVR/HAL_spi_AVR.o] Error 1
*** [.pioenvs/teensy20/src/src/HAL/HAL_AVR/MarlinSerial.o] Error 1
*** [.pioenvs/teensy20/src/src/HAL/HAL_AVR/persistent_store_impl.o] Error 1
*** [.pioenvs/teensy20/src/src/HAL/HAL_AVR/servo_AVR.o] Error 1
*** [.pioenvs/teensy20/src/src/HAL/HAL_AVR/watchdog_AVR.o] Error 1
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:34:0,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:691:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
^
*** [.pioenvs/teensy20/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o] Error 1
========================== [ERROR] Took 3.74 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[ERROR]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 3.74 seconds ==========================
```

From readme originally created by @Wackerbarth :

> # Example Configuration for OpenBeam Kossel Pro [BRAINWAVE_PRO](http://www.openbeamusa.com/3d-printers/kossel/)
> * Configuration files for the **Openbeam Kossel Pro** as delivered in their KickStarter distribution
> 
> I [@Wackerbarth](https://github.com/Wackerbarth) tested this version on my Kossel Pro and Arduino 1.6.5 for Mac.
> This configuration is a transition to merge Terence Tam's configuration with up-to-date Marlin source and a current Arduino IDE

-=dave

Wackerbarth, 2017-12-15:
Dave,

This configuration has been passed down from years ago. At that time, all of the processor/pins stuff was quite different. I upgraded the processor mine a long time ago and haven’t used Marlin on any delta printer in some time.

Perhaps the configuration should just “disappear” unless there are others interested in keeping it active. (The same can be said for other systems also. At some time, we need to call “END OF LIFE” on the dinosaurs)

fiveangle, 2017-12-15:
I'd tend to agree and depending on outcome of some of these old config compile failures, suggest we keep them but move these unmaintained configs into an "Archived" directory within configs with a readme in it that simply states they are no longer maintained.

Wackerbarth, 2017-12-15:
On Dec 15, 2017, at 3:18 PM, Dave Johnson <notifications@github.com> wrote:
> I'd tend to agree and depending on outcome of some of these old config compile failures, suggest we keep them but move these unmaintained configs into an "Archived" directory within configs with a readme in it that simply states they are no longer maintained.

I’m not sure that “Archiving” them in a position within the active tree is necessary. The GIT repository has them throughout history and they can be recovered from a release tag which was more contemporary with their actual usage.

I tried floating the idea of periodically placing a “soon to be removed unless someone reverifies this configuration” sunset on every configuration and, just before each release, purging those which have not been “claimed”. But that idea never gained traction.

pavelking, 2017-12-28:
I'm a user of Kossel Pro and just run into this problem.
Looks like on older Marlin build I used the value was not explicitly set and used a value specified somewere else. I will try to comment again if I find a solution (pin number and possibly other configuration issues). There are probably some other Kossel Pro users who would like to have this configuration working.

thinkyhead, 2018-02-11:
If #9586 does the trick, then we can apply the same change to 1.1.x.

fiveangle, 2018-05-22:
Thanks for confirming Bob !

closed 2018-05-22 by fiveangle

antidamage, 2019-01-01:
Is this configuration still maintained? I see some suggestions above to abandon it. I'm running into this issue and it's not clear from this thread what to do.

pavelking, 2019-01-02:
I am a Kossel Pro printer owner (it uses this board in orginal
configuration) and personally stopped using this board since it is a bit
slow for a delta and it was supported only up to Arduino 1.0.6
Was trying to do some tricks to use Marlin for latter versions of Arduino
and finally bricked it.
I suppose if it is kept in old versions of Marlin that can compile on
Arduino 1.0.6 it should be enough.

Bob-the-Kuhn, 2019-01-05:
Here's the modified  pins_BRAINWAVE_PRO.h file.

[pins_BRAINWAVE_PRO.zip](https://github.com/MarlinFirmware/Marlin/files/2729461/pins_BRAINWAVE_PRO.zip)

PR #9586 (2018-02-11): [2.0.x] Have Kossel Pro use Z_MIN_PROBE_ENDSTOP
Fix #8795

Since the Brainwave Pro only has 3 endstop plugs, the Kossel Pro Z min probe should use `Z_MIN_PROBE_PIN`.