MarlinFirmware/Marlin #8796: [2.0.x] STM32F10 example compile failure due to various issues

fiveangle, 2017-12-15:
Found while trying to fix Travis builds:

```
$ use_example_configs STM32F10
The command "use_example_configs STM32F10" exited with 0.
13.14s$ build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}
PlatformManager: Installing ststm32
Downloading...
Unpacking...
ststm32 @ 3.7.0 has been successfully installed!
The platform 'ststm32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinoststm32 @ ~2.0.0
Downloading...
Unpacking...
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0
Downloading...
Unpacking...
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:29:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../../Configuration.h:289:2: warning: #warning temp_sensor set to 998/999 (fake) [-Wcpp]
#warning temp_sensor set to 998/999 (fake)
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:29:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../../Configuration.h:447:2: warning: #warning "disabled temperature protection" [-Wcpp]
#warning "disabled temperature protection"
^
In file included from /home/travis/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/boards.h:39:0,
from /home/travis/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish.h:54,
from /home/travis/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/Arduino.h:30,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:43,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
/home/travis/.platformio/packages/framework-arduinoststm32/STM32F1/variants/generic_stm32f103r/board/board.h:41:0: warning: "CYCLES_PER_MICROSECOND" redefined [enabled by default]
#define CYCLES_PER_MICROSECOND  72
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:27:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../core/macros.h:48:0: note: this is the location of the previous definition
#define CYCLES_PER_MICROSECOND (F_CPU / 1000000L) // 16 or 20 on AVR
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:61:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:49:25: warning: extra tokens at end of #ifdef directive [enabled by default]
#ifdef MCU_STM32F103CB  || defined(MCU_STM32F103C8)
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:61:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_set_count(uint8_t, hal_timer_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:120:5: error: 'StepperTimer' was not declared in this scope
StepperTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:126:5: error: 'TempTimer' was not declared in this scope
TempTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'hal_timer_t HAL_timer_get_count(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:140:12: error: 'StepperTimer' was not declared in this scope
temp = StepperTimer.getCompare(STEP_TIMER_CHAN);
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:143:12: error: 'TempTimer' was not declared in this scope
temp = TempTimer.getCompare(TEMP_TIMER_CHAN);
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_set_current_count(uint8_t, hal_timer_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:154:26: error: 'StepperTimer' was not declared in this scope
case STEP_TIMER_NUM: StepperTimer.setCount(count); break;
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:155:26: error: 'TempTimer' was not declared in this scope
case TEMP_TIMER_NUM: TempTimer.setCount(count); break;
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'hal_timer_t HAL_timer_get_current_count(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:163:12: error: 'StepperTimer' was not declared in this scope
temp = StepperTimer.getCount();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:166:12: error: 'TempTimer' was not declared in this scope
temp = TempTimer.getCount();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_isr_prologue(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:180:5: error: 'StepperTimer' was not declared in this scope
StepperTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:186:5: error: 'TempTimer' was not declared in this scope
TempTimer.pause();
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h: At global scope:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:157:12: warning: 'int freeMemory()' defined but not used [-Wunused-function]
static int freeMemory() {
^
*** [.pioenvs/STM32F1/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o] Error 1
========================== [ERROR] Took 12.96 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[ERROR]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 12.96 seconds ==========================
```

-=dave

victorpv, 2017-12-16:
Please check https://github.com/MarlinFirmware/Marlin/issues/8585#issuecomment-351600792
I posted a link to a newer timer file, you need to replace it. I can send a PR for it, but I want to send it only once someone else other than me confirm everything works for him, otherwise I'm keeping thinkyhead busy with partial fixes.

fiveangle, 2017-12-16:
I don't have this board.

victorpv, 2017-12-16:
Can you check if with that file Travis builds? I  don't know much about Travis yet.

fiveangle, 2017-12-16:
Travis just uses PlatformIO so if it builds in PIO, it will build in Travis.  I just copied this file to a clean bf2 branch and it didn't resolve:

<details><summary>PIO Compile Log…</summary>

```
[ Deviot 2.2.0-dev.12 ] Starting...
[Sat Dec 16 09:57:03 2017] Processing STM32F1 (build_flags: !python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py; lib_deps: U8glib-HAL, LiquidCrystal_I2C@1.1.2, TMC2130Stepper, https://github.com/teemuatlut/TMC2208Stepper.git, Adafruit NeoPixel, https://github.com/lincomatic/LiquidTWI2.git, https://github.com/trinamic/TMC26XStepper.git, https://github.com/ameyer/Arduino-L6470.git; platform: ststm32; board: genericSTM32F103RE; framework: arduino; src_filter: +<src/*> -<src/config>)
--------------------------------------------------------------------------------
Library Storage: /Users/speedster/dev/github/Marlin/.piolibdeps
LibraryManager: Installing id=1932
U8glib-HAL @ 0.4 is already installed
LibraryManager: Installing id=576 @ 1.1.2
LiquidCrystal_I2C @ 1.1.2 is already installed
Looking for TMC2130Stepper library in registry
Found: http://platformio.org/lib/show/1493/TMC2130Stepper
LibraryManager: Installing id=1493
TMC2130Stepper @ 2.0.2 is already installed
LibraryManager: Installing TMC2208Stepper
TMC2208Stepper @ af2b60a is already installed
Looking for Adafruit NeoPixel library in registry
Found: http://platformio.org/lib/show/28/Adafruit NeoPixel
LibraryManager: Installing id=28
Adafruit NeoPixel @ 1.1.3 is already installed
LibraryManager: Installing LiquidTWI2
LiquidTWI2 @ 30aa480 is already installed
LibraryManager: Installing TMC26XStepper
TMC26XStepper @ c1921b4 is already installed
LibraryManager: Installing Arduino-L6470
Arduino-L6470 @ 3cd0993 is already installed
PlatformManager: Installing ststm32
Downloading...
Unpacking...
ststm32 @ 3.7.0 has been successfully installed!
The platform 'ststm32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinoststm32 @ ~2.0.0
Downloading...
Unpacking...
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0
Downloading...
Unpacking...
arm-none-eabi-g++ -o "/Users/speedster/dev/github/Marlin/Marlin/Marlin.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/tmpFq86XE"
Collected 39 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <U8glib-HAL> v0.4 (/Users/speedster/dev/github/Marlin/.piolibdeps/U8glib-HAL_ID1932)
|-- <LiquidCrystal_I2C> (/Users/speedster/dev/github/Marlin/.piolibdeps/LiquidCrystal_I2C_ID576)
|   |-- <Wire> (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/Wire)
|-- <TMC2130Stepper> v2.0.0 (/Users/speedster/dev/github/Marlin/.piolibdeps/TMC2130Stepper)
|   |-- <SPI> v1.0 (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/SPI)
|-- <TMC2208Stepper> v0.0.2 (/Users/speedster/dev/github/Marlin/.piolibdeps/TMC2208Stepper)
|-- <Adafruit NeoPixel> v1.1.2 (/Users/speedster/dev/github/Marlin/.piolibdeps/Adafruit NeoPixel)
|-- <LiquidTWI2> (/Users/speedster/dev/github/Marlin/.piolibdeps/LiquidTWI2)
|   |-- <Wire> (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/Wire)
|-- <TMC26XStepper> (/Users/speedster/dev/github/Marlin/.piolibdeps/TMC26XStepper)
|   |-- <SPI> v1.0 (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/SPI)
|-- <Arduino-L6470> (/Users/speedster/dev/github/Marlin/.piolibdeps/Arduino-L6470)
|-- <SPI> v1.0 (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/SPI)
|-- <Wire> (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/Wire)
|-- <STM32ADC> v1.0 (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/STM32ADC)
|-- <EEPROM> (/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/EEPROM)
arm-none-eabi-g++ -o .pioenvs/STM32F1/src/src/HAL/HAL_AVR/HAL_AVR.o -c -fno-rtti -fno-exceptions -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP --param max-inline-insns-single=500 -march=armv7-m -DF_CPU=72000000L -DGENERIC_STM32F103R -DPLATFORMIO=30500 -DARDUINO_GENERIC_STM32F103R -DMCU_STM32F103RE -DTARGET_STM32F1 -DARDUINO=10610 -DBOARD_generic_stm32f103r -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -DDEBUG_LEVEL=DEBUG_NONE -D__STM32F1__ -DARDUINO_ARCH_STM32F1 -DVECT_TAB_ADDR=134217728 -IMarlin/src/HAL -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/system/libmaple -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/system/libmaple/include -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/system/libmaple/usb/stm32f1 -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/system/libmaple/usb/usb_lib -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/variants/generic_stm32f103r -IMarlin -Iinclude -I.piolibdeps/U8glib-HAL_ID1932/src -I/Users/speedster/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/Wire -I.piolibdeps/LiquidCrystal_I2C_ID576 -I [... cut ...]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[ERROR]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 28.20 seconds ==========================
```

</details>

fiveangle, 2017-12-16:
To get up to speed on PIO quickly,

Windows: Install Python2.7, git commandline for windows
Mac: Install Xcode with commandline tools
-Install Sublime Text 3
-Install "Package Control" from within ST3
-Package Control: Install Package: Select "Deviot..."
-Enable developer version using these instructions (heed advice to restart ST3 between each step): https://github.com/gepd/Deviot/wiki/Setup#developer
-Select Deviot:Options:Use Development PIO
-Select Deviot:Options:Full Verbose Output
-Select Deviot:Select Environment (then select your STM32F1 target)
-Select Deviot:Compile
-🍻 

If anything is unclear above, google for it.

-=dave

Spawn32, 2017-12-16:
@victorpv really wish i could test it for you , but the timers files for my F7 is totally different.. :/

victorpv, 2017-12-16:
Thanks guys. I don't understand why Dave still gets the errors about TempTimer.getcount etc, since those are removed in that file, but I'll try to get PIO setup and test.

On a different note, I just enabled Laser support today and engraving a few things for a present, works really well.

victorpv, 2017-12-17:
I just spent quite a few hours with platformio, and is not going anywhere. Throws the most stupid errors I could image, and neither Arduino or Eclipse throw. First, it can't find the SPI library because there is an SPI.h file in the HAL. The stm32 SPI library doesn't show in the list of libraries, I added in manually to the list of deps in platformio.ini, and still would not find the right file. Ended up renaming /HAL/SPI.h to HAL/HAL_SPI.h. Then it has started throwing errors about the Hardware timer class, USBSerial class...
```
C:\Users\Victor\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/usb_serial.h:41:1: error: unknown type name 'class'
class USBSerial : public Stream {
^
C:\Users\Victor\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/usb_serial.h:41:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
class USBSerial : public Stream {
```
The errors don't make any sense to me, at the moment unless someone can point to a solution for this, consider the STMF1 HAL doesn't compile in platformio.

victorpv, 2017-12-17:
After several more hours, I got platformio to compile the F1 HAL. It fails if I enable i2c eeprom or M100 support, which both work fine in with Eclipse. They compile and they do work.

victorpv, 2017-12-17:
@thinkyhead I'm trying my best to get platformio to compile all the options that work fine for me with Eclipse, but I can't find a way to get it to load the STM32F1 SPI library files correctly since instead it confuses it with this file:
`Marlin/src/HAL/SPI.h`

I have tried all I could think, including partial paths and what not, but so far the only thing that has worked for me is renaming that one file to HAL_SPI.h. 
I can only find it included from HAL.h.
Can we rename Marlin/src/HAL/SPI.h to Marlin/src/HAL/HAL_SPI.h?
If you are ok, I will send a PR with that change.

With that, and some changes to platformio.ini, the F1 HAL compiles fine for a bunch of options, and I am trying to sort out the others.

fiveangle, 2017-12-19:
> Can we rename Marlin/src/HAL/SPI.h to Marlin/src/HAL/HAL_SPI.h?

If this resolves and it's not a band-aid for a problem somewhere else in the Marlin code, I'm sure it would be fine.  Fastest way to find out is to submit a PR with the change :)

-=dave

p3p, 2017-12-19:
@victorpv I had some issues with platformios LDF (not the best idea in the world) getting confused when setting up the LPC1768, I just [disabled it](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini#L102) as I prefer to specify needed libraries anyway, may help in this case aswell.

thinkyhead, 2018-02-11:
The above PR should help reduce confusion about the timer functions.

victorpv, 2018-02-15:
Thanks @p3p I'll give it a shot. I don't know much about plaformio, so any pointers like that help.
Just searching for what ldf is, it makes sense that may be related to the problem with SPI. In eclipse I manually select what libraries should be included. I do not know yet how to manually tell plaformio, but at least now I know how to disable the automatic mode.

thinkyhead, 2018-02-23:
LDF = Library Dependency Finder. They seem very proud of it!

boelle, 2019-02-19:
@fiveangle does latest bugfix 2.0 solve the issue?

fiveangle, 2019-02-20:
Anyone could test and answer this, but since no good deed goes unpunished... 😁 

```
avr-objcopy -O ihex -R .eeprom .pioenvs/melzi_optiboot/firmware.elf .pioenvs/melzi_optiboot/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [========= ]  91.6% (used 15003 bytes from 16384 bytes)
PROGRAM: [========= ]  93.5% (used 121540 bytes from 130048 bytes)
.pioenvs/melzi_optiboot/firmware.elf  :
section                      size      addr
.data                         310   8388864
.text                      121230         0
.bss                        14693   8389174
.comment                       17         0
.note.gnu.avr.deviceinfo       64         0
.debug_aranges                512         0
.debug_info                  4690         0
.debug_abbrev                2153         0
.debug_line                  2632         0
.debug_str                    662         0
Total                      146963
========================= [SUCCESS] Took 47.20 seconds =========================
```

-=dave

closed 2019-02-20 by fiveangle