MarlinFirmware/Marlin #8798: [2.0.x] UltiMachine/Archim2 example compile failure due to various PIN issues

fiveangle, 2017-12-15:
Found while trying to fix Travis builds:

```
The command "use_example_configs UltiMachine/Archim2" exited with 0.
19.01s$ build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}
PlatformManager: Installing atmelsam
Downloading...
Unpacking...
atmelsam @ 2.7.1 has been successfully installed!
The platform 'atmelsam' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinosam @ ~2.10616.1
Downloading...
Unpacking...
Marlin/src/HAL/HAL_DUE/HAL_spi_Due.cpp:232:15: warning: 'void spiSend(const uint8_t*, size_t)' defined but not used [-Wunused-function]
static void spiSend(const uint8_t* buf, size_t n) {
^
In file included from Marlin/src/inc/../HAL/HAL_DUE/HAL_Due.h:36:0,
from Marlin/src/inc/../HAL/HAL.h:40,
from Marlin/src/inc/MarlinConfig.h:28,
from Marlin/src/Marlin.h:25,
from Marlin/src/Marlin.cpp:31:
Marlin/src/Marlin.cpp: In function 'void enable_all_steppers()':
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:71:25: error: 'DIO105_WPORT' was not declared in this scope
volatile Pio* port = (DIO ##  IO ## _WPORT); \
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:103:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^
Marlin/src/module/stepper_indirection.h:280:36: note: in expansion of macro 'WRITE'
#define E0_ENABLE_WRITE(STATE) WRITE(E0_ENABLE_PIN,STATE)
^
Marlin/src/Marlin.h:105:26: note: in expansion of macro 'E0_ENABLE_WRITE'
#define  enable_E0() E0_ENABLE_WRITE( E_ENABLE_ON)
^
Marlin/src/Marlin.cpp:296:3: note: in expansion of macro 'enable_E0'
enable_E0();
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:72:24: error: 'DIO105_PIN' was not declared in this scope
uint32_t mask = MASK(DIO ## IO ## _PIN); \
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:47:28: note: in definition of macro 'MASK'
#define MASK(PIN)  (1 << PIN)
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:103:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^
Marlin/src/module/stepper_indirection.h:280:36: note: in expansion of macro 'WRITE'
#define E0_ENABLE_WRITE(STATE) WRITE(E0_ENABLE_PIN,STATE)
^
Marlin/src/Marlin.h:105:26: note: in expansion of macro 'E0_ENABLE_WRITE'
#define  enable_E0() E0_ENABLE_WRITE( E_ENABLE_ON)
^
Marlin/src/Marlin.cpp:296:3: note: in expansion of macro 'enable_E0'
enable_E0();
^
Marlin/src/Marlin.cpp: In function 'void disable_e_steppers()':
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:71:25: error: 'DIO105_WPORT' was not declared in this scope
volatile Pio* port = (DIO ##  IO ## _WPORT); \
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:103:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^
Marlin/src/module/stepper_indirection.h:280:36: note: in expansion of macro 'WRITE'
#define E0_ENABLE_WRITE(STATE) WRITE(E0_ENABLE_PIN,STATE)
^
Marlin/src/Marlin.h:106:26: note: in expansion of macro 'E0_ENABLE_WRITE'
#define disable_E0() E0_ENABLE_WRITE(!E_ENABLE_ON)
^
Marlin/src/Marlin.cpp:304:3: note: in expansion of macro 'disable_E0'
disable_E0();
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:72:24: error: 'DIO105_PIN' was not declared in this scope
uint32_t mask = MASK(DIO ## IO ## _PIN); \
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:47:28: note: in definition of macro 'MASK'
#define MASK(PIN)  (1 << PIN)
^
Marlin/src/inc/../HAL/HAL_DUE/fastio_Due.h:103:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^
Marlin/src/module/stepper_indirection.h:280:36: note: in expansion of macro 'WRITE'
#define E0_ENABLE_WRITE(STATE) WRITE(E0_ENABLE_PIN,STATE)
^
Marlin/src/Marlin.h:106:26: note: in expansion of macro 'E0_ENABLE_WRITE'
#define disable_E0() E0_ENABLE_WRITE(!E_ENABLE_ON)
^
Marlin/src/Marlin.cpp:304:3: note: in expansion of macro 'disable_E0'
disable_E0();
^
*** [.pioenvs/DUE/src/src/Marlin.o] Error 1
========================== [ERROR] Took 18.78 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[ERROR]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 18.78 seconds ==========================
```

Once resolved, could be re-enabled in Travis tests.

-=dave

teemuatlut, 2017-12-15:
I don't think Travis downloads the Archim core (modified from Due core) that is required for this board.

thinkyhead, 2018-02-11:
Looks like #7776 can replace this.

closed 2018-02-11 by thinkyhead