MarlinFirmware/Marlin #8800: [FR] Control RGB LEDs from LCD

Zacharandir, 2017-12-16:
I would like to be able to control my RGB LED strip from the LCD screen. I have been trying to make my own menu entry that behaves similar to Case Light, but my coding skills/understanding of Marlin are not up to the task. 

I was envisioning a menu that can adjust RGB Brightness, or even just the ability to toggle the LED strip between a default R,G,B setting and off.

Thanks,
-Zachrandir

P.S. If someone could point me in the direction of a guide on how Marlin is put together, I would greatly appreciate it. I would love to learn how to define my own G/M codes, add new LED feedback options and create/edit LCD entries. I do have some entry level experience in C++ and am willing to learn.

psavva, 2017-12-16:
For things like this, i will typically create a gcode file in my sd card, and execute it from there...

Bergerac56, 2017-12-16:
Am I wrong or what you ask is enabled by uncommenting : #define LED_CONTROL_MENU in configuration_adv ? (At least in bugfix 2.0)

Zacharandir, 2017-12-16:
@Bergerac56 Thanks! I should have checked newer versions before I posted it was added in 1.1.7.

closed 2017-12-16 by Zacharandir