MarlinFirmware/Marlin #8801: Probing points setup

Thesecky, 2017-12-16:
Hi. After i installed my clips that hold the glass to the bed, my hotend always crashes into them when manualy leveling the bed using LCD. There is probbaly a settign that can offset the probe points, i just cant find the right one. I have spend 2 days searching for the right line of code, i still didnt find it. Can anyone point me to where this could be?

HekTron802, 2017-12-16:
Increase the value of `#define MESH_INSET`. This will be located under the section "Unified Bed Leveling" if you're using UBL with manual probing or under "MESH" if you're using the Mesh. Both of these are at the end of the bed leveling section.

Thesecky, 2017-12-16:
#define MESH_INSET 10          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 3    // Don't use more than 7 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

I have upped that from 10 to 30.  Is that to much, my bed is 220*220?

HekTron802, 2017-12-16:
> I have upped that from 10 to 30. Is that to much, my bed is 220*220?

I don't know, the value of the inset is in mm, so when you run the bed leveling marlin will for example, home the X axis then move in the value of mesh inset and it's max travel will be 220-Mesh_inset. To get a more "precise" measure, home your printer then move the X axis until you clear the clips, then use that number as the mesh inset.  Basically the Mesh inset in an inscribed square(or rectangle) on your print bed.

Another option and the one I use, is to move the endstops inward and then set the size of your bed so that at full travel the nozzle won't hit the clips. You'll want software max endstop option on and I assuming that your printer has physical min endstops. Although I do lose some print space I don't have to hassle around with inset settings(just set it to 0) or moving objects in the slicer if they might hit the clamps (or forgetting to do that and ruining a print!). 

Don't spend all day on this, get back to printing!

Thesecky, 2017-12-16:
Thanks, your suggestion with meassuring worked. In the future, mybe i will also made the endstops as you suggested.

closed 2017-12-16 by Thesecky