MarlinFirmware/Marlin #8805: [2.0.x] LEVEL_BED_CORNERS times out after one lap around bed

DaveAhrendt, 2017-12-16:
Using the LEVEL_BED_CORNERS has a small issue.  After boot you can make one lap around the bed with the Next prompt remaining for an indefinite period.  After moving to the 4th point the menu will go back to the main menu after about 15 seconds. Each time after the next point will return to the main menu and you get to start over.   I'd like to be able to make several laps around the bed setting the corners.  You need to reboot to get back to the indefinite period for each position.

thinkyhead, 2017-12-25:
Try this:
```diff
    void _lcd_level_bed_corners() {
-     defer_return_to_status = true;
      lcd_goto_screen(_lcd_corner_submenu);
+     defer_return_to_status = true;
      bed_corner = 0;
      _lcd_goto_next_corner();
    }
```

thinkyhead, 2017-12-25:
What version of Marlin are you using?

DaveAhrendt, 2017-12-25:
I just tried this again using the 2.0.xbugfix from late yesterday.  Something has changed or I didn't see what was really happening earlier.  If I home the printer and then select Level Corners it will go to the left front corner as expected.  What I thought was happening before is that it would wait here "forever" for me to select the next corner.  What actually happens is that it will wait for 2 minutes +/- then return to the Info screen.  During this time I can move to the remaining 3 corners and it will wait for some time.  However, once I step again to get to the front left again it only waits 15 seconds before returning to the Info screen.  This 15 second behavior now remains until I reboot the printer.
I tried this suggested change to no effect:

    void _lcd_level_bed_corners() {
-     defer_return_to_status = true;
      lcd_goto_screen(_lcd_corner_submenu);
+     defer_return_to_status = true;
      bed_corner = 0;
      _lcd_goto_next_corner();
    }

DaveAhrendt, 2017-12-26:
A bit more information...
When I replace my RepRap Discount Full Graphic Display with a 20x4 line display the problem goes away. It will remain on the next corner page indefinitely. I have tried this now with both 1.1.8 and 2.0bugfix.

thinkyhead, 2018-02-11:
Presuming this is solved now.

DaveAhrendt, 2018-02-11:
Actually not.  Just downloaded the bugfix .zip this morning and have the same issue.  If UBL is defined for bed leveling the level corners function will “time-out” and return to the main menu after some delay. Without UBL it will stay at position for as long as you want.

From: Scott Lahteine 
Sent: Sunday, February 11, 2018 1:06 AM
To: MarlinFirmware/Marlin 
Cc: DaveAhrendt ; Author 
Subject: Re: [MarlinFirmware/Marlin] [2.0.x] LEVEL_BED_CORNERS times out after one lap around bed (#8805)

Presuming this is solved now.

Roxy-3D, 2018-02-11:
Until the full fix is identified...    You can probably work around the problem by giving the printer a 

G29 P1 U T      And let it get enough points to help you understand the bed's tilt.    It should get the corners probed and and some in-between points with the first 8 or 10 probes.

DaveAhrendt, 2018-02-11:
Thanks for the reply.  I assume and had intended to use the level corners feature to perform a mechanical setup before trying to create any mesh.  I wanted to be able to move to a corner and add/remove shims or whatever and then move to the next corner. I guess I’ll jut write 4 macros to move to the corners.

From: Roxy-3D 
Sent: Sunday, February 11, 2018 12:47 PM
To: MarlinFirmware/Marlin 
Cc: DaveAhrendt ; Author 
Subject: Re: [MarlinFirmware/Marlin] [2.0.x] LEVEL_BED_CORNERS times out after one lap around bed (#8805)

Until the full fix is identified... You can probably work around the problem by giving the printer a

G29 P1 U T And let it get enough points to help you understand the bed's tilt. It should get the corners probed and and some in-between points with the first 8 or 10 probes.

Roxy-3D, 2018-02-11:
Yes...   That is the purpose of the U parameter for G29 P1.   It is for Unlevel beds.   Instead of starting the probing where ever the nozzle is and trying to to find the next closest location to probe...   It does the opposite.   It searches for the next mesh point that is the furthest from any previously probed locations.   It tries to spread the points out as much as possible so you can see what needs to be done to level the bed.

thinkyhead, 2018-02-14:
@DaveAhrendt Does the timeout to the status screen happen to coincide with your stepper inactive time timeout? I see that we have this in the `manage_inactivity` function:
```cpp
if (MOVE_AWAY_TEST && stepper_inactive_time && ELAPSED(ms, gcode.previous_cmd_ms + stepper_inactive_time)
    && !ignore_stepper_queue && !planner.blocks_queued()) {
  . . .
  #if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(ULTIPANEL)  // Only needed with an LCD
    ubl.lcd_map_control = defer_return_to_status = false;
  #endif
}
```

This also seems to now be a general behavior for `M18` whenever UBL is enabled. Any call to `M18` will cause the LCD to go back to the status screen.

thinkyhead, 2018-02-14:
I suggest this change:
```diff
  #if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(ULTIPANEL)  // Only needed with an LCD
-   ubl.lcd_map_control = defer_return_to_status = false;
+ if (ubl.lcd_map_control) ubl.lcd_map_control = defer_return_to_status = false;
  #endif
```

DaveAhrendt, 2018-02-14:
I tried this change and it seems to fix the issue.  The head will stay in position for a indefinite period.  I waited for an hour so I assume this takes care of it.  I've not noticed any negative side effects. 

I made the change in both places in marlin_main.cpp. I assume that this is what is needed.

thinkyhead, 2018-02-23:
Yes, that would do it. Thanks for the feedback. I'm glad this is resolved.

commit 113a5fa8 (2018-02-23): Fix broken LEVEL_BED_CORNERS

Fix #8805
commit 65519263 (2018-02-23): Fix broken LEVEL_BED_CORNERS

Fix #8805
closed 2018-02-23 by thinkyhead