MarlinFirmware/Marlin #8807: [1.1.x] G26 Validation Print ignores home offset UBL

WheresWaldo, 2017-12-16:
### Bug Report
- Description: Attempts to print **G26** Mesh validation print, prints at Z endstop rather than Z0 using release 1.1.7---
- Expected behavior: **G26** should respect **M206 Znnn** when used for Z offset. ---
- Actual behavior: Mesh validation print prints in the air at the actual endstop position rather than accounting for the home offset. ---
- Steps to reproduce: All steps done from terminal
1. Clear all EEPROM with **M502** followed by **M500**.
2. **G28** to home bed bed homes in proper location above bed. 
3. Use any available method to determine how far above bed the endstop is triggered. 
4. **M206 Zn.nn** sets home offset. 
5. Verify home offset by issuing a **G1 Z0** command.
6. Initiate UBL auto probe with **G29 P1**.
7. Verify mesh is created with **G29 T**.
8. Save mesh to **G29 S1**.
9. Activate with **M420 S1**.
10. Save to EEPROM with **M500**.
11. Attempt validation print with **G26 R**. (all default setting are good).

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1565038/config.zip)
Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

thinkyhead, 2017-12-25:
- Where is your Z endstop located?
- What is your `M206` Z offset?

WheresWaldo, 2017-12-25:
Z min
M206 Z = 4.05  M851 Z = 0.00

thinkyhead, 2018-02-11:
Presuming solved.

closed 2018-02-11 by thinkyhead