MarlinFirmware/Marlin #8809: Blue LED turns off at full brightness

Zacharandir, 2017-12-16:
Bug Report
- Description: My blue LED light turns off when set to 255, works fine for values less than 255
- Expected behaviour:  Send M150 B255 LED strip turns blue
- Actual behaviour: Send M150 B255 Blue LED turns off 
- Steps to reproduce:
  Send M150 B255 light turns off 
  Send M150 B254 Light comes back on
OR 
  Navigate to LED Control --> Custom Lights --> Blue Intensity
  Set Blue intensity to 255

Configuration files attached:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1565076/Configurations.zip)

thinkyhead, 2017-12-16:
Just the blue pin? Can you try switching the pin numbers in your configs and see if it still affects only the blue pin? Swap blue with red so it's obvious…
```cpp
#if ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define RGB_LED_R_PIN 4 //34
  #define RGB_LED_G_PIN 5 //43
  #define RGB_LED_B_PIN 6 //35
  #define RGB_LED_W_PIN -1
#endif
```

thinkyhead, 2017-12-16:
If it changes so that the red pin comes on when it's supposed to be off, that could mean a pin conflict — pin 4 is also used for something else.

Tannoo, 2017-12-17:
Works fine for me on bugfix-2.0.x.

Tannoo, 2017-12-17:
I would say try `bugfix-1.1.x` and see if the issue is still there.

Tannoo, 2017-12-17:
This has always been my setup:
```
  #define RGB_LED_R_PIN 4
  #define RGB_LED_G_PIN 5
  #define RGB_LED_B_PIN 6
```

Zacharandir, 2017-12-17:
I swapped pins, and it appears to be associated with Pin 4. I haven't encountered anything turning on when it's supposed to be off, only the opposite. I am looking for possible conflicting pins. Case light is also assigned to pin 4 but I have it disabled. I will try the bug fix branch when I get a chance.

thinkyhead, 2017-12-17:
Try adding to your `Configuration.h`…
```cpp
#define FAN_PIN -1
```

Tannoo, 2017-12-17:
Ah... ok. I have `BOARD_RAMPS_14_EFB`

Zacharandir, 2017-12-17:
I added that and it seemed to fix my issue. Thank you for the help. Will adding that line affect the part cooling fan? I have one but I have not yet installed it.

Is there a physical difference between BOARD_RAMPS_14_EeB and BOARD_RAMPS_14_EFB or is it just a change in pin assignments? I recently had to replace my Ramps 1.4 and I'm not sure that I have an exactly identical replacement.

Tannoo, 2017-12-17:
It's just pin assignments.

Tannoo, 2017-12-17:
Those board assignments refer to the D8, D9, and D10 mosfet outputs.

EEB = Extruder, Extruder, Bed
EFB = Extruder, Fan, Bed

thinkyhead, 2017-12-18:
> Will adding that line affect the part cooling fan? I have one but I have not yet installed it.

At that time you'll change `FAN_PIN` from -1 to whatever pin your fan is attached to.

closed 2017-12-18 by thinkyhead