MarlinFirmware/Marlin #8812: Extruder motor not working after marlin update

jrubinstein, 2017-12-16:
After I had this issue: https://github.com/MarlinFirmware/Marlin/issues/8775, I tried to update to the latest bugfix release. Now, the extruder motor will not work. 
I have tried: 

- Putting the extruder onto the Z axis, which worked, so the motor is still good
- Putting a different driver onto the extruder, which did not work 
- Innumerable permutations of the config file (see attached) 
- New Arduino mega in case I'd burned out a pin

 I'm using this ramps board on an anet a8 https://www.amazon.com/gp/product/B06XS8G1BB/ref=oh_aui_search_detailpage?ie=UTF8&psc=1 

Let me know if you have any suggestions, I'm at my wit's end on this one. 

[jrubinstein-marlin-configs.zip](https://github.com/MarlinFirmware/Marlin/files/1565122/jrubinstein-marlin-configs.zip)

Spawn32, 2017-12-16:
I dont know the RAMPS boards, but my guess is that you need to set it up to something else than:
#define MOTHERBOARD BOARD_RAMPS_14_EFB

If you have a look in the pins.h file you will se that there is a lot of different versions of those boards..

jrubinstein, 2017-12-16:
The only pins files I see are:

- pins_RAMPS_13
- pins_RAMPS_OLD
- pins_RAMPS_PLUS 
- pins_RAMPS 

AFAIK, pins_RAMPS covers most of the ramps boards. pins_RAMPS_13 just points to pins_RAMPS. It _could_ be a RAMPS plus but that does seem a bit unlikely. 
Any thoughts on how I could figure out what the pin for that motor is supposed to be?

thinkyhead, 2017-12-17:
So, the E stepper still doesn't work with —for example— Marlin 1.1.3?

You might want to check the pins on the stepper driver plug with an oscilloscope and make sure that they all have the correct signals.

jrubinstein, 2017-12-17:
I can't get the older version to build, I have no idea why. I don't have an oscilloscope, either :(

thinkyhead, 2017-12-17:
> I can't get the older version to build, I have no idea why.

Getting a compile error?

> I don't have an oscilloscope, either :(

Ah well… It sounds like there might be a hardware issue, but we'll wait till you can build an earlier Marlin. (It may help to start with a clean config and just set your basic options.)

jrubinstein, 2017-12-17:
I tried a clean build of 1.1.3 - it built, but did not fix the issue. 
I tried going back to 1.1.1, but still the extruder did not work. The driver is good, the motor is good. So that leaves A) software or B) the RAMPS board itself. 
I've ordered another board ($7 on amazon!) and we'll see if that fixes the issue.

KevMags, 2017-12-17:
Marlin defaults to not allow cold extrusion.  Is the hotend hot or cold extrusion enabled (M302 S0) when you test the extruder motor?

jrubinstein, 2017-12-17:
@kevmags would that prevent me from even moving the extruder motor under the "prepare" menu? How can I disable that in configuration.h?

KevMags, 2017-12-17:
Yes, preheat the hotend and try it.  To change in configuration.h, Line #421 (or nearby) comment out "define PREVENT_COLD_EXTRUSION". (Replace with "/#define PREVENT_COLD_EXTRUSION".

jrubinstein, 2017-12-17:
That was it! Holy crap, thanks for pointing that out. :) 

Now, I need to get all of my settings dialed in, but the z-offset parameter isn't available on the LCD any more. :/

closed 2017-12-25 by thinkyhead

yuriamado, 2019-07-31:
I'm having the very same problem, but unfortunately, without the happy ending. Every single word of what jrubinstein said could be mine, except that I already had tried to deal with the cold extrusion prevention commands, and it doesn't change anything. Could anyone have another guess? 

Just reassuring,

I have tried:

Putting the extruder onto the Z axis, which worked, so the motor is still good
Putting a different driver onto the extruder, which did not work
Innumerable permutations of the config file (see attached)
New Arduino mega in case I'd burned out a pin