MarlinFirmware/Marlin #8816: [1.1.7] Height difference with no G29 command in GCODE after leveling

AkBKukU, 2017-12-17:
### Bug Report
- *Description*: I have two copies of the [same GCODE](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1). One of them I have a commented out lines [14&15](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-sim_adapter-gcode-L14) to disable a `G29` and `M500`. There is a difference in the Z axis on the printed object when there should not be.

- *Expected behavior*: Both files should print the object in the same way. I would like to not need to auto level the bed every time I have to print if I don't move the printer or change the bed temperature.

- *Actual behavior:* I run the first file that does auto leveling and I get [a print that was extruded too high](https://photos.app.goo.gl/VSkLhhei3zxCvMIG3). Without restarting(or with restarting since I run `M500`) I then run the file where `G29` & `M500` is disabled and I get [a print that is extruded too low](https://photos.app.goo.gl/3PRLucAchxu5Fd6b2).

- *Steps to reproduce*:
 1. Generate GCODE with a `G29`
 2. Make a copy of the GCODE file and disable the `G29` command
 3. Run file with `G29`
 4. Clean off bed and run file without `G29`

- *Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.*
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1565222/Config.zip)
Here is a link to a commit of my changes to the default Anet A8 configuration files: https://github.com/AkBKukU/Marlin/commit/e01842411d43885f0427b413ce7075e9a1999f62

Additional:
Printer: Anet A8
Relevant Modifications:
 - 4mm Inductive Sensor
 - Octoprint used to run GCODE files

thinkyhead, 2017-12-17:
We'll certainly look into this.

Note that you probably shouldn't include `M500` in your G-code. The EEPROM can handle a lot of writes before it dies, but its best not to tempt fate.

thinkyhead, 2017-12-17:
If you already leveled the bed once and have saved your data with `M500` then all you need in your G-code is `M420 S1` to enable leveling using the data from your previous run.

AkBKukU, 2017-12-17:
> Note that you probably shouldn't include `M500` in your G-code. The EEPROM can handle a lot of writes before it dies, but its best not to tempt fate.

That is something important to point out. In this case though I want to manually create several G-code files with different bed temperatures that I'll leave on the SD card. They will preheat the bed, run the auto level, save the data and be done. Than way I only need to run one of those files if I change my printer's location or need a different bed temp.

> If you don't use M420 S1 then you will be printing with leveling disabled.

That is most likely the issue. I'll add that to my same G-code and try it out again.

thinkyhead, 2017-12-17:
> That way I only need to run one of those files if I change my printer's location or need a different bed temp.

True. If you'd like to try UBL, it has an option to save and load more than one mesh on EEPROM.

AkBKukU, 2017-12-17:
> True. If you'd like to try UBL, it has an option to save and load more than one mesh on EEPROM.

I saw that in the wiki, that looks a lot like what I want to achieve. I'm definitely going to be looking into that.

> That is most likely the issue. I'll add that to my same G-code and try it out again.

I'm stilling doing some testing but so far adding `M420 S1` has not fixed the issue. In about 15m I'll be finished testing that and I'll post again with the results.

AkBKukU, 2017-12-17:
It is now working. I'm not sure what my problem was in the last post but now if I run[ the file with a `G29`](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c-gcode) then afterwards run [the file with a `M420 S1`](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c_nolevel-gcode) it works as expected now.

I'm going to get this working reliably for me with the separate G-code files concept then I'll look at UBL.

Thank you for the help!

AkBKukU, 2017-12-17:
This problem is plaguing me. I took the G-code files and put them on the SD card and printed them from there while connected to a computer logging the serial output. I ran one file after another without restarting the printer.

The [file with `G29`](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c-gcode) does correctly [output a bed level mesh](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c-log-L345) in the log. This gives me [this result](https://photos.app.goo.gl/HiTiyA4QWgbPt5Ek2) as a print.

The file without `G29` but [with `M420 S1`](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c_nolevel-gcode) does say that [it is using bed leveling](https://gist.github.com/AkBKukU/ca9380338dc87c67f3b97b6f75bd2aa1#file-herman_sim_60c_nolevel-log-L228) in the log. But the resulting print [is still different
](https://photos.app.goo.gl/jVydYi59xmmjo8rJ2).

I don't know if this would be an indication of anything but this is the resulting mesh:
```
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.123 +0.095 +0.020 -0.050 -0.130 -0.060 -0.037
 1 +0.118 -0.022 +0.048 -0.040 -0.027 +0.003 +0.075
 2 +0.035 -0.025 +0.018 -0.030 -0.040 +0.028 +0.113
 3 -0.097 -0.050 -0.063 -0.072 -0.025 +0.078 +0.187
 4 -0.130 -0.075 -0.055 -0.065 -0.020 +0.105 +0.253
 5 -0.185 -0.132 -0.107 -0.037 +0.045 +0.182 +0.345
6 -0.260 -0.140 -0.105 -0.030 +0.073 +0.225 +0.413
```

The area where I am printing has a maximum deviation of -0.072, but when I tried to calibrate my Z-offset for the prints with `G29` that were too high I had to take off more than -0.4 to get it anywhere near close to the prints without `G29`. This resulted in any prints without `G29` driving the nozzle into the bed.

AkBKukU, 2017-12-17:
I may have found a clue as to what is happening
```
[10:28:16:761] Bilinear Leveling Grid:␊
[10:28:16:761]       0      1      2      3      4      5      6␊
[10:28:16:766]  0 +0.182 +0.100 +0.030 -0.007 -0.015 +0.013 +0.100␊
[10:28:16:770]  1 +0.150 +0.093 +0.083 +0.050 +0.025 +0.080 +0.168␊
[10:28:16:774]  2 +0.123 +0.078 +0.060 +0.023 +0.025 +0.093 +0.230␊
[10:28:16:780]  3 +0.020 +0.010 +0.063 +0.003 +0.080 +0.160 +0.305␊
[10:28:16:782]  4 -0.037 -0.010 +0.010 +0.030 +0.088 +0.228 +0.375␊
[10:28:16:788]  5 -0.117 -0.127 -0.012 +0.015 +0.093 +0.260 +0.430␊
[10:28:16:793]  6 -0.087 -0.075 +0.015 +0.060 +0.155 +0.320 +0.523␊
[10:28:16:796] ␊
[10:28:16:796] X:203.00 Y:206.00 Z:10.69 E:0.00 Count X:20300 Y:20600 Z:4484␊
[10:28:16:803] ok␊
[10:32:02:994] Bilinear Leveling Grid:␊
[10:32:02:994]       0      1      2      3      4      5      6␊
[10:32:02:999]  0 -0.270 -0.357 -0.440 -0.492 -0.530 -0.510 -0.407␊
[10:32:03:003]  1 -0.427 -0.400 -0.447 -0.502 -0.520 -0.460 -0.352␊
[10:32:03:008]  2 -0.455 -0.465 -0.490 -0.505 -0.502 -0.415 -0.330␊
[10:32:03:013]  3 -0.522 -0.520 -0.527 -0.535 -0.465 -0.377 -0.260␊
[10:32:03:016]  4 -0.587 -0.540 -0.537 -0.522 -0.430 -0.327 -0.185␊
[10:32:03:021]  5 -0.577 -0.527 -0.507 -0.447 -0.367 -0.245 -0.092␊
[10:32:03:027]  6 -0.655 -0.602 -0.520 -0.462 -0.387 -0.212 +0.023␊
[10:32:03:029] ␊
[10:32:03:029] X:203.00 Y:206.00 Z:10.69 E:0.00 Count X:20300 Y:20600 Z:4484␊
[10:32:03:036] ok␊
[10:35:54:825] Bilinear Leveling Grid:␊
[10:35:54:825]       0      1      2      3      4      5      6␊
[10:35:54:830]  0 -0.867 -0.895 -0.965 -1.030 -1.075 -1.035 -0.960␊
[10:35:54:833]  1 -0.942 -0.967 -1.000 -1.042 -1.065 -0.970 -0.895␊
[10:35:54:838]  2 -0.975 -1.007 -1.025 -1.020 -1.045 -0.940 -0.807␊
[10:35:54:843]  3 -0.967 -0.997 -0.995 -1.020 -0.970 -0.880 -0.745␊
[10:35:54:846]  4 -1.035 -1.045 -1.042 -1.005 -0.977 -0.870 -0.720␊
[10:35:54:852]  5 -1.175 -1.115 -1.100 -1.035 -0.950 -0.792 -0.625␊
[10:35:54:857]  6 -1.210 -1.127 -1.067 -0.988 -0.935 -0.770 -0.520␊
[10:35:54:860] ␊
[10:35:54:860] X:203.00 Y:206.00 Z:10.69 E:0.00 Count X:20300 Y:20600 Z:4484␊
[10:35:54:867] ok␊
[10:40:00:013] Bilinear Leveling Grid:␊
[10:40:00:013]       0      1      2      3      4      5      6␊
[10:40:00:018]  0 -1.312 -1.430 -1.472 -1.530 -1.575 -1.545 -1.465␊
[10:40:00:021]  1 -1.377 -1.457 -1.480 -1.535 -1.557 -1.500 -1.400␊
[10:40:00:026]  2 -1.430 -1.472 -1.490 -1.522 -1.542 -1.442 -1.357␊
[10:40:00:032]  3 -1.502 -1.505 -1.527 -1.542 -1.490 -1.395 -1.260␊
[10:40:00:035]  4 -1.560 -1.572 -1.560 -1.545 -1.482 -1.352 -1.202␊
[10:40:00:040]  5 -1.707 -1.645 -1.585 -1.512 -1.457 -1.310 -1.132␊
[10:40:00:045]  6 -1.697 -1.620 -1.562 -1.510 -1.417 -1.237 -1.027␊
[10:40:00:049] ␊
[10:40:00:049] X:203.00 Y:206.00 Z:10.69 E:0.00 Count X:20300 Y:20600 Z:4484␊
[10:40:00:055] ok␊
```

The bed level mesh doesn't seem to clear after running it repeatedly.

[Full log.](https://gist.github.com/AkBKukU/5ac83b622204c26f7fbc56acf171f61b#file-bed-level-log)

AkBKukU, 2017-12-17:
I've been getting my self acquainted with the code base(`Marlin_main.cpp` is a *steep* hill to climb). I've run basically the same test with`DEBUG_LEVELING_FEATURE` enabled and have noticed that the bed level information seems to change by an increasing multiple of the result from `bilinear_z_offset(current_position))`. [Here is a log of those results](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log). The commands run in that were:
```
G28
M111 S32
G29
G29
G29
```

- The first `G29`(n=0) gave [+0.250 for [0,0]](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log-L127). `0.25 - n *0.52 = 0.25`
- The second `G29` (n=1) stated "ABL Adjustment" [was +0.52](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log-L144) and [gave -0.262 for [0,0]](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log-L226). `0.25 - n *0.52 = -0.27`
- The third `G29` (n=2)[ gave -0.795 for [0,0]](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log-L325) . `0.25 - n *0.52 = -0.79`
- The forth `G29` (n=3)[ gave -1.315 for [0,0]](https://gist.github.com/AkBKukU/4010e77ec32aa6bfe88b642d3971cc2a#file-level-debug-log-L325) . `0.25 - n *0.52 = -1.31`

I may be on a wild goose chase here but this seems like something is wrong. I'm going to do the same print with a `G29` two times in a row now and see if anything unexpected happens.

I also looked at the what the most recent commit was that changed `bilinear_z_offset` which ended up being [Operate in Native Machine Space](https://github.com/MarlinFirmware/Marlin/commit/640526f0c89278983cc9393d54eb75106ebb0d40#diff-1cb08de130a6ece2d1b5b9c37bcfef48) and I reverted to [the commit just before that](https://github.com/MarlinFirmware/Marlin/commit/13a3fb1accec51adb12f7af4f45d6d63f4bc8a3f#diff-1cb08de130a6ece2d1b5b9c37bcfef48) and still had the issue.

AkBKukU, 2017-12-18:
I tried the consecutive prints of the same G-code file over and over and it worked as expected. I had a suspicion that `G28` might be resetting it. So I reset and ran a couple `G29`s to get the exaggerated mesh issue followed by a `G28` and one more `G29` and the mesh was back to where it should be. So I removed the `G28` from the G-code, ran the file twice, and it did go too low the second time. So the level mesh is effecting it.

The easy answer to this problem is "Always add a `G28` & `G29` to every G-code file". And while that would work I'm concerned about the z-offset discrepancy between files with and without `G29`. If I set the offset for files with a `G29` then any files without it would try to print about 0.5mm below the bed. 

In the mean time I was able to cram UBL onto the lowly ATMEGA1284P after disabling the SDCARD, the LCD, and after quite a lot of head scratching `DEBUG_LEVELING_FEATURE`. So far it seems to be working well.

I think I'm not the only one with this issue, both #8831 and #8817 seem similar to this to me.

ikarisan, 2017-12-18:
I have the same (similar)  problem. If auto leveling is disabled everything works fine. With enabled "bilinear leveling" my prints are starting below the configured / measured Z zero.
https://github.com/MarlinFirmware/Marlin/issues/8831

thinkyhead, 2017-12-25:
@AkBKukU Thanks for the detective work! I will look very closely at the situation. I'm not sure how there ends up being some cumulative error, but that makes perfect sense to the results you're seeing.

Can you also test whether `M420 S0` before each `G29` prevents the accumulation of error?

Meanwhile, I will pore over the `G29` code and see where something might be persisting that shouldn't.

> I was able to cram UBL onto the lowly ATMEGA1284P after disabling the SDCARD, the LCD, and after quite a lot of head scratching DEBUG_LEVELING_FEATURE. So far it seems to be working well.

Wow. Somehow Timothy Hoogland gets his EZABL system to fit on the 1284P, keeping the LCD and SD. It's probably very simplified.

thinkyhead, 2017-12-25:
> Always add a G28 & G29 to every G-code file

Or, `G28`, `M420 S1` (to use the last-saved mesh).

thinkyhead, 2017-12-25:
@AkBKukU — Try this change in `gcode_G29`…
```diff
      // Disable auto bed leveling during G29
-     planner.leveling_active = false;
+     set_bed_leveling_enabled(false);
```

thinkyhead, 2017-12-25:
@AkBKukU — I made this change in `gcode_G29` along with some others.
```diff
      stepper.synchronize();

      // Disable auto bed leveling during G29
-     planner.leveling_active = false;
+     set_bed_leveling_enabled(false);

      if (!dryrun) {
```
It's always better to disable bed leveling with the function instead of turning it off directly. The cumulative error is probably due to not resetting the current position back to the current physical position.

Test the current `bugfix-1.1.x` which has the patch and see if the issue is resolved.

AkBKukU, 2017-12-28:
I reflashed with the current `bugfix-1.1.x` branch and switched back to ABL. 

Consecutive `G29` commands now work as expected. The mesh is reset between measurements.

I ran a `G29` and `M500` then did two test prints with just a `M420 S1`. Then I shut the printer off and ran one more `M420 S1` print. All prints were at the correct Z-Offset. 

I would say my problem has now been resolved, thank you! I came to really like UBL so I'll probably go with that in a while. But for now I'm going to continue using ABL.

closed 2017-12-28 by AkBKukU