MarlinFirmware/Marlin #8817: Bi-Linear autolevel seems to ignore Z offset only when printing

Festivejelly, 2017-12-17:
### Bug Report
- Description: When using bi-linear bed leveling with a BL touch for example the offset between nozzle and bed seems to be ignored.
- Expected behaviour: Gap between nozzle and bed should be the same as the Z value
- Actual behaviour: Gap between nozzle and bed is greater than Z value (ignores the offset)
- Steps to reproduce:
  - Set your Z offset with M851 to Z-1.5 for example
  - Heat up the bed and home
  - Do G29
  - Go to middle of the bed and set Z to 0. Probe should be touching bed, or just about.
  - Go and do a print with G28 then G29 in the starting code.
  - Appears the nozzle starts way too high from the bed.

Ive tested this several times and it respects the offset when outside of the print. This only seems to happen during the printing phase.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1565288/Marlin.zip)

I've tried doing M501 to read the values from Eeprom, tweaking with M851 Z-1.5 to set the correct offset then doing M500.

As mentioned when controlling the printer via repetier and doing G29 etc then going to Z0.2 for example. The gap is 0.2. But during a print its starting much higher, like its ignoring m -1.5 offset.

I inspected the Gcode and it had the correct layer heights in there of 0.2

Festivejelly, 2017-12-17:
This has already been fixed in a later release

Festivejelly, 2017-12-17:
This issue is still present in 1.1.7 and can be reproduced.

Festivejelly, 2017-12-17:
My latest updates on Configs.

Using a feeler guage, outside of printing (homing then G29 then going to centre of bed) @ Z=0.1 the feeler gage just about can squeeze under at 0.1.

When doing a print when stopping in the centre of the bed with a layer height of 0.1 the feeler gauge can find a 0.3mm under there.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1565310/Marlin.zip)

ikarisan, 2017-12-17:
Hi!
Maybe we both have the same problem:

https://github.com/MarlinFirmware/Marlin/issues/8831

Festivejelly, 2017-12-17:
I think you could be right. I'm going to test 1.1.5 just to make sure im not going crazy. But it can reproduce it every time. Though for me the print starts higher than its supposed to rather than lower.

ikarisan, 2017-12-17:
Please test 1.1.6. This version worked for me.

Festivejelly, 2017-12-17:
1.1.6 didnt work for me nor did 1.1.7 currently testing 1.1.5

thinkyhead, 2017-12-20:
> Go to middle of the bed and set Z to 0

Set? You mean, with `G92`? Probably a bad idea.

Roxy-3D, 2017-12-20:
My thoughts are we should remove the z_offset variable and code.   It just adds complexity.

thinkyhead, 2017-12-20:
Whether the probe Z offset is constant or a variable, all code that uses the value will be the same.

Maybe getting rid of `M851` would eliminate confusion about the current value (e.g., when forgetting to save to EEPROM), but it appears that @Festivejelly has verified that the probe offset is -1.5 and still has issues. Making the probe offset unchangeable probably wouldn't help this case.

Roxy-3D, 2017-12-20:
> Maybe getting rid of M851 would eliminate confusion about the current value (e.g., when forgetting to save to EEPROM), 

No...   I'm thinking z_offset is the one that needs to go.  M851 controls the zprobe_zoffset value.   These are two different animals.
```
  inline void gcode_M851() {
    SERIAL_ECHO_START();
    SERIAL_ECHOPGM(MSG_PROBE_Z_OFFSET);
    if (parser.seen('Z')) {
      const float value = parser.value_linear_units();
      if (!WITHIN(value, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX)) {
        SERIAL_ECHOLNPGM(" " MSG_Z_MIN " " STRINGIFY(Z_PROBE_OFFSET_RANGE_MIN) " " MSG_Z_MAX " " STRINGIFY(Z_PROBE_OFFSET_RANGE_MAX));
        return;
      }
      zprobe_zoffset = value;
    }
    SERIAL_ECHOLNPAIR(": ", zprobe_zoffset);
  }
```

thinkyhead, 2017-12-20:
ABL doesn't have a `z_offset`, although it accepts a Z offset parameter that will be applied to all mesh points. If left out, there's no offset. UBL used to have a `z_offset`. I removed it as obsolete a while ago, since no code was using it anymore.

v1ru3, 2017-12-20:
I tested 1.1.5, 1.1.6 and now 1.1.7 and the z offset and bilinear bed leveling issues seem to persist. I am using an induction z probe that has the same height distance from the nozzle (fixed probe).
In 1.1.4 i used to be able to set the offset using:
G28
G29
G1 Z0
;add paper between nozzle and bed and decrease height.
M851 Z-"value from above".

And that value stayed the same across all future G29s as it should.

In recent versions (1.5, 1.6 and now 1.7) the behavior is chaotic to say the least, even though the documentation says nothing changed.

In 1.5 it was a hit and miss, in 1.6 after every G29 i had to babystep-substract the previous set z offset to avoid bed scratches. In 1.7, the only way to get bilinear working is to not use z offset at all, leave it at 0 and do babystepping before the print starts.
And by the way, the z offset i measure is nothing like the babystepping value (eg, z-offset is -0.9 when i do babystepping the value is -1.050).

I am at loss, not sure what else to try. Can you please help?

fflosi, 2017-12-22:
What I could found about this is that if I do a G29, then a G28, and reenable the leveling with "M420 S1" it starts to work perfectilly. The problem only happen when I try to test the leveling right after the G29 finishes.
It appears as if the last position of the Z axis after G29 completes is geting lost in some way.

thinkyhead, 2017-12-25:
> It appears as if the last position of the Z axis after G29 completes is geting lost in some way.

What is the value of Z immediately after `G29`, versus when you use `G28`, `M420 S1` ?

The last thing the `G29` procedure does before enabling leveling is:
```cpp
          // Unapply the offset because it is going to be immediately applied
          // and cause compensation movement in Z
          current_position[Z_AXIS] -= bilinear_z_offset(current_position);
```
You might try commenting out that line to see if it has a positive effect.

fflosi, 2017-12-28:
> You might try commenting out that line to see if it has a positive effect

I could not try to comment it yet. I will post a result as soon as I try it out.

> After G29 do a simple G1 X0 Y0 move, then do M420 S0. What is the current Z?

I tried the `M420 S0` and `S1` but every time I issue an `M420 So and S1` i get a different result.
After X0 Y0 Z=1
S0 - Z=0.39
S1 - Z=1.04
S0 - Z=0.43
S1 - Z=1.08
S0 - Z=0.48
S1 - Z=1.12

> •After doing G28, M420 S1, what is the current Z?

Just after G28 Z=0
M420 S1 - Z=0.25
M420 S0 - Z=0.01
M420 S1 - Z=0.26
M420 S0 - Z=0.02
Less change than just after a G29, but still there.

thinkyhead, 2017-12-31:
I'd expect a very small amount of imprecision that could accumulate, but not to the degree that we're seeing here. I wouldn't be surprised if there is some greater element of imprecision due to the plethora of changes to the related functions.

I will have to block out time to add logging and do a lot of intensive testing…

thinkyhead, 2018-02-02:
> the printer does not react to either the `M851` command nor to a change in the `Z_PROBE_OFFSET_FROM_EXTRUDER` setting.

What kind of "reaction" are you looking for?

Try `M851 Z10` followed by `M503 S0`. Does the output show `M851 Z10.00` or has it not been changed? If it has been changed then `M851` is working!

RoyKea, 2018-02-02:
I never tried using `M503 S0`, but instead made changes through e.g. `M851 Z10.00` --˃ `M500` --˃ `M501`. This saved my changes and did output the changed value in the terminal. Is this what you were asking? In any case, even though the output showed that the value was changed, the printer didn't seem to apply it. `Z10`, `Z-4`, it didnt matter. The printer never adjusted the Z height.

In any case, I reflashed the newest 1.1.x bugfix version and now `M851` works as intended for me.

thinkyhead, 2018-02-10:
`M503 S0` tells you what your current settings are.

closed 2018-02-10 by thinkyhead

patternseek, 2018-04-28:
@thinkyhead the original issue seems to be unresolved, I'm confused as to why this is closed.

thinkyhead, 2018-04-29:
@patternseek — Please open a new issue if you're having a similar problem with the current bugfix branches. It may be a regression, it may be a configuration issue, or it may be a matter of procedure. There's nothing unique about the BLTouch Z probe offset. If there's a general problem with the M851 Z value being "ignored" we have not had widespread reports about it. For most setups ABL bilinear works as designed.

Shadesh, 2018-07-30:
Same problem here. Got a Creality Ender 3 with original BLTouch. Everything is set but when printing Marlin seems to ignore the Z Offset completely. Using the 1.1.8 Version.

maqndon, 2018-08-28:
I'm usingtje 1.1.8 too. The same problem. I fixed that 3 months ago but I don't remember how...

thinkyhead, 2018-08-29:
1.1.8 is no longer recommended. Marlin 1.1.9 fixes all the bugs.

maqndon, 2018-08-30:
I have the zoffsets just good. Then I made a m206 x0 y0 and my zoffset turn crazy again... z0 when its leveling but when print it says z4.8 or so...

spilz87, 2018-08-30:
I solved the issue for me like this in marlin_main.c, not sure it's the good way, but it works for me : 
`// Calculate leveling, print reports, correct the position
    if (!isnan(measured_z)) {
      #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

        if (!dryrun) extrapolate_unprobed_bed_level();
        print_bilinear_leveling_grid();

        // PHR
        // moyenne locale pour l'autolevel
        report_current_position();
        const float l = (z_values[0][0] + z_values[1][0] + z_values[2][0] + z_values[0][1] + z_values[1][1] + z_values[2][1] + z_values[0][2] + z_values[1][2] + z_values[2][2]) / 9,
                  v = LOGICAL_TO_NATIVE(l, Z_AXIS),
                  d = current_position[Z_AXIS] - v;
        current_position[Z_AXIS] = d;
        update_software_endstops((AxisEnum)Z_AXIS);
        report_current_position();
        for (uint8_t i = 0; i < 3; ++i) {
          for (uint8_t j = 0; j < 3; ++j) {
            z_values[i][j] -= l;
          }
        }
        print_bilinear_leveling_grid();

        
        
        refresh_bed_level();

        #if ENABLED(ABL_BILINEAR_SUBDIVISION)
          print_bilinear_leveling_grid_virt();
        #endif

    `

spilz87, 2018-08-30:
but I never try to change homing or else after bed leveling, direct print

thinkyhead, 2018-08-30:
The effects of `M206` may be unpredictable. What is it being used for?

Hawkeye0914, 2019-02-07:
Im having this exact same issue.  Is there still no fix for this?

tomdk, 2019-03-17:
Same here in Marlin 1.1.9 and 1.1.x bugfix. Have been going crazy and so happy I have found this now.

spilz87, 2019-03-17:
Hi
Did my code help ? 
If yes I have an small update if you want/need 
Else happy you found this :)

treschlet, 2019-10-14:
I just upgraded my Lulzbot TAZ 6 from 1.1.5 to 1.1.9.34 and am having this issue. z-offset is completely ignored. Doesn't matter what I set it to, from the controls on machine or with the M851 code.

Excentertex, 2020-01-28:
Oh dear! I cannot believe I spent the better part of the week finding an explanation for this problem. 

I have an Anet A8 with stock board and latest bugfix-1.1.x firmware. But it also happened on the 1.1.9 branch. I'm not entirely sure I had the problem before, because I always had to do "some" manual corrections in Cura via the Z offset plugin. However I believe it worsened considerably after I had to re-download Marlin (following a complete computer reinstall). Since then the problem has been very persistent and considerably worse.