MarlinFirmware/Marlin #8823: MESH_TEST_LAYER_HEIGHT ignored

lavato, 2017-12-17:
Not sure if I understand the functionality correctly but I think the MESH_TEST_LAYER_HEIGHT is ignored when doing "Mash Edit" or "Fine Tune All".

When I want to fine tune the Mesh the head stops about 10mm above bed level, so it is impossible to tune the bed Mash.

Expected behaviour:
If I set MESH_TEST_LAYER_HEIGHT to 0.2mm I expect the head to be constantly at that height while doing Mesh adjustments.

Is above expected behaviour correct?

Roxy-3D, 2017-12-17:
> When I want to fine tune the Mesh the head stops about 10mm above bed level, so it is impossible to tune the bed Mash.

The nozzle does not need to move up or down to fine tune the mesh.  The only reason the nozzle parks over various mesh points during a G29 P4 R is so you know which mesh point is being edited.    The Nozzle's Z height has nothing to do with anything.     

And in fact...  You can control whether it moves or not during the editing of the mesh point with
```
  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
```
Most sohpisticated UBL users prefer that the nozzle does not move during a mesh point editing session.

lavato, 2017-12-17:
OK I see  ... however I am confused in that case,
Once the UBL "Cold mash is built how can I "Fine Tune" the mash, as I have no point of reference?
The head is 10mm above the bed level, if I need to adjust the Mesh value for a particular Mesh point using LCD dial I would like to be able to use e.g. Paper and manualy adjust the Mesh offset based on resistance.

In any case as the head parks about 10mm above the bed - is there any way I can adjust this so I can use to Live edit the Mesh points.

Roxy-3D, 2017-12-17:
Well...   You are making some assumptions that are not true.   Once the initial mesh is built and filled out...   It should be fairly close to 'usable' across the entire bed.   But especially in the case of inductive probes where there is an 'edge effect' some probed values may be off a little bit.

That is why you do a G26 and generate a Mesh Validation Pattern.   From that you can see which mesh points might be a little bit too high or a little bit too low.   Usually, for most of the mesh points nothing needs to be done.   But for the few mesh points that need adjustment...   Shift the ones that are off a lot by .1 mm in the correct direction.   For ones that are close, but not exactly right, shift them .05mm in the right direction.

Then do another G26 and see if everything is correct.   If not...   Repeat the edit cycle.    And just use the LCD panel to tell you how much you have shifted the mesh point.

lavato, 2017-12-17:
> Well... You are making some assumptions that are not true. 

I am coming to the Mesh functionality fresh, so my comments are somewhat from user perspective. 

>  Once the initial mesh is built and filled out... It should be fairly close to 'usable' across the entire bed

Yes - the Mesh is actually really usable after a first pass,  especially in the middle of the bed, for me. Tribute to contributors!

> That is why you do a G26 and generate a Mesh Validation Pattern.

The issue for me is that the pattern does not print fully, as I have big variances at the sides <0.5mm.
Which is why I think there should be a set Z height which can be used for LIVE mash edits. I also think many users may prefer to do a live edit for a given Mesh position, instead of doing it by eye.
Also in my case it does not work.

**Request:**
For this reason I think we should have a static value that can be set for this purpose?
E.g.
MESH_LIVE_EDIT_Z_HEIGHT. - This would use the existing Mash values and set the Nozzle height to that value.
Possibly even to add another menu option below Mash Edit , e.g. Mesh Edit - Live

**Question**
Once clicking on edit value the probe  goes up for about 10mm, which value is used here to lift the nozzle, the Z_CLEARANCE_BETWEEN_PROBES?

Roxy-3D, 2017-12-18:
>> That is why you do a G26 and generate a Mesh Validation Pattern.
> The issue for me is that the pattern does not print fully, as I have big variances at the sides <=0.5mm.
Which is why I think there should be a set Z height which can be used for LIVE mash edits. I also think some users may prefer to do a live edit for a given Mesh position, instead of doing it by eye.

The fact the pattern does not print completely is exactly why that command is there.    

If you let the pattern start printing at the center of the bed...   It will work its way outward.   If the nozzle gets too close to the glass, you can stop it and raise those mesh points.   And then restart the print.   If the nozzle is too far from the glass to print correctly...   Just leave it be and let the pattern finish.   When the pattern is done, go edit those mesh points and lower them .10 or .15 mm and re-run the command to generate a new pattern.

If you really want to use the nozzle's height to set or edit mesh points, you can do a manual probe by issuing a G29 P2 command.   But really...   You just need the G26 Mesh Validation Pattern to print well enough you can see which areas are too high or too low...  Edit them...   And repeat.   After 2 or 3 iterations, everything will converge.

lavato, 2017-12-18:
> The fact the pattern does not print completely is exactly why that command is there.

Sorry your argument just does not make sense at all. There is actually a bigger risk of a hot nozzle crashing into a bed during a "Pattern print" than during a suggested live edit.

Also the new variable would be e.g. >0.5mm so there is even less chance for this.

>If you really want to use the nozzle's height to set or edit mesh points, you can do a manual probe by issuing a G29 P2 command. But really... You just need the G26 Mesh Validation Pattern to print well enough you 

Yes I could do a manual probe or change the code but I think this would be better for some usecases.
I guess you got used to the current process which is why you may not see the benefit for where printing a pattern don't work.

lavato, 2017-12-19:
I did manage to configure using G26 but it was very painful.
After more thinking, I suggest another step is added, e.g. Live Mesh Edit, to the UBL workflow, just before "Validation", which can also be optional as any other step.

This will allow people who have massive variances between what the probe is giving for some parts of the bed to be able to more easily calibrate Mesh settings.

Roxy-3D, 2017-12-19:
> After more thinking, I suggest another step is added, e.g. Live Mesh Edit, to the UBL workflow, just before "Validation", which can also be optional as any other step.

It would be helpful to see the full description of how this feature works.   Just a title isn't enough.   And it probably needs to be described for Cartesian, Delta's and Scara's.

lavato, 2017-12-19:
> It would be helpful to see the full description of how this feature works. And it probably needs to be described for Cartesian, Delta's and Scara's.

Sure, I'll detail this in a spec, as I see it, in next couple of days.

I  had a quick look at the code ( like the command pattern with the FILO approach design choice ), as I am looking to do this for me, since I am regularly changing hot ends and often need Mash calibration.

So, I could make a pull request for this but have never done it for this project. Is this something the core contributor group allows?

Roxy-3D, 2017-12-20:
> So, I could make a pull request for this but have never done it for this project. Is this something the core contributor group allows?   

Yes..  Just fork one of the bugfix branches...   Make your changes and test them...   And make a pull request.

lavato, 2017-12-20:
Oh ... OK. 
I'll get back to this thread once I finish local Dev.

Roxy-3D, 2017-12-20:
> Sure, I'll detail this in a spec, as I see it, in next couple of days.

If possible, please also do this.   This will allow others to follow the work and suggest improvements...

lavato, 2017-12-20:
>If possible, please also do this. This will allow others to follow the work and suggest improvements...

Yes ... perhaps more important than the pull request - in case there are any major objections, given all the comments above.

NP, will do so in the next couple of days.

Roxy-3D, 2017-12-20:
Here is something to consider...    If the 'Live Adjustment' happens with the nozzle moving, what gets affected?   The nozzle is typically surrounded by 4 mesh points making the corners of a mesh cell.

I think I would argue all 4 mesh points should be affected.  But the closer the nozzle is to a given mesh point, the more that mesh point will be affected.

lavato, 2017-12-20:
I am still going trough the code ,,, so still collecting thoughts but I would tend to agree.

One thing I am still trying to get my head around is how to configure UBL to "see" unreachable areas:

In my case, bed size 190 x 190 and the probe offset is X+30.

The mesh map only includes area (30,0) x (190,190).  So (0,0) x (30,190) is not covered by the mash grid.
How can the missing section be included in the mesh?

E.g.  Map of the bed: (o - included, x -not included)
xooooo
xooooo
xooooo

Roxy-3D, 2017-12-20:
Where are you homing the printer.   That location should have mesh points very close to 0.00 mm.   While no required, it is typically best to home the Z-Axis in the center of the bed.  

> The mesh map only includes area (30,0) x (190,190). So (0,0) x (30,190) is not covered by the mash grid.  How can the missing section be included in the mesh?

Yes...   That may be because you have set a 30mm inset.   Or you have your X_MIN_POS at 30.    If you post your configuration.h and configuration_adv.h it won't be difficult to figure out where that is coming from.

lavato, 2017-12-21:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1577446/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1577447/Configuration_adv.txt)

>Yes... That may be because you have set a 30mm inset. Or you have your X_MIN_POS at 30. If you post your configuration.h and configuration_adv.h

 it won't be difficult to figure out where that is coming from.

(attached) 

I only have:
- HOTEND_OFFSET_X  set to (0,30.40) , as I have 2 extruders and 
- X_PROBE_OFFSET_FROM_EXTRUDER 30 => which is requred

As a user, I would expect area  (0,0) x (30,190) to be unreachable and for me to be able to extrapolate the numbers using the current "Auto Fill" function e.g. (G29 P3.1)

Roxy-3D, 2017-12-21:
Can you also cut & paste a G29 W for me?  That will give all the calculations to determine bed size and mesh point locations.

The 2nd extruder should not be factoring into this issue.  Pretty much, the mesh is setup using the first extruder.   There may be locations on the mesh that the 2nd extruder can not reach.

lavato, 2017-12-21:
>Can you also cut & paste a G29 W for me? 

Here it is, thank you,

```
SENDING:G29 W
Unified Bed Leveling System v1.01 inactive.
Mesh 3 Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
# of samples: 36
Mean Mesh Height: 0.277482
Standard Deviation: 0.332422
zprobe_zoffset: -1.7099999
MESH_MIN_X  ((((((190) / 2) - (190) / 2) + 0)>(0 + 30)?((((190) / 2) - (190) / 2) + 0):(0 + 30)))=30
MESH_MIN_Y  ((((((190) / 2) - (190) / 2) + 0)>(0 + 0)?((((190) / 2) - (190) / 2) + 0):(0 + 0)))=0
MESH_MAX_X  ((((((190) / 2) + (190) / 2) - (0))<(190 + 30)?((((190) / 2) + (190) / 2) - (0)):(190 + 30)))=190
MESH_MAX_Y  ((((((190) / 2) + (190) / 2) - (0))<(190 + 0)?((((190) / 2) + (190) / 2) - (0)):(190 + 0)))=190
GRID_MAX_POINTS_X  6
GRID_MAX_POINTS_Y  6
MESH_X_DIST  32.00
MESH_Y_DIST  38.00
X-Axis Mesh Points at: 30.000  62.000  94.000  126.000  158.000  190.000
Y-Axis Mesh Points at: 0.000  38.000  76.000  114.000  152.000  190.000
Kill pin on :41  state:1
Unified Bed Leveling sanity checks passed.
```

lavato, 2017-12-23:
@Roxy-3D - Do you think this is a bug?

Roxy-3D, 2017-12-24:
I suspect so.   The nozzle shouldn't be doing that.   The big problem is I don't have a machine with me to duplicate the problem.    But as soon as I'm home I'll try to duplicate the problem.    If I can duplicate it, I should be able to find (and fix) what is causing it.

Can you try turning off SEGMENT_LEVELED_MOVES and see if the problem still exists?
```
  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)
```

lavato, 2017-12-25:
I'll am off for next couple of days, so will do above and check back in next couple of days.
Basically I wanted to ensure I can have a fully working UBL before I start writing the user story and start changing code.

Also I have couple of questions:

-  Will Marlin 2 directory structure be reverted beack,  due to issues with W10. I tried compiling but had to revert to 1.1 bugfix? I think Marlin team should provide a step by step workaround otherwise many people will not use it.
- Do you have any standard way of step debugging Marlin using e.g. a special IDEs or some kind of SPI "hack" which can be utilised for this purpose or is it just via Serial logs?

Oh and since it is, Merry Christmas!

Roxy-3D, 2017-12-25:
Marlin 2's file organization is the future direction.   When everything works and all the build environments do reasonable things, that will become the main development branch.

The 32-bit platforms have J-Link capability.  But the 8-bit boards pretty much are debugged with SERIAL_ECHO() commands.

closed 2018-02-11 by thinkyhead