MarlinFirmware/Marlin #8826: suggestion needed for compile

ruggb, 2017-12-17:
### Compile Error with ver 2.0.x-957bbab and Arduino 1.8.6
When I compile with my config files I get this error.

fork/exec C:\Users\Bill\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

It does not occur compiling the distribution.
how do I resolve it? I can't shorten anything past LOCAL..........

------------------------------------------------------------

Bob-the-Kuhn, 2017-12-17:
Try the Arduino 1.9 beta.  

The "too long" solution is slated for the 1.9 release.

ruggb, 2017-12-17:
didn't work

hicsvntdracones, 2017-12-17:
Remove the IDE's preferences.txt file line 54 # to build.path = F: \ tmp

ruggb, 2017-12-17:
I reinstalled Arduino 1.8.5 and retried. Got the same error with a different path. BUT noticed that the path is not Widows compliant. The same as the previous one -- ..tools\**avr/bin/avr-gcc.exe**
So it isn't TOO LONG, it is looking for a file AVR/BIN/AVR-GCC.EXE in tools folder.
Is that from the IDE or Marlin?

thinkyhead, 2017-12-20:
Seems to an Arduino for Windows issue. Please report to the Arduino project.

closed 2017-12-20 by thinkyhead