MarlinFirmware/Marlin #8830: compile error with 1.1.x-

ruggb, 2017-12-17:
### Compile Error with 1.1.x-a0d452a as distributed - ie virgin

In file included from sketch\MarlinConfig.h:40:0,

                 from D:\Users\Bill\Documents\#Arduino\libraries\Marlin-bugfix-1.1.x. a0d452a new\Marlin-bugfix-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:266: error: #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

     #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

      ^

exit status 1
#error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

Tannoo, 2017-12-17:
a virgin `bugfix-1.1.x` builds fine for me.

```
platformio run --environment megaatmega2560
AVR Memory Usage
----------------
Device: atmega2560
 
Program:   54284 bytes (20.7% Full)
(.text + .data + .bootloader)
 
Data:       2613 bytes (31.9% Full)

(.data + .bss + .noinit)
 
 
 [SUCCESS] Took 57.97 seconds
 
 [SUMMARY]
Environment megaatmega2560      [SUCCESS]
Environment megaatmega1280      [SKIP]
Environment printrboard         [SKIP]
Environment printrboard_revf    [SKIP]
Environment brainwavepro        [SKIP]
Environment rambo               [SKIP]
Environment anet10              [SKIP]
Environment sanguino_atmega644p [SKIP]
 [SUCCESS] Took 57.98 seconds
 
Linter
Severity 	Provider 	Description	Line 
PIO Build (megaatmega2560)
clean-bugfix-1.1.x1 file1 update
```

ruggb, 2017-12-18:
I reinstalled arduino, recompiled, and get the same error

In file included from sketch\MarlinConfig.h:40:0,

                 from D:\Users\Bill\Documents\#Arduino\libraries\Marlin-bugfix-1.1.x. a0d452a\Marlin-bugfix-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:266: error: #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

     #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

      ^

exit status 1
#error "RX_BUFFER_SIZE must be a power of 2 greater than 1."

I think the problem is the sanity check.
It is asking about the buffer, but the buffer isn't defined.
Not sure how it works for you??????

Tannoo, 2017-12-18:
Oooh... I haven't tried the Arduino IDE. Sorry. I dropped that a while ago for Marlin.

Tannoo, 2017-12-18:
Try setting the `RX_BUFFER_SIZE` to `2` and see if it compiles.

Tannoo, 2017-12-18:
In `Configuration_adv.h`, comment out this line: `#define TX_BUFFER_SIZE 0`.

Tannoo, 2017-12-18:
But, it is defined in `MarlinSerial.h`:
```
// Define constants and variables for buffering serial data.
// Use only 0 or powers of 2 greater than 1
// : [0, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048, ...]
#ifndef RX_BUFFER_SIZE
  #define RX_BUFFER_SIZE 128
#endif
```

Tannoo, 2017-12-18:
How about trying to change `MarlinSerial.h`, LN# 94 to:
```cpp
#if RX_BUFFER_SIZE && (!IS_POWER_OF_2(RX_BUFFER_SIZE) || RX_BUFFER_SIZE < 2)
```

Tannoo, 2017-12-18:
I compiled with this in the Arduino IDE:

`MarlinSerial.h`
```cpp
  #if ENABLED(SERIAL_XON_XOFF) && RX_BUFFER_SIZE < 1024
    #error "SERIAL_XON_XOFF requires RX_BUFFER_SIZE >= 1024 for reliable transfers without drops."
  #endif

  #if !IS_POWER_OF_2(RX_BUFFER_SIZE) || RX_BUFFER_SIZE < 2
    #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."
  #endif

  #if TX_BUFFER_SIZE && (TX_BUFFER_SIZE < 2 || TX_BUFFER_SIZE > 256 || !IS_POWER_OF_2(TX_BUFFER_SIZE))
    #error "TX_BUFFER_SIZE must be 0, a power of 2 greater than 1, and no greater than 256."
  #endif
```

```
Sketch uses 53610 bytes (21%) of program storage space. Maximum is 253952 bytes.
Global variables use 2612 bytes (31%) of dynamic memory, leaving 5580 bytes for local variables. Maximum is 8192 bytes.
```

Also changed this in `MarlinSerial.h`:
```cpp
  #if RX_BUFFER_SIZE && (!IS_POWER_OF_2(RX_BUFFER_SIZE) || RX_BUFFER_SIZE < 2)
    #error "RX_BUFFER_SIZE must be a power of 2 greater than 1."
  #endif
```
```
Sketch uses 53610 bytes (21%) of program storage space. Maximum is 253952 bytes.
Global variables use 2612 bytes (31%) of dynamic memory, leaving 5580 bytes for local variables. Maximum is 8192 bytes.
```

No errors. Bone stock, virgin, `bugfix-1.1.x` from this morning.

Tannoo, 2017-12-18:
All using Arduino IDE 1.8.4

thinkyhead, 2017-12-20:
These tests also exist in `MarlinSerial.h`. They should probably be removed from there now that they exist in `SanityCheck.h`.

closed 2017-12-20 by thinkyhead