MarlinFirmware/Marlin #8834: Azteeg X3 Pro and Viki I2c Lcd issues

Gfbubba18, 2017-12-17:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`
Viki_i2c_lcd and azteeg x3 Pro proper code.
------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
In file included from sketch\ultralcd.cpp:97:0:

ultralcd_impl_DOGM.h:209: error: 'DOGLCD_CS' was not declared in this scope

   U8GLIB_DOGM128_2X u8g(DOGLCD_CS, DOGLCD_A0);  // HW-SPI Com: CS, A0 // 4 stripes

                         ^

ultralcd_impl_DOGM.h:209: error: 'DOGLCD_A0' was not declared in this scope

   U8GLIB_DOGM128_2X u8g(DOGLCD_CS, DOGLCD_A0);  // HW-SPI Com: CS, A0 // 4 stripes

                                    ^

sketch\ultralcd.cpp: In function 'void lcd_buzz(long int, uint16_t)':

ultralcd.cpp:736: error: 'lcd' was not declared in this scope

       lcd.buzz(duration, freq);

       ^

In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from sketch\fastio.h:32,

                 from sketch\MarlinConfig.h:26,

                 from sketch\ultralcd.cpp:23:

sketch\ultralcd.cpp: In function 'void lcd_update()':

ultralcd.h:92: error: 'BLEN_C' was not declared in this scope

     #define EN_C (_BV(BLEN_C))

                       ^

sketch\ultralcd.h:172:36: note: in expansion of macro 'EN_C'

     #define LCD_CLICKED (buttons & EN_C)

                                    ^

sketch\ultralcd.cpp:4617:26: note: in expansion of macro 'LCD_CLICKED'

     if (UBL_CONDITION && LCD_CLICKED) {

                          ^

sketch\ultralcd.cpp: In function 'bool lcd_detected()':

ultralcd.cpp:5089: error: 'lcd' was not declared in this scope

     bool lcd_detected() { return lcd.LcdDetected() == 1; }

                                  ^

exit status 1
'DOGLCD_CS' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

------------------------------------------------------------
### Bug Report
- Description: using viki l2c lcd issues on an azteeg x3 Pro board
- Expected behaviour: working LCD
- Actual behaviour: not working on encorders and display pins are not in proper locations according to Panucatt Devices
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin_1.1.7_UPDATED5_12-17-17.zip](https://github.com/MarlinFirmware/Marlin/files/1566158/Marlin_1.1.7_UPDATED5_12-17-17.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2017-12-25:
Looking at your configuration, the first thing I notice is you need to comment out this line:
```cpp
#define DOGLCD      // Full graphics display
```

boelle, 2019-02-19:
@Gfbubba18 did thinky's suggest help?

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

InsanityAutomation, 2019-08-25:
At a glance, it seems there are currently issues with the button assignment. They should be read over 12c, but there is a compile error looking for the pin. next_button_update_ms in the check function has an undeclared error as well. Id say to get support for this lcd working properly, hardware will be needed. Its probably been broken too long to just get it to compile again and call good. Itll need validation.