MarlinFirmware/Marlin #8836: PARK_HEAD_ON_PAUSE does nothing

witorwitor, 2017-12-18:
I have enabled  ADVANCED_PAUSE_FEATURE and PARK_HEAD_ON_PAUSE but when I choose pause on the graphical screen of my Anet A6 print just stops without moving away from the print. Is it a bug or am I missing something? 

#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_X_POS 3                  // X position of hotend
  #define PAUSE_PARK_Y_POS 3                  // Y position of hotend
  #define PAUSE_PARK_Z_ADD 10                 // Z addition of hotend (lift)
  #define PAUSE_PARK_XY_FEEDRATE 100          // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define PAUSE_PARK_Z_FEEDRATE 5             // Z axis feedrate in mm/s (not used for delta printers)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // Initial retract feedrate in mm/s
  #define PAUSE_PARK_RETRACT_LENGTH 2         // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 0     // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
  #define PARK_HEAD_ON_PAUSE                  // Go to filament change position on pause, return to print position on resume
  //#define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change
#endif

tcm0116, 2017-12-19:
I was able to replicate this issue, but only by setting the extruder temperature to less than `EXTRUDE_MINTEMP` prior to pausing the print.

Could you possibly connect the printer to a computer and monitor the output from the terminal while attempting to pause a print? There might be some output that could be helpful in diagnosing the issue.

witorwitor, 2017-12-19:
I was testing PARK_HEAD_ON_PAUSE feature with a COLD extruder. I tried to pause real print (nozzle at 230*) and the nozzle moved away from the print. So it looks like this feature works with hot nozzle only.

tcm0116, 2017-12-19:
The reason for that is because it will retract the filament before it parks
the nozzle. We could modify the process such that it the nozzle isn't warm
enough, then it'll just skip retracting, but continue with the park.

witorwitor, 2017-12-19:
Parking without retraction when cold sounds like a good idea to me. At least people will not be confused (as I was) trying this feature with a cold nozzle.

Roxy-3D, 2017-12-19:
> Parking without retraction when cold sounds like a good idea to me. At least people will not be confused (as I was) trying this feature with a cold nozzle.

A couple of thoughts (and debate is welcome).   If we are going to decide to ignore the retraction when the nozzle is too cold...   Perhaps we should have some limit defined.  Right now, UBL assumes +/- 3 degrees is close enough for normal operation.   Perhaps that should be changed to what ever limit we define here?

The second though is if we park a cold nozzle without retraction, we probably want to heat and purge the nozzle just as if it was parked hot and retracted.   Right?

thinkyhead, 2017-12-20:
> The second thought is if we park a cold nozzle without retraction, we probably want to heat and purge the nozzle just as if it was parked hot and retracted. Right?

If the nozzle is already heating, but not yet up to temperature, then in that case it should continue to heat after it's parked, then purge as usual. The nozzle shouldn't heat up in any other case, since the machine may just be in dry-run mode or running G-code that doesn't involve heating the nozzle.

tcm0116, 2017-12-20:
I agree with @thinkyhead.

If the nozzle is too cold to extrude when the pause occurs, then the
retract will just be skipped and the nozzle will be parked.

If the nozzle is too cold to extrude when the print is resumed, then the
Continue/Extrude screen will be skipped and the print will immediately
resume.

This also has the benefit of making M125 and M24 work for machines without
a heated extruder (i.e. router or laser).

Roxy-3D, 2017-12-20:
OK.   That sounds logical...

tcm0116, 2017-12-27:
@witorwitor can you verify if this issue is resolved in 1.1.8? If so, you can close this issue.

Witgang, 2017-12-29:
> If the nozzle is too cold to extrude when the print is resumed, then the
Continue/Extrude screen will be skipped and the print will immediately
resume.
are you sure that's logical? like do you really want to resume printing with a cold hotend?

thinkyhead, 2017-12-31:
> are you sure that's logical? like do you really want to resume printing with a cold hotend?

I believe the actual full behavior would be that if the nozzle was heating or heated (and a print job was active) when it was parked, then when the print job is resumed, the nozzle will stay parked while it heats up, purge a little and do a wipe (if that's enabled) before continuing the print job.

That isn't completely implemented yet, but with @tcm0116's work in PRs like #7107 we are getting closer.

closed 2018-01-23 by thinkyhead