MarlinFirmware/Marlin #8838: auto level ,Servo  angle save to eeprom?

jackshi618, 2017-12-19:
open auto level,Servo  angle save to eeprom?

psavva, 2017-12-19:
Can you please be more specific in your question?

Roxy-3D, 2017-12-19:
I think he wants the servo angles to be adjustable and saved to the EEPROM without having to edit and rebuild the firmware.     This sounds like a good thing to do.   But we probably would want to hard allocate 6 or 8 bytes and handle 3 or 4 servos (even if the system only has one).    And of course, there would need to be a way to change the per servo values.   (Yet another MCode...)

jackshi618, 2017-12-20:
Exactly what I want to say

thinkyhead, 2017-12-20:
Anything is possible. You just need to recruit a willing developer to code it.

jackshi618, 2017-12-20:
I want to do it, but I don't know where to start.

Roxy-3D, 2017-12-20:
> I want to do it, but I don't know where to start.

Fork the Marlin repository, modify a Bugfix branch to have your changes working.  And then issue a Pull Request.

Probably the first thing to do is modify Configuration_Store.cpp to have the space allocated for the values.   Then, modify the initialization code to pick up and use those values.   And then define and implement some MCode function to set the values.

closed 2018-08-25 by thinkyhead