MarlinFirmware/Marlin #8839: compile error - "#define FILAMENT_WIDTH_SENSOR"

empakoso, 2017-12-19:
### Compile Error
When I compile with only this feature enabled in configuration_adv.h (after a clean download of 1.1.#-BugFix:

#define FILAMENT_WIDTH_SENSOR

I was trying to test the new volumentric and linear code for the filament width sensor. https://github.com/MarlinFirmware/Marlin/issues/8590 which is a bug fix found here: https://github.com/MarlinFirmware/Marlin/issues/8767

Arduino IDE 1.8.5 compile error:
```
	sketch\temperature.cpp: In static member function 'static void Temperature::manage_heater()':

	temperature.cpp:802: error: 'calculate_volumetric_for_width_sensor' was not declared in this scope

		   calculate_volumetric_for_width_sensor(measurement_delay[meas_shift_index])

																					^

	exit status 1
	'calculate_volumetric_for_width_sensor' was not declared in this scope
```

psavva, 2017-12-19:
Are you using the latest bugfix?

empakoso, 2017-12-19:
Yes downloaded it yesterday and only enabled this feature.  I did not enable or modify the downloaded version in any other way.

thinkyhead, 2017-12-20:
I'll find and patch this asap.

empakoso, 2017-12-20:
Thank you.  Huge gratitude!

thinkyhead, 2017-12-20:
Ok, all patched. I hope the LCD display patches end up working!

closed 2017-12-20 by thinkyhead

empakoso, 2017-12-20:
Ok, thank you.
Stand by.  This is gonna be good.

empakoso, 2017-12-20:
Yes! It works with Linear (or M200 D0) mode

@thinkyhead to add to your comment about the LCD updates;

It shows only 100% on the LCD for the entire print.  I set the length of M405 D1 for my next print just to ensure it is engaged after only 1cm of filament consumed.

I will try again tomorrow.

thinkyhead, 2017-12-20:
If it shows 100% even though the filament width value is changing then the volumetric multiplier isn't being changed. We can try adding some logging to find the source of that issue.

AnHardt, 2017-12-20:
What is changed, applier, displayed at the front of the buffer and what at the end?
When a reinitialisation of the buffer is needed?
Does it make sense to make 'big' changes in the filaments diameter? Changing from 1.75 to 3 usually needs an other nozzle and a reslice.

empakoso, 2017-12-20:
On that note of logging, could this be an option the same way the temperature is reported (without being polled)?  Not sure what the scope is for this ask (my apologies in advance).

empakoso, 2017-12-21:
Using bugfix-v1.1.# downloaded 2017Dec19 19:38
Just completed a considerable sized print (largest done to date).  I would like to add that these new sensor calculations have really improved my print quality.  I also see a difference in the overall consistency. (thank you).

Print completed:
; filament used = 19856.8mm (47.8cm3)
; filament used = 59.7

The display seems to have shown " V100% " throughout the entire print.  Atleast it been like this every time it was checked upon.  The LCD reported a filament measured width of close to " Dia 1.69 " for the entire print.

The startup Gcode in OctoPi is as follows:
M200 D0				; volumetric turned off
M404 W1.75			; Display or set Nom. Fil. Width: "W<dia_mm>"
M405 D1			; Enable FWS flow control. "M405 D<delay_cm>"
;M406				; Disable Filament Sensor flow control
M407                ; show FWS measurement
M75					; start print job timer
M117 Print Started	; LCD note
M150 R255 U255 B0	; LED yellow (for Camera)

Please let me know how I can add some useful logging to track this bug.

empakoso, 2017-12-22:
I just loaded the bugfix-1.1.# downloaded 2017Dec21 8:00am

I am glad to report that the LCD is now properly reporting the multiplier value being applied.  Currently between V102% to V104%

Thank you guys for whatever has been done to make a great feature run perfect.

As reported before, my prints are coming out perfectly!

commit ce6a6919 (2017-12-20): Fix #8839 - filament width sensor