MarlinFirmware/Marlin #8840: [2.0.x] MKS Sbase and ANET_FULL_GRAPHICS_LCD LCD12864 doesn't work.

teddybee, 2017-12-19:
I have a problem with my Anet A6 LCD12864 on MKS Sbase. I enabled the feature in the config file, connect the LCD plug to the Exp1 connector, the J3 to Exp2, but only the backlight is on and keeps beeping.
I have tried to rotate the plugs by 180° without success.
This is the LCD: 
[Anet-3D-printer-Part-controller-RAMPS-1-4-LCD-12864-control-panel-blue-screen.jpg_640x640.jpg](https://ae01.alicdn.com/kf/HTB1kIlwQFXXXXbuXXXXq6xXFXXXz/Freeshipping-1Pcs-Anet-3D-printer-Part-controller-RAMPS-1-4-LCD-12864-control-panel-blue-screen.jpg_640x640.jpg)

kining77, 2018-01-19:
i have the same problem here. Got the KFB2.0 (MKS_base) and the Anet LCD12864. Only backlight and beeper

Ady1306, 2018-04-04:
I had the same Issue, I am very new but I re loaded Marlin conf, then opened the configuration.h files, scrolled down and selected my board (Anet full graphics) It failed to work the first time but then I went back and unchecked Zonestar and just had the Anet selected, sent it to the board once again and it works excellently .... "Load it twice then reset machine"

haudamekki, 2018-04-13:
I have made me a custom cable and connected it like this shematic: https://www.thingiverse.com/thing:2428214
It is for a base 1.3, but works also with the sbase.

thinkyhead, 2018-04-14:
@Ady1306 — What is your `MOTHERBOARD` setting in `Configuration.h`?

boelle, 2019-02-19:
@Ady1306 What is your `MOTHERBOARD` setting in `Configuration.h`?

boelle, 2019-02-19:
@teddybee do you still have this problem?

teddybee, 2019-02-20:
I don't have a working turnbutton, but I will check the wiring that haudamekki shared above. I am sure I didn't do that way.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2020-09-11:
Is anybody still paying attention to this feature request, and interesting in seeing this happen? It would require someone with the right combination of hardware to test any changes with, so can't be completely easily without involvement from one of you.

sjasonsmith, 2020-11-11:
Given the lack of continuing interest I am closing this. I believe this combination of hardware can potentially be damaging if custom cables are not created, and compatible displays are ready available for this board.

closed 2020-11-11 by sjasonsmith