MarlinFirmware/Marlin #8841: [FR] Incorporating height sensor. (Not for auto-bed leveling)

Solari1339, 2017-12-19:
Good day all! 

First off, thanks for making such a great firmware. Do appreciate the struggle and effort that were put into it. 

I am trying to incorporate a simple displacement/height sensor into my 3D printer so that I could measure the height of an object and print on top of it (i.e. 3D printed parts. I know, why don't I just print the entire 3D print the object again?). And potentially/ideally, with the new measured height (in a txt file), I can change (combine the .txt file with .gcode file) the height in the .gcode file (rather than doing it manually with a measuring caliper). I have seen 3D printers/machines that would do that and was curious if I could incorporate the idea using Marlin FW. I am using RAMBo 1.3 and Marlin 1.1.6. This is just an idea/hobby. 

With a step by step approach, the sensor is an Adafruit Time-of-Flight I2C Distance Sensor VL53L0X (it is not accurate but just to get the ball rolling) and I connected it to the RAMBo's I2C pin. Enabling #define EXPERIMENTAL_I2CBUS, I tried a couple of times by pulling the data using M261 A41 but **could not** get the any response and the board restarted (sensor was tested and worked with simple arduino code + Mega2560). (A41 = 0x29 for VL53L0X). I am using Printrun:Pronterface and when I request (using M261), there is no I2C reply and board restarted. 

What am I doing wrong **or** is there a **better way** of doing this? Or am I missing something?
 
Please enlighten me with any information. I have searched and read through previous issues about I2C (#2895 , #3258) and still could not figure it out. 

Thank you very much. I do appreciate all the help I can get.

revilor, 2017-12-19:
So you just enabled I2C in Marlin and connected the VL53L0X to the I2C bus, right? Then I suppose what's missing is the initialization of the sensor. It's no plug'n'play device, you have to tell the sensor what you what it to do. Have a look at the VL53L0X datasheet or check the Adafruit VL53L0X library for what's necessary to get the sensor working.

boelle, 2019-02-19:
@Solari1339 is this one still relavant? if your "issue" is a question then Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

closed 2019-07-16 by Solari1339